espressif/esp32-camera #128: ESP32-Cam with 2 servo (LedcWrite) not responding after 40 seconds

kingyyp, 2020-03-04:
Hello, 
I bought a ESP32-Cam module from Aliexpress, 
https://www.aliexpress.com/item/32971094258.html?spm=a2g0s.9042311.0.0.27424c4dreFgYS 

I am making a Pan / Tilt camera using ESP32-cam module,
after some study and testing, I am able get this uploaded to this esp32-cam module.
however, the code only allow me to control the servo for around 10-40 seconds, after that the module will not responds , 
serial console do not pop up any error or reset message, it just hang and not responding.
need to do a hard reset by pressing the reset button, and then it restart and works for another 10-40 seconds.

I tried 3 different esp32-cam module from different seller, all having the same issue.

I tried to remove all servo and only connect the esp32-cam module with usb power. 
I can open the webpage and see the video, 
However, when i tried to move the servo control. again ,it will stop responding after 10-40 seconds.
( mean this is not power issue ?)

This is a very interesting device and with lots of potential , 
I guess is the problem caused by LedCWrite ? Timer Library ?
as the Video using Ledc library as well 
Please help , 
Full code attached,

[Pan_Tilt_ESP32_LEDC_testing_txt.txt](https://github.com/espressif/esp32-camera/files/4286858/Pan_Tilt_ESP32_LEDC_testing_txt.txt)
`
[image]

MrFielding, 2020-08-15:
Hi, I just have the same issue, did you find a solution???

hfmicro, 2020-10-01:
Hi . this problem accured when you use Camera and Servo Lib because Servo lib using timer 1 that is used by Camera!!
you should try with another servo lib that use another  timer .