espressif/esp32-camera #130: Where to find "esp_camera.h"

dibia65, 2020-03-08:
Pls can anyone help me with where to find the included esp_camera.h and all other included library in the example code of camera web server. I need it for me to better understand what is going on...thanks

vvdevadass, 2020-03-08:
esp_camera.h exists in the following location in the same repo: driver/include/esp_camera.h
https://github.com/espressif/esp32-camera/blob/master/driver/include/esp_camera.h

If you are using arduino, for the rest of the libs, refer https://github.com/espressif/arduino-esp32/tree/master/libraries

dibia65, 2020-03-14:
> esp_camera.h exists in the following location in the same repo: driver/include/esp_camera.h
> https://github.com/espressif/esp32-camera/blob/master/driver/include/esp_camera.h
> 
> If you are using arduino, for the rest of the libs, refer https://github.com/espressif/arduino-esp32/tree/master/libraries

So what you are saying is that these files do not exist in any directory in my personal computer

me-no-dev, 2020-03-16:
what he is saying is where the files are located in the source tree. Where you have downloaded that source tree on your computer, we have no clue. You did not say wether youy use Arduino or IDF and how you try to compile and run the code, so this is as much as anyone could say.

closed 2020-03-16 by me-no-dev

Mayoogh, 2020-07-10:
@me-no-dev  I have installed the ESP32 library on Arduino IDE using the additional board URL. But I am not able to find the `esp_camera.h`

me-no-dev, 2020-07-10:
https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/include/esp32-camera

puyiot, 2020-12-29:
C:\Users\USER NAME\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\tools\sdk\include\esp32-camera\esp_camera.h

me-no-dev, 2020-12-29:
Did you have a look at the link right above your comment?

deepu9876, 2022-02-07:
#include "esp_camera.h"
error: no such file or directory

Viktorinno, 2022-03-10:
I have the same problem. 
#include "esp_camera.h"
error: no such file or directory
I have tried all the links above, I can see its there but there is an error downloading it all the time.

Naveeshdurairaj, 2023-04-23:
> I have the same problem. #include "esp_camera.h" error: no such file or directory I have tried all the links above, I can see its there but there is an error downloading it all the time.

@Viktorinno how to solve this issue bro, please help me