espressif/esp32-camera #134: Core panic error when using frame2bmp

XerXeX, 2020-03-15:
I'm having a problem when using the `frame2bmp`function. As soon as a call the conversion to BMP it reboots.
I am using the lowest resolution. What could be the problem?

My code
`
    #include "esp_camera.h"

    //WROVER-KIT PIN Map
    #define CAM_PIN_PWDN    32 //power down is not used
    #define CAM_PIN_RESET   -1 //software reset will be performed
    #define CAM_PIN_XCLK    0
    #define CAM_PIN_SIOD    26
    #define CAM_PIN_SIOC    27

    #define CAM_PIN_D7      35
    #define CAM_PIN_D6      34
    #define CAM_PIN_D5      39
    #define CAM_PIN_D4      36
    #define CAM_PIN_D3      21
    #define CAM_PIN_D2      19
    #define CAM_PIN_D1      18
    #define CAM_PIN_D0       5
    #define CAM_PIN_VSYNC   25
    #define CAM_PIN_HREF    23
    #define CAM_PIN_PCLK    22

    static camera_config_t camera_config = {
      .pin_pwdn  = CAM_PIN_PWDN,
      .pin_reset = CAM_PIN_RESET,
      .pin_xclk = CAM_PIN_XCLK,
      .pin_sscb_sda = CAM_PIN_SIOD,
      .pin_sscb_scl = CAM_PIN_SIOC,

      .pin_d7 = CAM_PIN_D7,
      .pin_d6 = CAM_PIN_D6,
      .pin_d5 = CAM_PIN_D5,
      .pin_d4 = CAM_PIN_D4,
      .pin_d3 = CAM_PIN_D3,
      .pin_d2 = CAM_PIN_D2,
      .pin_d1 = CAM_PIN_D1,
      .pin_d0 = CAM_PIN_D0,
      .pin_vsync = CAM_PIN_VSYNC,
      .pin_href = CAM_PIN_HREF,
      .pin_pclk = CAM_PIN_PCLK,

      //XCLK 20MHz or 10MHz for OV2640 double FPS (Experimental)
      .xclk_freq_hz = 20000000,
      .ledc_timer = LEDC_TIMER_0,
      .ledc_channel = LEDC_CHANNEL_0,

      .pixel_format = PIXFORMAT_JPEG,//YUV422,GRAYSCALE,RGB565,JPEG
      .frame_size = FRAMESIZE_QQVGA,//QQVGA-QXGA Do not use sizes above QVGA when not JPEG

      .jpeg_quality = 12, //0-63 lower number means higher quality
      .fb_count = 1 //if more than one, i2s runs in continuous mode. Use only with JPEG
    };

    esp_err_t camera_init(){
    //power up the camera if PWDN pin is defined
      if(CAM_PIN_PWDN != -1){
        pinMode(CAM_PIN_PWDN, OUTPUT);
        digitalWrite(CAM_PIN_PWDN, LOW);
      }

      //initialize the camera
      esp_err_t err = esp_camera_init(&camera_config);
      if (err != ESP_OK) {
        ESP_LOGE(TAG, "Camera Init Failed");
        return err;
      }

      return ESP_OK;
    }

    void getColor(){
      camera_fb_t * fb = esp_camera_fb_get();
      if (!fb) {
        Serial.println("Camera capture failed");
      }

      uint8_t * buf = NULL;
      size_t buf_len = 0;
      bool converted = frame2bmp(fb, &buf, &buf_len);
      esp_camera_fb_return(fb);
      if(!converted){
        Serial.println("Convert to BMP failed");
      }
    }

    void setup() {
      Serial.begin(115200);
      camera_init();
    }

    void loop() {
      getColor();
      delay(10000);
    }`

And the error on serial

15:53:44.912 -> ets Jun  8 2016 00:22:57
15:53:44.912 -> 
15:53:44.912 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
15:53:44.912 -> configsip: 0, SPIWP:0xee
15:53:44.912 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:53:44.912 -> mode:DIO, clock div:1
15:53:44.912 -> load:0x3fff0018,len:4
15:53:44.912 -> load:0x3fff001c,len:1216
15:53:44.912 -> ho 0 tail 12 room 4
15:53:44.946 -> load:0x40078000,len:9720
15:53:44.946 -> ho 0 tail 12 room 4
15:53:44.946 -> load:0x40080400,len:6352
15:53:44.946 -> entry 0x400806b8
15:53:45.362 -> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
15:53:45.362 -> Core 1 register dump:
15:53:45.362 -> PC      : 0x400ea92a  PS      : 0x00060430  A0      : 0x800f02dd  A1      : 0x3ffb1d40  
15:53:45.399 -> A2      : 0x00000036  A3      : 0x00000036  A4      : 0x000001e0  A5      : 0x00000010  
15:53:45.399 -> A6      : 0x00000008  A7      : 0x3ffc2228  A8      : 0x00000000  A9      : 0x00000002  
15:53:45.399 -> A10     : 0x00000026  A11     : 0x00000036  A12     : 0x00000030  A13     : 0x00000000  
15:53:45.399 -> A14     : 0x00000f00  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
15:53:45.433 -> EXCVADDR: 0x00000036  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
15:53:45.433 -> 
15:53:45.433 -> Backtrace: 0x400ea92a:0x3ffb1d40 0x400f02da:0x3ffb1d60 0x40061b1b:0x3ffb1d80 0x400ecdbf:0x3ffb1e00 0x400ea9b1:0x3ffb1ec0 0x400eaa69:0x3ffb1ef0 0x400eac6d:0x3ffb1f30 0x400d12b7:0x3ffb1f60 0x400d12f7:0x3ffb1f90 0x400d1a7d:0x3ffb1fb0 0x4008ab09:0x3ffb1fd0
15:53:45.468 -> 
15:53:45.468 -> Rebooting...

me-no-dev, 2020-03-16:
the buf that gets returned by the conversion MUST be freed once used. You are probably running out of memory.

XerXeX, 2020-03-16:
Oh great. I'd think that it got freed when it gets reinitialised. Bud that did the trick. Thanks.

dahmadjid, 2021-06-19:
> the buf that gets returned by the conversion MUST be freed once used. You are probably running out of memory.
how is this done ?