espressif/esp32-camera #142: menuconfig don't work : Unknown CMake command "register_component"

marchanderic44, 2020-04-07:
On windows 10, I manage to make menuconfig work (idf.py menuconfig with ESP-IDF Command Prompt) with the example given by espressif (esp-idf / examples / get-started / hello_world).
Now, if I try this example (esp32-camera), it doesn't understand the "register_component" command in the CMakeLists.txt file.
I tried to force the update of cmake to 3.17.0, but the idf.py of expressif does not want it and only wants its latest version, namely 3.13.4.

Here is my error message:
C:\Users\PROPRIETAIRE\Desktop\MicroControleur\esp-idf_modif\esp\esp32-camera-master>idf.py menuconfig
Checking Python dependencies...
Python requirements from C:\Users\PROPRIETAIRE\Desktop\MicroControleur\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory c:\users\proprietaire\desktop\microcontroleur\esp-idf_modif\esp\esp32-camera-master\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 c:\users\proprietaire\desktop\microcontroleur\esp-idf_modif\esp\esp32-camera-master"...
Warn about uninitialized values.
-- The C compiler identification is GNU 4.6.3
-- The CXX compiler identification is GNU 4.6.3
-- Check for working C compiler: C:/Users/PROPRIETAIRE/gcc/bin/gcc.exe
-- Check for working C compiler: C:/Users/PROPRIETAIRE/gcc/bin/gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/PROPRIETAIRE/gcc/bin/c++.exe
-- Check for working CXX compiler: C:/Users/PROPRIETAIRE/gcc/bin/c++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:28 (register_component):
  Unknown CMake command "register_component".

-- Configuring incomplete, errors occurred!
See also "C:/Users/PROPRIETAIRE/Desktop/MicroControleur/esp-idf_modif/esp/esp32-camera-master/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

If someone has an idea...
Thank you in advance.
Beautiful day.

me-no-dev, 2020-04-07:
this is a component and not a ready to go application. See https://github.com/espressif/esp-who/tree/master/examples/single_chip/camera_web_server

closed 2020-04-07 by me-no-dev

marchanderic44, 2020-04-07:
Thank you for your reply.
I looked at the example, but I don't quite understand.
I thought that all the files in espressif / esp32-camera were sufficient to open idf.py menuconfiguration, then, after adjustment in this interface, to obtain my firmware by construction.
Used in this way, I thought it was not a ready to go application.
So I have to update another file? or do I modify the CMakeLists.txt file?
In the link that you indicate to me, they specify that you must execute second "2 / Configure the example through idf.py menuconfig;" But it is this environment that I cannot reach.
Thank you for guiding me a little more because I am a beginner.
Thanks again.

igrr, 2020-04-07:
@me-no-dev means that you should enter the example directory and run idf.py there. The root of the repository is a component (library), not an application project. Components (libraries) can not be built on their own, they must be included into some application. The example linked above does precisely that.

marchanderic44, 2020-04-10:
Thank you @igrr and @me-no-dev ,
It's clearer now, and I understand better the esp-idf presentation (which is by default better explained for linux than for windows, and since I don't work on linux ...).
I manage to get the menuconfig environment for the example given by @me-no-dev.

However, I still can't get the menuconfig environment from the tsaarni example.
Here is my overall objective: I am able to obtain an image from camera.capture () in micropython with esp32-cam, in jpeg format, and I will wish to obtain it in bmp format directly on esp32-cam to make image processing locally on the esp32-cam thereafter.

I have to make a mistake somewhere, but I redid the procedure, reread the information and I still can't do it. So I present what I did, telling myself that it will probably be useful later on to someone who may have the same difficulties as me.

I created a main working directory esp /
In this one, I cloned esp-idf:
git clone https://github.com/espressif/esp-idf.git
I go to esp-idf / components / and clone the library that interested me:
git clone https://github.com/tsaarni/esp32-camera-for-micropython.git esp32-camera
I go back to the main directory esp / and clone the tsaarni project:
git clone https://github.com/tsaarni/micropython-with-esp32-cam.git micropython
I go to this esp / micropython directory and run idf.py menuconfig:
C: \ Users \ OWNER \ Desktop \ MicroControleur \ esp \ micropython> idf.py menuconfig
Checking Python dependencies ...
Python requirements from C: \ Users \ OWNER \ Desktop \ MicroControleur \ esp \ esp-idf \ requirements.txt are satisfied.
Executing action: menuconfig
CMakeLists.txt not found in project directory c: \ users \ proprietaire \ desktop \ microcontroller \ esp \ micropython

The project proposed by tsaarni is therefore not configurable with esp-idf menuconfig? and can it only be configured in unix? Could the esp-who project answer my problem in order to obtain an image in bmp format locally? ...

Thank you in advance.

igrr, 2020-04-10:
The micropython project you have linked to describes different installation steps. Have you tried following them?
https://github.com/tsaarni/micropython-with-esp32-cam/wiki#how-to-compile

marchanderic44, 2020-04-10:
Yes, it is among other things this approach that I follow, but without success. I think the problem is that I'm on windows. I can take this one by one:

1 / mkdir micropython
No problem even under windows

2 / cd micropython
No problem

3 / export MICROPYTHON = $ PWD
Windows does not know this command. export can be replaced by SET, $ PWD by echo% cd%. However, SET MICROPYTHON = echo% cd% does not work. I think it is not dramatic, and I move in the directories with the CDs. But maybe my error comes from there. Maybe MICROPYTHON, ESPIDF and SDKCONFIG should be defined for compilation?

4 / virtualenv -p python3 venv
. venv / bin / activate
pip install pyserial pyparsing
I did not create a virtual environment because Pyserial and pyparsing are already installed on my machine.

5 / cd $ MICROPYTHON
I stay at the root of my project.

6 / git clone https://github.com/espressif/esp-idf.git
I can do it again but that has no interest because under windows, I already installed this directory with https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows -setup.html

7 / git checkout <Current supported ESP-IDF commit hash> # run `make` to find out hash for ESP-IDF
I found the hash 5c88c5996dbde6208e3bec05abc21ff6cd822d26 and I do:
git checkout 5c88c5996dbde6208e3bec05abc21ff6cd822d26
But I have the message:
fatal: not a git repository (or any of the parent directories): .git
I am in the main directory of my project (see 5 /). I say to myself that maybe we should go to the esp-idf / directory. But then I have the following error:
error: Your local changes to the following files would be overwritten by checkout:
        components / app_update / otatool.py
        components / efuse / efuse_table_gen.py
        components / efuse / test_efuse_host / efuse_tests.py
        components / espcoredump / espcoredump.py components / nvs_flash / nvs_partition_generator / nvs_partition_gen.py
        components / partition_table / gen_esp32part.py
        components / partition_table / parttool.py components / partition_table / test_gen_esp32part_host / gen_esp32part_tests.py
        components / spiffs / spiffsgen.py
        docs / check_doc_warnings.sh
        docs / gen-version-specific-includes.py
        examples / build_system / cmake / idf_as_lib / build-esp32.sh
        examples / build_system / cmake / idf_as_lib / build.sh
        examples / build_system / cmake / idf_as_lib / run-esp32.sh
        examples / build_system / cmake / idf_as_lib / run.sh
        examples / storage / parttool / parttool_example.py
        examples / system / ota / otatool / otatool_example.py
        install.sh
        tools / build_apps.py
        tools / check_kconfigs.py
        tools / check_python_dependencies.py
        tools / ci / build_examples.sh
        tools / ci / build_examples_cmake.sh
        tools / ci / check-executable.sh
        tools / ci / check_build_warnings.py
        tools / ci / check_deprecated_kconfigs.py
        tools / ci / check_examples_cmake_make.sh
        tools / ci / check_idf_version.sh
        tools / ci / checkout_project_ref.py
        tools / ci / get-full-sources.sh
        tools / ci / get_supported_examples.sh
        tools / ci / mirror-submodule-update.sh
        tools / ci / test_build_system.sh
        tools / ci / test_build_system_cmake.sh
        tools / ci / test_configure_ci_environment.sh
        tools / cmake / convert_to_cmake.py
        tools / docker / entrypoint.sh
        tools / docker / hooks / build
        tools / elf_to_ld.sh
        tools / esp_app_trace / logtrace_proc.py
        tools / esp_app_trace / sysviewtrace_proc.py
        tools / esp_app_trace / test / logtrace / test.sh
        tools / esp_app_trace / test / sysview / test.sh
[... 29 lines cut ...]
The tool chain is normally configured during installation under windows (https://docs.espressif.com/projects/esp-idf/en/v4.0/get-started/index.html#step-3- set-up-the-tools) as well as the environment variables (https://docs.espressif.com/projects/esp-idf/en/v4.0/get-started/index.html#step-4- set-up-the-environment-variables).

8 / cd $ MICROPYTHON / esp-idf / components
git clone https://github.com/tsaarni/esp32-camera-for-micropython.git
No problem

9 / export ESPIDF = ~ / packages / micropython / esp-idf
I don't do it, same as point 3 /

10 / cd $ MICROPYTHON
git clone https://github.com/tsaarni/micropython-with-esp32-cam.git
no problem

11 / cd $ MICROPYTHON / micropython / ports / esp32
make V = 1 SDKCONFIG = boards / sdkconfig.esp32cam -j
Forced problem :(
** WARNING **
The git hash of ESP IDF does not match the supported version
The build may complete and the firmware may work but it is not guaranteed
ESP IDF path:       C:\Users\PROPRIETAIRE\Desktop\MicroControleur\esp\esp-idf
Current git hash:   463a9d8b7f9af8205222b80707f9bdbba7c530e1
Supported git hash: 5c88c5996dbde6208e3bec05abc21ff6cd822d26
Makefile:270: *** target pattern contains no `%'.  Stop.

If you see my error. Thank you in advance.