espressif/esp32-camera #143: library does not handle correctly deep sleep nor light sleep

pabloandresm, 2020-04-17:
When going deepsleep or even lightsleep, there are good chances the esp32 will trigger a watchdog after waking up.

Before entering any sleep mode I tried esp_camera_deinit(), gpio_isr_handler_remove((gpio_num_t)VSYNC_GPIO_NUM), gpio_uninstall_isr_service(), disabling the watchdog, rtc_gpio_isolate(), gpio_deep_sleep_hold_en(), and combinations of the above, and it is the same problem: the esp32 wakes up, a few seconds later watchdog reset.

My guess is that the esp_camera_deinit() is not correct at all, missing very important unitialization code, probably with the ISR, and probably killing a task created for the vsync.

It is a shame, because the most important proyects for the camera are going to sleep and be waken up by a PIR or something, to save battery.

me-no-dev, 2020-04-20:
it's not deinit that is needed. it's holding the camera either in reset or on powerdown mode (depends which pins you have exposed on your particular board). Another question is why would the camera trigger wdt after wakeup? The chip in effect reboots, so it will reset and reinit the camera.

pabloandresm, 2020-04-21:
The ov2640 does not have a reset pin in the ESP32-CAM board, but it does have the PWDN_GPIO pin. I already tried to set it to HIGH before going to sleep. No change. Deepsleep or lightsleep. With or without the gpio_deep_sleep_hold_en() and/or rtc_gpio_isolate(). Problem remains.

I didn't say that the camera triggered the WDT. When the board resumes, either after 1 second light sleep or after a deepsleep starting the setup() cycle again, some seconds after that, it triggers WDT.

One thing I was doing after wakeup is to perform a udp raw socket send, and in most of the cases for that example the watchdog is triggered when the operation finishes and I am in the close(udpsocket).

Another test I did was just to wake up and enter a while(1) { sys_msleep(1);taskYIELD();vTaskDelay(1);yield();sched_yield();delay(0);delay(1);}
and after a couple of seconds....WDT
I added all of those yield functions inside the while loop to yield the CPU I am on and avoid WDT, and it stills trigger.

I even added esp_task_wdt_reset() inside that while loop so I signal any watchdog in my CPU that I m alive and it should not be triggered. Same result. Some seconds. and WDT.

I mentioned the esp_camera_deinit() because in the init() I saw an xTaskCreate() and, without knowing the behaviour of the ESP+WDT+multiCPU+deep/lightsleep, I imagined that it could be a possible source of problem. Meaning that the xTaskCreate was in the esp_camera_init() but didn't see any xTaskDestroy() in the esp32_camera_deinit.

This ESP32 has 2 cores, and there is nothing in the core I am currently executing that could hang to a point of triggering a WDT. I did not create any extra task.

I could reproduce the error and give you a copy/paste of the WDT message if you want.

....and if, instead of using the lightsleep(1 second), I use a delay(1000), no WDT. So the issue has to do with the camera and the esp32 sleep modes, definitely.

I would like to help you debugging and diagnosing this problem. For me is a "showstopper".
thank you

me-no-dev, 2020-04-21:
I definitely would like some logs :)
One thing to point out is that by default the tick rate of FreeRTOS is 100, which means that task switching happens every 10ms. Delaying anything less or equal to that in many cases will not switch the context and you can end up in WDT triggered from the idle task (one with lowest prio, so only when all tasks are sleeping, does it have a chance to run).
You can find the option in the freertos section of menuconfig.

pabloandresm, 2020-04-22:
glad I can help.
please give me some days to debug and upload logs
I was not compiling the FreeRTOS to enable debug. I was just using the Arduino IDE with the esp32 version 1.0.4 from repo https://dl.espressif.com/dl/package_esp32_index.json
thank you

pabloandresm, 2020-04-24:
After playing a lot testing different codes, I arrive to the following..

My code basically does in the setup(): esp_camera_init() and esp_camera_sensor_get()
And in the loop I take a picture and go to sleep: esp_camera_fb_get() and esp_camera_fb_return() and sleep

In the case of DEEPSLEEP:
---------------------------
It works for a couple of times, until I receive the following message and there is no way to recover unless a hard reset via button press:
E (27185) i2c: i2c driver install error
[E][sccb.c:154] SCCB_Write(): SCCB_Write Failed addr:0x30, reg:0xff, data:0x01, ret:-1
[E][sccb.c:154] SCCB_Write(): SCCB_Write Failed addr:0x30, reg:0x12, data:0x80, ret:-1
[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

In the case of LIGHTSLEEP:
----------------------------
After going out of the lightsleep of 1000milliseconds, I call esp_camera_fb_get() and I always receive:
[E][camera.c:495] i2s_run(): Timeout waiting for VSYNC

In the scope of lightsleep, if after this error I perform a esp_camera_deinit() and also a reset via the camera PWDN_GPIO, and again the esp_camera_init(), it works for a while, but with the following message:
E (46306) i2c: i2c driver install error
E (47711) gpio: gpio_install_isr_service(410): GPIO isr service already installed
[E][camera.c:205] skip_frame(): Timeout waiting for VSYNC
[E][camera.c:1270] esp_camera_init(): Camera init failed with error 0x20003

Until a couple of minutes, that even the deinit() does not unblock the situation, and from that point I receive forever this:
E (408207) i2c: i2c driver install error
[E][sccb.c:154] SCCB_Write(): SCCB_Write Failed addr:0x30, reg:0xff, data:0x01, ret:-1
[E][sccb.c:154] SCCB_Write(): SCCB_Write Failed addr:0x30, reg:0x12, data:0x80, ret:-1
[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

In the case of delay() to replace the deepsleep or lightsleep:
---------------------------------------------------------------
Everything works fine.

I also have one small question about the esp_camera_deinit(). The esp_camera_init() performs a gpio_isr_handler_add((gpio_num_t)VSYNC_GPIO_NUM) and a gpio_install_isr_service() and it creates a task with xTaskCreate(). But the esp_camera_deinit() frees resources but does not perform a  gpio_isr_handler_remove((gpio_num_t)VSYNC_GPIO_NUM) nor gpio_uninstall_isr_service() nor xTaskDelete(). Is that OK? I do not know what happens in deepsleep with those tasks and ISR, that is why I ask. I just want to understand and fix things.

Thank you very much

KevWal, 2020-11-11:
hi @pabloandresm, sorry I know this is old but could you upload your code please?  Thanks very much, Kevin

pabloandresm, 2020-11-12:
> 
> 
> hi @pabloandresm, sorry I know this is old but could you upload your code please? Thanks very much, Kevin

Yes sure. Let me clear a little the code so it is the minimum necessary to reproduce the bug and I will give it to you.

For sure there is a bug, and if you have the will to fix it, I will help in everything you need.

My last "big" message was a complete report of what happens in case of lightsleep, deepsleep, and delay.

Do you think you can also give some light on my last question? ----->
---->"I also have one small question about the esp_camera_deinit(). The esp_camera_init() performs a gpio_isr_handler_add((gpio_num_t)VSYNC_GPIO_NUM) and a gpio_install_isr_service() and it creates a task with xTaskCreate(). But the esp_camera_deinit() frees resources but does not perform a gpio_isr_handler_remove((gpio_num_t)VSYNC_GPIO_NUM) nor gpio_uninstall_isr_service() nor xTaskDelete(). Is that OK? I do not know what happens in deepsleep with those tasks and ISR, that is why I ask. I just want to understand and fix things."

Thank you very much. As soon as I clean the code I ll give it to you.

pabloandresm, 2020-11-12:
I attach the code I use. The main part is the setup() and loop().
There are 2 defines. One USE_DEEP_SLEEP, and another USE_LIGHT_SLEEP. If both are undefined, then the code uses vTaskDelay().
So you can test the 3 scenarios. And the outcome is what I've described above.

I use the Arduino IDE with the ESP32 package (in preferences, "additional board manager = https://dl.espressif.com/dl/package_esp32_index.json"). I select board "AI-Thinker ESP32-CAM".
Unfortunately that package was not updated since it's version 1.0.4, from october 2019. I do not know what version of the camera library it has embedded.

I attach the code. Please let me know how could I help.
Thank you very much.

[esp32-cam-pir-v2-for-debug.zip](https://github.com/espressif/esp32-camera/files/5532397/esp32-cam-pir-v2-for-debug.zip)