espressif/esp32-camera #144: Color and monochrome image with short delay

Schaggo, 2020-04-24:
Hi, 

I am trying to take a hq monochrome image and a lq color image with the smallest possible delay between them.

I declared two frame buffers (fb, fb1) and a sensor_t (*s)
I init the cam with 2 frame buffers

my function 
````
s->set_special_effect(s, 2);
  s->set_quality(s, 1);
  fb1 = esp_camera_fb_get();
  DBG("bw img taken.");

  //take color lq image
  
   s->set_special_effect(s, 0);
   s->set_quality(s, 7);
  fb = esp_camera_fb_get();
  DBG("Color img taken.");
  counter++;
  EEPROM.write(Countaddr, counter);
  EEPROM.commit();
  return;
}
````
After filling the buffers I upload both via ftp.

The hole show works the first time after boot, but the second time (without reboot) both images are coloured and low quality. 

Does anybody have a hint what's going on?

Schaggo, 2020-04-24:
Ok, reverting the sensor to hq bw at the end of the function works

Schaggo, 2020-04-24:
If anybody knows why, I'd love to learn.

wzli, 2020-04-29:
When you have multiple frame buffers in queue, the frame is taken in background before the call to esp_camera_fb_get().

closed 2021-07-09 by Schaggo