espressif/esp32-camera #146: explain chunked transfer with Multipart Content-Type, streaming protocol esp32

xenpac, 2020-05-15:
Hi All,
i dived into the streaming webserver code.
Server stream response:

HTTP/1.1 200 OK
Content-Type: multipart/x-mixed-replace;boundary=123456789000000000000987654321
Transfer-Encoding: chunked
Access-Control-Allow-Origin: *

Hmmm... so we are using two (2) different splitting schemes  together??

First: Content-Type: multipart/x-mixed-replace;boundary=123456789000000000000987654321
This is "The Multipart Content-Type" using boundary to seperate frames/parts.
The content type multipart/x-mixed-replace was developed as part of a technology to emulate server push and streaming over HTTP. 

Then: Transfer-Encoding: chunked
We are also using chunked http1.1 transfer like:
length\r\n
data\r\n
......

Why are we doing this? wouldnt it be just one or the other?

maybe someone can explain the reasoning...
thanks

me-no-dev, 2020-05-18:
because you have to keep pushing JPEGs that do not yet exists over an already open HTTP connection. The boundary separates the images, chunked encoding allows us to send data without knowing it's total size first. If you want JPG streaming, you need both.

closed 2020-05-18 by me-no-dev

xenpac, 2020-05-22:
Ok, but we are already using "Content-Type: multipart/x-mixed-replace; boundary=BoundaryString",
so we seperate each frame by a "boundary" delimiter, and we are telling that we are sending multible parts.

Also i could not find any "chunk" framing like "length\r\ndata\r\n" in the data stream.

Moreover i did not find where the "Transfer-Encoding: chunked" header-part is added in the "app_httpd.c" file.

It is a bit confusing;)

found this, but??
https://stackoverflow.com/questions/20334859/difference-between-multipart-and-chunked-protocol

I guess that VLC is not able to display the stream because of the "chunked" header?!
effectively it works without it, i already tried.

s60sc, 2020-05-22:
When I wrote my esp32-cam mjpeg streaming extension [https://github.com/s60sc/ESP32-CAM_MJPEG2SD](url), I found that VLC only worked if the mjpeg stream had a character on the very first line prior to the first separator

```
<a space>\r\n
--123456789000000000000987654321\r\n
Content-Type: image/jpeg\r\n
Content-Length: 39102\r\n
\r\n
...
```

me-no-dev, 2020-05-22:
don't know about any of it... latest esp32-camera driver with latest esp-who web-server example works fine on VLC and other players. @xenpac you need to better investigate the stream. Everything that you say is missing is there. I see that you know this and that for HTTP, so how come you do not get this part? If you want to send without chunked you need to provide content-length header. Do you know the length of the stream you are about to send? I don't think so... so that is why chunked is used. It is defined in the idf http server library, don't look for it in the example code. Please go read some more HTTP. You seem to miss the point of multipart and chunked (which are totally different things all together). Then you should maybe read a bit on MJPEG HTTP streaming. It should clear things up.

xenpac, 2020-05-27:
thanks guys ;)

eriktrk, 2021-12-04:
Hi! Sorry for digging this up, but I just encountered something similar and 
Using php builtin server
`HTTP/1.1 200 OK                                                                                                         Host: tarkodu:8001                                                                                                      Date: Fri, 03 Dec 2021 09:14:27 GMT                                                                                     Connection: close                                                                                                       X-Powered-By: PHP/7.3.29-1~deb10u1                                                                                      Content-Type: multipart/x-mixed-replace;boundary=123456789000000000000987654321                                         Transfer-Encoding: chunked                                                                                              Access-Control-Allow-Origin: *                                                                                                                                                                                                                  Content-Type: image/jpeg                                                                                                Content-Length: 4315                                                                                                                                                                                                                            ����^@^PJFIF^@^A^A^A^@^@^@^@^@^@��^@C^@                                                                                 ^G^H    ^H^F                                                                                                                ^H  ^K^K                                                                                                            ^L^O^Y^P^O^N^N^O^_^V^W^R^Y$ &&$ #"(-:1(+6+"#2D36;=@A@'0GLF?K:?@>��^@C^A^K^K^K^O
...`

Using nginx
`HTTP/1.1 200 OK                                                                                                         
Server: nginx/1.14.2                                                                                                    
Date: Fri, 03 Dec 2021 09:15:51 GMT                                                                                     Content-Type: multipart/x-mixed-replace;boundary=123456789000000000000987654321                                         Transfer-Encoding: chunked                                                                                              Connection: keep-alive                                                                                                  Access-Control-Allow-Origin: *                                                                                                                                                                                                                  32                                                                                                                      Content-Type: image/jpeg                                                                                                Content-Length: 4333                                                                                                                                                                                                                                                                                                                                                    10ed                                                                                                                    ����^@^PJFIF^@^A^A^A^@^@^@^@^@^@��^@C^@                                                                                 ^G^H    ^H^F                                                                                                                ^H  ^K^K                                                                                                            ^L^O^Y^P^O^N^N^O^_^V^W^R^Y$ &&$ #"(-:1(+6+"#2D36;=@A@'0GLF?K:?@>��^@C^A^K^K^K^O 
...`