espressif/esp32-camera #147: README should state that Component config > driver configurations > RTCI0 configuration should be active to use this driver

pushtoopen, 2020-05-17:
twi.c uses rtc_gpio_desc and it requires Component config > driver configurations > RTCI0 configuration > "Support array 'rtc_gpio_desc' for ESP32 The README should state this.

also the "Enable PSRAM in `menuconfig`" line should have "Component config > Esp32-Specific > Support for external, SPI supported RAM" added onto it to be very clear for new users.

1st is required to use the driver, 2nd is a suggestion.

me-no-dev, 2020-05-18:
I will guess you are using IDF master. The camera driver is tracking IDF 4.0 currently (as it's a stable version). Things have changed quite a bit in between.

closed 2020-05-18 by me-no-dev

WilliamRoyle, 2021-01-08:
I ran into the same issue. Maybe add it to the readme as a suggestion?