espressif/esp32-camera #148: "JPG Decompression Failed! Data format Error"

pushtoopen, 2020-05-24:
This will be a little complicated. Im using the ESP-EYE and for the most part everything works wonderfully. I'm using the library to execute differencing of two pictures. During the camera capture routine, I execute a calloc of two pictures and i use the esp_camera_fb_get function while initializing to a PIXFORMAT_JPEG and frame buffer = 1. Then i use fmt2rgb888 to convert/load the calloc'ed buffers with the JPEG data. Between pictures, i use the esp_camera_fb_return function before taking the 2nd picture and doing the same thing. This means Finally i ensure all reserved memory is free. Below is my routine.

The 1st time this is executed after a reset, the routine works perfectly. Once the routine is executed again without a reset (again, all calloc'ed buffers are freed and re-reserved), the 2nd execution of the routine fails on the 2nd picture and use of "fmt2rgb888". It fails with the warning: "esp_jpg_decode:JPG Decompression Failed! Data format Error"(4th total time fmt2rgb888 is run). Is anyone aware of how the jd_decomp routine in the esp_jpg_decode method works? It is what is returning the failure. If all i need to do is clean up a buffer somwhere, i would, but i've NULL'ed the fb pointer before and it still doesnt seem to work. 

static esp_err_t capture_handler(httpd_req_t *req){

	camera_fb_t * fb = NULL;

	curfreeheap = esp_get_free_heap_size();
	minfreeheap = esp_get_minimum_free_heap_size();
	ESP_LOGI(TAG, "start current free heap %d",curfreeheap);
	ESP_LOGI(TAG, "minimum free heap seen %d",minfreeheap);

	oldbuf.data = (uint8_t *)calloc(oldbuf.width*oldbuf.height,sizeof(uint8_t)*oldbuf.num_ch);
	if (oldbuf.data == NULL){
		ESP_LOGI(TAG, "oldbuf failed to initialize");
	} else {
		ESP_LOGI(TAG, "oldbuf initialized");
	}
	newbuf.data = (uint8_t *)calloc(newbuf.width*newbuf.height,sizeof(uint8_t)*newbuf.num_ch);
	if (newbuf.data == NULL){
		ESP_LOGI(TAG, "newbuf failed to initialize");
	} else {
		ESP_LOGI(TAG, "newbuf initialized");
	}

	esp_err_t res = ESP_OK;
	int64_t fr_start = esp_timer_get_time();
	fb = esp_camera_fb_get();
	delay(5);
	if (!fb) {
		ESP_LOGE(TAG, "Camera capture failed");
		httpd_resp_send_500(req);
		return ESP_FAIL;
	}
	ESP_LOGI(TAG, "1st fb pointer 1 len: %d",fb->len);
	ESP_LOGI(TAG, "1st fb pointer 1 val: %p",fb->buf);
	ESP_LOGI(TAG, "1st fb format: %d",fb->format);

	fmt2rgb888(fb->buf,fb->len,PIXFORMAT_JPEG,oldbuf.data);
	esp_camera_fb_return(fb);
	ESP_LOGI(TAG, "oldbuf should be filled with camera buffer");
	delay(5);
	fb = esp_camera_fb_get();
	delay(5);
	int64_t fr_end = esp_timer_get_time();
	if (!fb) {
		ESP_LOGE(TAG, "Camera capture failed");
		httpd_resp_send_500(req);
		return ESP_FAIL;
	}
	ESP_LOGI(TAG, "2nd fb pointer 1 len: %d",fb->len);
	ESP_LOGI(TAG, "2nd fb pointer 1 val: %p",fb->buf);
	ESP_LOGI(TAG, "2nd fb format: %d",fb->format); // should be 3/JPEG

	delay(5);
//FAILS AT THIS fmt2rgb888 CALL 2ND TIME ITS CALLED
	fmt2rgb888(fb->buf,fb->len,PIXFORMAT_JPEG,newbuf.data);
	ESP_LOGI(TAG, "newbuf should be filled with camera buffer");

	for (int i = 0; i < newbuf.height*newbuf.width*newbuf.num_ch; i++){
		if (newbuf.data[i] < oldbuf.data[i]){
			newbuf.data[i] = 0;
		} else {
			newbuf.data[i] = newbuf.data[i] - oldbuf.data[i];
		}
	}
	ESP_LOGI(TAG, "oldbuf subtracted from newbuf");

	ESP_LOGI(TAG, "beginning image processing");

	int j = 0;
	//grabbing red channel from new buffer
	for (int i = 2; i < newbuf.width*newbuf.height*newbuf.num_ch; i += 3){
		procbuf.data[j] = newbuf.data[i];
		//procbuf.data[j] /= 255.;
		j++;
	}
	ESP_LOGI(TAG, "copied newbuf to procbuf");

	curfreeheap = esp_get_free_heap_size();
	minfreeheap = esp_get_minimum_free_heap_size();
	ESP_LOGI(TAG, "pre-Binary current free heap %d",curfreeheap);
	ESP_LOGI(TAG, "minimum free heap seen %d",minfreeheap);

	fb->buf = newbuf.data;
	fb->format = PIXFORMAT_RGB888;
	fb->len = newbuf.width*newbuf.height*newbuf.num_ch;
	fb->height = PHOTO_HEIGHT;
	fb->width = PHOTO_WIDTH;

	httpd_resp_set_type(req, "image/jpeg");
	httpd_resp_set_hdr(req, "Content-Disposition", "inline; filename=capture.jpg");

	size_t fb_len = 0;
	if(fb->format == PIXFORMAT_JPEG){
		fb_len = fb->len;
		res = httpd_resp_send(req, (const char *)fb->buf, fb->len);
	} else {
		jpg_chunking_t jchunk = {req, 0};
		res = frame2jpg_cb(fb, 80, jpg_encode_stream, &jchunk)?ESP_OK:ESP_FAIL;
		httpd_resp_send_chunk(req, NULL, 0);
		fb_len = jchunk.len;
	}

	esp_camera_fb_return(fb);
	//fb = NULL;
	//int64_t fr_end = esp_timer_get_time();
	ESP_LOGI(TAG, "JPG: %uB %ums", (uint32_t)(fb_len), (uint32_t)((fr_end - fr_start)/1000));
	if(newbuf.data != NULL){
		free(newbuf.data);
		newbuf.data = NULL;
	}
	if(oldbuf.data != NULL){
		free(oldbuf.data);
		oldbuf.data = NULL;
	}

	return res;
}

me-no-dev, 2020-05-24:
so does this code run at least once successfully?

pushtoopen, 2020-05-25:
yes

me-no-dev, 2020-05-26:
```cpp
	fb->buf = newbuf.data;
	fb->format = PIXFORMAT_RGB888;
	fb->len = newbuf.width*newbuf.height*newbuf.num_ch;
	fb->height = PHOTO_HEIGHT;
	fb->width = PHOTO_WIDTH;
```

You are overwriting the frame structure here and then you are giving it back wrong to the driver with `esp_camera_fb_return(fb);`

pushtoopen, 2020-05-28:
I tried things such as  making a separate camera_fb_t, but assigning the buf pointer of that new camera_fb_t to the newbuf.data caused a crash. (error is below for that). 

So instead, i kept the fb the same, ran esp_camera_fb_return(fb) line after  fmt2rgb888(fb->buf,fb->len,PIXFORMAT_JPEG,newbuf.data); is run.  Then ran the standard code as seen above. Then to finish, before returning the fb at the end of the code, i ran an fb = esp_camera_fb_get() before running esp_camera_fb_return(fb) at the end of the capture_handler function. I did this in hopes that it would reassign the camera_fb_t fb to something completely different (since fb = NULL didnt seem to work). Doing all this still caused the original Data format Error.

The difficulty is understanding how the fb buffer is filled, or if frame2jpg_cb is modifying the camera_fb_t that is input into it. 

Either way, in my standard code i execute an additional step which fills a buffer with one channel from the newbuf.data and i set the fb to a PIXFORMAT_GRAYSCALE  and the 1 channel buffer before frame2jpg_cb is executed, this error does not seem to happen. 

Again, i have difficulty understanding the code that is executed inside of frame2jpg_cb and it seems to modify the camera_fb_t thats fed into it.

I do not believe its because im giving back the wrong buffer with esp_camera_fb_return(fb).

assigning the newbuf.data to a newly declared camera_fb_t->buf caused the error below. I assume it's because the length is not pre-allocated. I do not believe this issue is fixed, but i will close this erro as there is plenty of other issues that need resoilution and it does not currently directly affect my operation.
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d7aa4  PS      : 0x00060530  A0      : 0x80109e32  A1      : 0x3ffc5d40  
0x400d7aa4: capture_handler at /home/mccroryj/esp/WORK/PW_camera/build/../main/app_httpd.c:223 (discriminator 9)

A2      : 0x3ffe7f5c  A3      : 0x3ffe1524  A4      : 0x00000000  A5      : 0x00fcdced  
A6      : 0x3ffb0a38  A7      : 0x3f838420  A8      : 0x800d7a96  A9      : 0x3ffc5cf0  
A10     : 0x00000003  A11     : 0x3f403724  A12     : 0x3f4040f0  A13     : 0x00004821  
A14     : 0x3f403724  A15     : 0x3ffb5350  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4009b509  LEND    : 0x4009b519  LCOUNT  : 0xfffffffc  
0x4009b509: strlen at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:84

0x4009b519: strlen at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:96

closed 2020-05-28 by pushtoopen