espressif/esp32-camera #151: Help: ESP32-CAM crashing while changing the configuration

rrobinet, 2020-06-18:
Environment:
- ESP32-CAM ( CAMERA_MODEL_AI_THINKER )
- ESp32 libray 1.04 (with camera.c and sccb.c updated)
- Arduino IDE 1.8.9

Hi,
I need to change on the fly the ESP32-CAM configuration/settings from QVGA to VGA ,and UXGA the simple sketch below illustrates the issue.
Most of the time the ESP is crashing, adding some delays between changes helps, but is not an absolute solution.
I don't speak about the messages ...
```
E (2164) i2c: i2c driver install error
E (3291) gpio: gpio_install_isr_service(410): GPIO isr service already installed
```
That looks to be normal from what I could read about the reset of the i2c, but after the system crash the system can even not recover from this issue, so I can't rely on EEPROM to run the program from the crash point.
I hope this a known problem with some work around.
Thanks in advance
Robert
**Crash Dump**
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
System is starting
OK
E (2164) i2c: i2c driver install error
E (3291) gpio: gpio_install_isr_service(410): GPIO isr service already installed
E (4380) i2c: i2c driver install error
E (5507) gpio: gpio_install_isr_service(410): GPIO isr service already installed
E (6159) i2c: i2c driver install error
E (7349) gpio: gpio_install_isr_service(410): GPIO isr service already installed
E (9610) i2c: i2c driver install error
E (10737) gpio: gpio_install_isr_service(410): GPIO isr service already installed
E (11826) i2c: i2c driver install error
E (12953) gpio: gpio_install_isr_service(410): GPIO isr service already installed
E (13605) i2c: i2c driver install error
E (14796) gpio: gpio_install_isr_service(410): GPIO isr service already installed
E (17056) i2c: i2c driver install error
E (18183) gpio: gpio_install_isr_service(410): GPIO isr service already installed
E (19272) i2c: i2c driver install error
[E][sccb.c:154] SCCB_Write(): SCCB_Write Failed addr:0x30, reg:0xff, data:0x01, ret:-1
[E][sccb.c:154] SCCB_Write(): SCCB_Write Failed addr:0x30, reg:0x12, data:0x80, ret:-1
[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d13fa  PS      : 0x00060930  A0      : 0x800d14ac  A1      : 0x3ffb1f70  
A2      : 0x00000004  A3      : 0x00000002  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00020004  A7      : 0x00000001  A8      : 0x800d13f8  A9      : 0x3ffb1f50  
A10     : 0x00000000  A11     : 0x00020004  A12     : 0x3ffbedf8  A13     : 0x3ff000c4  
A14     : 0x00989680  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000003c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace: 0x400d13fa:0x3ffb1f70 0x400d14a9:0x3ffb1f90 0x400d276d:0x3ffb1fb0 0x4008cd89:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
System is starting
OK
E (2164) i2c: i2c driver install error
[E][sccb.c:154] SCCB_Write(): SCCB_Write Failed addr:0x30, reg:0xff, data:0x01, ret:-1
[E][sccb.c:154] SCCB_Write(): SCCB_Write Failed addr:0x30, reg:0x12, data:0x80, ret:-1
[E][camera.c:1049] camera_probe(): Detected camera not supported.
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d13fa  PS      : 0x00060630  A0      : 0x800d1490  A1      : 0x3ffb1f70  
A2      : 0x00000006  A3      : 0x00000003  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00020004  A7      : 0x00000001  A8      : 0x800d13f8  A9      : 0x3ffb1f50  
A10     : 0x00000000  A11     : 0x00020004  A12     : 0x3ffbedf8  A13     : 0x3ff000c4  
A14     : 0x3ffb9294  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000003c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace: 0x400d13fa:0x3ffb1f70 0x400d148d:0x3ffb1f90 0x400d276d:0x3ffb1fb0 0x4008cd89:0x3ffb1fd0
```
**Script**
```
#include "esp_camera.h"
// CAMERA_MODEL_AI_THINKER
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27
#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22
camera_config_t config;

float frameTime;

void setup() 
{
    Serial.begin(115200);
    Serial.println ("System is starting");
    Serial.println(setup_camera((framesize_t)FRAMESIZE_QVGA, (pixformat_t)PIXFORMAT_GRAYSCALE) ? "OK" : "ERR INIT");
    camera_fb_t *frame_buffer = esp_camera_fb_get();
}

void loop() 
{
  for (int i = 0; i <10; i++)
  {
   capture_still() ;  
  }
//delay (500); 
  esp_camera_deinit();
//delay (500);  
  setup_camera(FRAMESIZE_VGA, PIXFORMAT_JPEG);
delay (100);
  for (int i = 0; i <10; i++)
  {
    capture_still() ;  
  }  
//delay (500);  
  esp_camera_deinit();
//delay (500);  
  setup_camera(FRAMESIZE_QVGA,PIXFORMAT_GRAYSCALE);
delay (100);
  for (int i = 0; i <10; i++)
  {
    capture_still() ;  
  }  
//delay (500); 
  esp_camera_deinit(); 
//delay (500); 
  setup_camera(FRAMESIZE_UXGA,PIXFORMAT_JPEG);
delay (100);
}

bool setup_camera(framesize_t frameSize, pixformat_t pixFormat) 
{
    config.ledc_channel = LEDC_CHANNEL_0;
    config.ledc_timer = LEDC_TIMER_0;
    config.pin_d0 = Y2_GPIO_NUM;
    config.pin_d1 = Y3_GPIO_NUM;
    config.pin_d2 = Y4_GPIO_NUM;
    config.pin_d3 = Y5_GPIO_NUM;
    config.pin_d4 = Y6_GPIO_NUM;
    config.pin_d5 = Y7_GPIO_NUM;
    config.pin_d6 = Y8_GPIO_NUM;
    config.pin_d7 = Y9_GPIO_NUM;
    config.pin_xclk = XCLK_GPIO_NUM;
    config.pin_pclk = PCLK_GPIO_NUM;
    config.pin_vsync = VSYNC_GPIO_NUM;
    config.pin_href = HREF_GPIO_NUM;
    config.pin_sscb_sda = SIOD_GPIO_NUM;
    config.pin_sscb_scl = SIOC_GPIO_NUM;
    config.pin_pwdn = PWDN_GPIO_NUM;
    config.pin_reset = RESET_GPIO_NUM;
    config.xclk_freq_hz = 20000000;
    config.pixel_format = pixFormat;
    config.frame_size = frameSize;
    config.jpeg_quality = 30;
    config.fb_count = 1;
    bool ok = esp_camera_init(&config) == ESP_OK;
//delay (500);
    sensor_t *sensor = esp_camera_sensor_get();
    sensor->set_framesize(sensor, frameSize);
    sensor->set_pixformat(sensor, pixFormat);
    return ok;
}

bool capture_still() 
  {
    frameTime = millis();
    camera_fb_t *frame_buffer = esp_camera_fb_get();
    if (!frame_buffer)  
    {
      return false;
    }
    else 
    {
      float deltaTime = (float)millis ()-frameTime;
   //   Serial.print ("Frame per second: ");Serial.println (1000.0/deltaTime, 4);
      return true;
    }
}
```

rrobinet, 2020-06-18:
Well it looks that a Camera reset is needed after the esp_cam_deinit ().
Looking a the schematic, the only way to reset the Camera is through the AVDD 2.8 V power, managed by the PWDN_GPIO_NUM   on pin 32 (for CAMERA_MODEL_AI_THINKER).
As far a I could test it the power cycle need to be at least 5 ms.
This is achieved by replacing in my case the esp_camera_deinit() function by a new one:
```
void resetCam (byte tick)
{
 esp_camera_deinit();
 digitalWrite(PWDN_GPIO_NUM,LOW);
 delay (1);
 digitalWrite(PWDN_GPIO_NUM,HIGH);
 delay (tick);
}
```
the (time)tick should be at least 5 (ms).
Please let me know if a better solution is possible.

Schaggo, 2020-06-18:
It could be possible that you have to start with the allocation of the largest frame buffer size (jpegQ 0, res 1600x1200). I don’t know why but increasing the buffer requires a reboot.

rrobinet, 2020-06-22:
Of course changing the frame format requires a change of the camera buffer size. I was expected that the `esp_camera_deinit()` was resetting all camera parameters before a new configuration could take place. I had a look at this function, but I couldn't find the issue.
However this function is needed; power on/off the camera is not enough. So both are needed before changing the configuration.
I will close this topic because it is solved for me.
Robert

closed 2020-06-22 by rrobinet

eried, 2023-02-01:
hey @rrobinet I am trying to switch resolutions and pixel format but the camera always reboots, what else do I need to totally deinit it?

rrobinet, 2023-02-02:
@eried 
Sorry, I don't have a solution for you. I did stop the development of my camera, which was becoming, not reliable, too complex and ... more expensive than a professional one (see for instance  https://reolink.com/, cheap and very performant. Robert