espressif/esp32-camera #155: [E][camera.c:1344] esp_camera_fb_get(): Failed to get the frame on time! Camera capture failed

andifirman, 2020-06-24:
I had stuck for this error, I use ESP32 cam with OV 2460 camera module. The device is connect, and it works well when I open an IP that given by router, but the problem is the camera would not capture the image, and also i can't start streaming. Here my code

#include "esp_camera.h"
#include <WiFi.h>
#include "soc/soc.h" //disable brownout problems
#include "soc/rtc_cntl_reg.h"  //disable brownout problems

//
// WARNING!!! Make sure that you have either selected ESP32 Wrover Module,
//            or another board which has PSRAM enabled
//

// Select camera model
//#define CAMERA_MODEL_WROVER_KIT
//#define CAMERA_MODEL_ESP_EYE
//#define CAMERA_MODEL_M5STACK_PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE
#define CAMERA_MODEL_AI_THINKER
//#define CAMERA_MODEL_M5STACK_WITHOUT_PSRAM

#include "camera_pins.h"

const char* ssid = "my_ssid";
const char* password = "12345678";

void startCameraServer();

void setup() {
  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0);
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();

  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;
  //init with high specs to pre-allocate larger buffers
  if(psramFound()){
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }

#if defined(CAMERA_MODEL_ESP_EYE)
  pinMode(13, INPUT_PULLUP);
  pinMode(14, INPUT_PULLUP);
#endif

  // camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }

  sensor_t * s = esp_camera_sensor_get();
  //initial sensors are flipped vertically and colors are a bit saturated
  if (s->id.PID == OV3660_PID) {
    s->set_vflip(s, 1);//flip it back
    s->set_brightness(s, 1);//up the blightness just a bit
    s->set_saturation(s, -2);//lower the saturation
  }
  //drop down frame size for higher initial frame rate
  s->set_framesize(s, FRAMESIZE_QVGA);

#if defined(CAMERA_MODEL_M5STACK_WIDE)
  s->set_vflip(s, 1);
  s->set_hmirror(s, 1);
#endif

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");

  startCameraServer();

  Serial.print("Camera Ready! Use 'http://");
  Serial.print(WiFi.localIP());
  Serial.println("' to connect");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(10000);
}

I also have read the troubleshoot with esp32 cam, but is still won't work

[image]

Schaggo, 2020-06-24:
Can you get a still?

andifirman, 2020-06-25:
> 
> 
> Can you get a still?

Pardon me?

Schaggo, 2020-06-25:
Can you take a still image with the „still“ button?

KamranArain, 2020-06-30:
I am also facing the same issue my controller is HW esp-32 
In this [link](https://electronics-project-hub.com/esp32-cam-tutorial-video-streaming-from-rc-car/) he used it.

andifirman, 2020-07-01:
> 
> 
> I am also facing the same issue my controller is HW esp-32
> In this [link](https://electronics-project-hub.com/esp32-cam-tutorial-video-streaming-from-rc-car/) he used it.
> if you are using XRZ00DI then there is a problem it dont work with any library of arduino. [link] (https://github.com/imShara/ESP32-CAM-XRZ00D1-V240-HTTP-AP) some one wrote code for that one but its not for ARDUINO IDE it is for ESP SDK

Thank you very much, it very helpful for me, i'll try it now

papadenis, 2020-07-29:
I have three identical modules ESP32-cam purchased from the same store.
The first one worked without errors at first, then started giving the error "esp_camera_fb_get (): Failed to get the frame on time". After some random manipulations, the module started working again.
I wrote a program that takes a series of photos and writes them to an SD Card, and while this program was running, at one point the module started giving an error. That is, everything was going fine, UXGA photos were written to a flash drive, and suddenly everything broke. By the way, the last photo, which was saved before his death - with SVGA 800x600 resolution.
Then I tried different implementations from different authors in different IDEs (Arduino and Visual Studio), tried to disable the use of PSRAM, used forcibly the VGA resolution - all this did not affect the error in any way.
I tried the simplest example on the second module - the same error. I tried the same example on the third module - everything WORKED FINE. I tried again on the first module - it doesn't work.

LuisPignataro, 2020-08-06:
I have the same problem. In my case the module worked for a few hours transmitting without problems. Then he started giving the error.
----
I tried everything mentioned above and some other things that I found in various posts. For example changing the frequency of the camera.
There are a lot of queries about this problem on several pages, but no concrete answer.

tuxidog, 2020-08-27:
I have same problem with andifirman  but I have 2 set of ESP32 Cam (2 camera and 2 ESP32 boards.
In my case I got one defected camera from seller. 

I started to test my cameras follow the youtube link 
https://www.youtube.com/watch?v=q-KIpFIbRMk 

( just edit  

#define CAMERA_MODEL_AI_THINKER  
const char* ssid = "my_ssid";
const char* password = "12345678";)

and got the results same as  andifirman. I try to edit the code many times but nothing work.
I change the camera with other one and test cameras again and I got results work fine.
And I test two camera many times to make sure one of them is defect.

finally the first camera is defected.

andifirman, 2020-08-28:
> finally the first camera is defected.

after I tried repeatedly in the same way as you, it looks like my camera is also damaged, I don't understand how it can be damaged, even though I just bought it

me-no-dev, 2020-08-28:
That is because the sensors are EOL and whatever is left out there in stock is what get's to the cameras. It looks like manufacturers are no longer interested in producing sensors with DVP and JPEG.

NikunjChiluvuri, 2020-09-08:
Is the cam have good connection

andifirman, 2020-09-08:
> 
> 
> Is the cam have good connection

Yaa, i was tried supplying it with 5v and 3.3v and it still won't work. But I just found a library that might match my ESP32-Cam. i hope it will work

NikunjChiluvuri, 2020-09-09:
> > Is the cam have good connection
> 
> Yaa, i was tried supplying it with 5v and 3.3v and it still won't work. But I just found a library that might match my ESP32-Cam. i hope it will work

i meant is the camara correctly connected. because i also got the errror. and i cleaned the camara connector and it started working. so try this also. now it is not working for me. i'm getting an another error and I'm waiting for an answer on my post. if you found the solution, put it in your and mine post. but lets hope that this works.

NikunjChiluvuri, 2020-09-09:
> Yes, i've take a still image but the result is same when i tried to start stream
> [image]
> 
> it is need an internet connection to start the camera?

no. you just have to press start stream.

NikunjChiluvuri, 2020-09-09:
> I had stuck for this error, I use ESP32 cam with OV 2460 camera module. The device is connect, and it works well when I open an IP that given by router, but the problem is the camera would not capture the image, and also i can't start streaming. Here my code
> 
> #include "esp_camera.h"
> #include <WiFi.h>
> #include "soc/soc.h" //disable brownout problems
> #include "soc/rtc_cntl_reg.h" //disable brownout problems
> 
> //
> // WARNING!!! Make sure that you have either selected ESP32 Wrover Module,
> // or another board which has PSRAM enabled
> //
> 
> // Select camera model
> //#define CAMERA_MODEL_WROVER_KIT
> //#define CAMERA_MODEL_ESP_EYE
> //#define CAMERA_MODEL_M5STACK_PSRAM
> //#define CAMERA_MODEL_M5STACK_WIDE
> #define CAMERA_MODEL_AI_THINKER
> //#define CAMERA_MODEL_M5STACK_WITHOUT_PSRAM
> 
> #include "camera_pins.h"
> 
> const char* ssid = "my_ssid";
> const char* password = "12345678";
> 
> void startCameraServer();
> 
> void setup() {
> WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0);
> Serial.begin(115200);
> Serial.setDebugOutput(true);
> Serial.println();
> 
> camera_config_t config;
> config.ledc_channel = LEDC_CHANNEL_0;
> config.ledc_timer = LEDC_TIMER_0;
> config.pin_d0 = Y2_GPIO_NUM;
> config.pin_d1 = Y3_GPIO_NUM;
> config.pin_d2 = Y4_GPIO_NUM;
> config.pin_d3 = Y5_GPIO_NUM;
> config.pin_d4 = Y6_GPIO_NUM;
> config.pin_d5 = Y7_GPIO_NUM;
> config.pin_d6 = Y8_GPIO_NUM;
> config.pin_d7 = Y9_GPIO_NUM;
> config.pin_xclk = XCLK_GPIO_NUM;
> config.pin_pclk = PCLK_GPIO_NUM;
> config.pin_vsync = VSYNC_GPIO_NUM;
> config.pin_href = HREF_GPIO_NUM;
> config.pin_sscb_sda = SIOD_GPIO_NUM;
> config.pin_sscb_scl = SIOC_GPIO_NUM;
> config.pin_pwdn = PWDN_GPIO_NUM;
> config.pin_reset = RESET_GPIO_NUM;
> config.xclk_freq_hz = 20000000;
> config.pixel_format = PIXFORMAT_JPEG;
> //init with high specs to pre-allocate larger buffers
> if(psramFound()){
> config.frame_size = FRAMESIZE_UXGA;
> config.jpeg_quality = 10;
> config.fb_count = 2;
> } else {
> config.frame_size = FRAMESIZE_SVGA;
> config.jpeg_quality = 12;
> config.fb_count = 1;
> }
> 
> #if defined(CAMERA_MODEL_ESP_EYE)
> pinMode(13, INPUT_PULLUP);
> pinMode(14, INPUT_PULLUP);
> #endif
> 
> // camera init
> esp_err_t err = esp_camera_init(&config);
> if (err != ESP_OK) {
> Serial.printf("Camera init failed with error 0x%x", err);
> return;
> }
> 
> sensor_t * s = esp_camera_sensor_get();
> //initial sensors are flipped vertically and colors are a bit saturated
> if (s->id.PID == OV3660_PID) {
> s->set_vflip(s, 1);//flip it back
> s->set_brightness(s, 1);//up the blightness just a bit
> s->set_saturation(s, -2);//lower the saturation
> }
> //drop down frame size for higher initial frame rate
> s->set_framesize(s, FRAMESIZE_QVGA);
> 
> #if defined(CAMERA_MODEL_M5STACK_WIDE)
> s->set_vflip(s, 1);
> s->set_hmirror(s, 1);
> #endif
> 
> WiFi.begin(ssid, password);
> 
> while (WiFi.status() != WL_CONNECTED) {
> delay(500);
> Serial.print(".");
> }
> Serial.println("");
> Serial.println("WiFi connected");
> 
> startCameraServer();
> 
> Serial.print("Camera Ready! Use 'http://");
> Serial.print(WiFi.localIP());
> Serial.println("' to connect");
> }
> 
> void loop() {
> // put your main code here, to run repeatedly:
> delay(10000);
> }
> 
> I also have read the troubleshoot with esp32 cam, but it still won't work
> 
> [image]
hmmmmmmmmm, did you write this code yourself?

andifirman, 2020-09-09:
>

No, i used the examples that they provide at the Example menu

andifirman, 2020-09-09:
> No, i used the examples that they provide at the Example menu

My bad, i found this code from internet and tried to my ESP32-Cam

xgarb, 2020-09-09:
> That is because the sensors are EOL and whatever is left out there in stock is what get's to the cameras. It looks like manufacturers are no longer interested in producing sensors with DVP and JPEG.

Will this affect the other sensors? - OV3660, OV5640 and OV7725

LuisPignataro, 2020-09-09:
In my case I unplugged the flat cable and plugged it back in and it worked. I think there was a false contact or dirt on the connector.

NikunjChiluvuri, 2020-09-10:
> > That is because the sensors are EOL and whatever is left out there in stock is what get's to the cameras. It looks like manufacturers are no longer interested in producing sensors with DVP and JPEG.
> 
> Will this affect the other sensors? - OV3660, OV5640 and OV7725

yes

F99, 2020-09-16:
I (surprise) also have this problem. But the Camera chip is working on 3 other samples with the original image still in flash mem of the AI Thinker ESP32 Cam board clone. Hence, the OV2640 is OK and being detected correctly from8-20MHz XCLK.
If only esp_camera_fb_get would be functioning.
I am using 
Arduino IDE 1.8.5
ESP32-Arduino 1.0.4
Win7 64 bit
CameraWebServer example, provided by espressif

Any hints would be great.

ianiliff, 2020-10-30:
I am having the same issue. I have 4 of these devices and after a few days of use the camera stops capturing images. I have tried multiple different sketches including the one posted here. I think these devices may just be junk hardware. I originally thought it had something to do with the SPIFFs writes, but i cant even get the thing to post to a external web server or write to SD.

May have to switch back to a normal ESP32 and use a standard cam. I was hoping for a low cost solution like this this board to kill two birds with one stone (A camera and a board together in one inexpensive unit). Get what you pay for I guess.

me-no-dev, 2020-11-02:
i have reasons to suspect that heat is the issue. I tested the same firmware on two of the same boards, one cooled and one not and the cooled one did not stop at all, while the other failed in a few days.

eesati, 2021-01-06:
I just figured out more, when you capture an image you should use the esp_camera_fb_return(fb); which allows to use the buffer again. And if this function is not used within scope it will cause this error. There is no issued with camera. 

Cheers

jameszah, 2021-01-26:
Hi, I ran into this post with the same problem. 

I was using a ESP32-CAM board and a ov5640 camera, and it all worked fine, but then I put the software on a different ESP32-CAM module, along with the same camera, and it kept failing with this "fail to get frame on time error".  This old ESP32-CAM board worked fine at full speed with a ov2640 camera, but would not work at high resolutions of the ov5640.

I went back to the new ESP32-CAM board, and both cameras worked fine with the same software, and the old ESP32-CAM worked fine with the ov2640, but the stress of the ov5640 on the old ESP32-CAM board was too much.  I don't think it was "heat" because there was a 2 hour gap in debugging, but the old ESP32-CAM board which had fallen out of trees, and had 5 volts accidently plugged into the IO4 various times over the years, was damaged enough to fail with the ov5640, but would work fine with the ov2640.

So, try another ESP32-CAM board if you have this problem. 😄

arthurborgesdev, 2021-05-09:
> I just figured out more, when you capture an image you should use the esp_camera_fb_return(fb); which allows to use the buffer again. And if this function is not used within scope it will cause this error. There is no issues with camera.
> 
> Cheers

This solved completely my problem!! I changed the code, which makes heavy use of those returns and when I did it, I forgot to add them again. Now it's perfectly solved! Thank you!

muhit313, 2021-05-22:
> Cheers

where to change in code?

muhit313, 2021-05-22:
> This solved completely my problem!! I changed the code, which makes heavy use of those returns and when I did it, I forgot to add them again. Now it's perfectly solved! Thank you!

where to change in code?

Ameyd33, 2021-05-27:
I am facing same problem with mu espcam board please suggest changes

muhit313, 2021-06-05:
> I am facing same problem with mu espcam board please suggest changes
> 
> edited
> I just replace
> config.pixel_format = PIXFORMAT_JPEG;
> to
> config.pixel_format = PIXFORMAT_GRAYSCALE;
> now i am getting image in grey scale

It is not working for me.

ikunovic, 2021-06-06:
I have just make it work, it was this error: 
"E][camera.c:1344] esp_camera_fb_get(): Failed to get the frame on time! Camera capture failed"

**I have just reconnected camera** (cable was little bent, I have disconnected, blow in connectors and carefully connected it deeply into connector and it worked.! 

It works on 3.3 and 5V, streams all the time

I succeded with default example (CameraWebServer, esp32 wroom board and #define CAMERA_MODEL_AI_THINKER // Has PSRAM row uncommented

I followed exactly the directions from this video:
https://randomnerdtutorials.com/esp32-cam-video-streaming-face-recognition-arduino-ide/
direct link :  https://youtu.be/MicAM_A0_lU

ardenpm, 2021-08-23:
I can confirm I am another user that had this issue where re-seating the camera cable fixed the problem for me.

dexterac, 2022-02-19:
I'm getting same error

```
[E][camera.c:1483] esp_camera_fb_get(): Failed to get the frame on time!
Camera capture failed

```

Santi-Sun, 2022-09-22:
I share my experience on the camera capture error.
After further checks of my code I solved this problem by gently overwriting the camera contacts with graphite (a pencil) and then polishing the contacts with an eraser.