espressif/esp32-camera #163: built in ESP32-CAM LED goes to half power during deep sleep

TonyLHansen, 2020-07-28:
I am using one of many ESP32-CAM boards. I have a short sketch that takes a picture, uploads the resulting screen shot to a web site, and goes into deep sleep. Before taking the picture, the built-in LED is turned on, then back off after taking the picture. However, as soon as the board goes into deep sleep though, the LED turns back on at partial power. My code shuts down everything it can, but nothing seems to make a difference.

[image]

[Esp32Cam-Server-DeepSleep.ino.txt](https://github.com/espressif/esp32-camera/files/4990551/Esp32Cam-Server-DeepSleep.ino.txt)

Schaggo, 2020-07-28:
Did you try pulling PIN 4 LOW before sleep?

TonyLHansen, 2020-07-28:
that's on line 202

jameszah, 2020-08-03:
Try this:

```
   pinMode(4, OUTPUT);
   digitalWrite(4, LOW);
   rtc_gpio_hold_en(GPIO_NUM_4);
   gpio_deep_sleep_hold_en();
   esp_deep_sleep_start();
```
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html

void gpio_deep_sleep_hold_en(void)
Enable all digital gpio pad hold function during Deep-sleep.

When the chip is in Deep-sleep mode, all digital gpio will hold the state before sleep, and when the chip is woken up, the status of digital gpio will not be held. Note that the pad hold feature only works when the chip is in Deep-sleep mode, when not in sleep mode, the digital gpio state can be changed even you have called this function.

Power down or call gpio_hold_dis will disable this function, otherwise, the digital gpio hold feature works as long as the chip enter Deep-sleep.

TonyLHansen, 2020-08-05:
YAY

yes, adding the two function calls rtc_gpio_hold_en(GPIO_NUM_4); and gpio_deep_sleep_hold_en(); do indeed appear to fix the issue with the LED going into half-power mode.

Thank you.

closed 2020-08-30 by TonyLHansen

bingxuanying, 2021-04-07:
Hi, does it save the board more power when turning off the build-in LED, or calling these two functions will actually cause the board consuming more energy for holding the status of the digital gpio in sleep mode?

jameszah, 2021-04-07:
I haven't measured it, but these calls just connect that pin to ground, rather than having it floating during deepsleep.  So that would be zero cost, and if the floating state turns on that transistor and its big led, but could cost your 10 or 20 mA.  So shut it off is better.  On some of my boards, I ds-soldered that transistor to keep that big LED off.

bingxuanying, 2021-04-07:
Thank you, James! That's really helpful.