espressif/esp32-camera #166: Double FPS not working with esp-idf but with arduino

NikolasPascher, 2020-08-04:
With the esp32-arduino I can get the double fps (around 26) from my OV2640 CAM in jpg mode using SVGA. 
When I switch to esp-idf I only got 13. 
I do measure the time using (esp_timer_get_time()) before the call of esp_camera_fb_get and after the call. 
In the Arduino I get 39ms, on esp-idf I got 69ms. 

I am using the config
```
          __config.ledc_channel = LEDC_CHANNEL_0;
	  __config.ledc_timer = LEDC_TIMER_0;
	  __config.pin_d0 = Y2_GPIO_NUM;
	  __config.pin_d1 = Y3_GPIO_NUM;
	  __config.pin_d2 = Y4_GPIO_NUM;
	  __config.pin_d3 = Y5_GPIO_NUM;
	  __config.pin_d4 = Y6_GPIO_NUM;
	  __config.pin_d5 = Y7_GPIO_NUM;
	  __config.pin_d6 = Y8_GPIO_NUM;
	  __config.pin_d7 = Y9_GPIO_NUM;
	  __config.pin_xclk = XCLK_GPIO_NUM;
	  __config.pin_pclk = PCLK_GPIO_NUM;
	  __config.pin_vsync = VSYNC_GPIO_NUM;
	  __config.pin_href = HREF_GPIO_NUM;
	  __config.pin_sscb_sda = SIOD_GPIO_NUM;
	  __config.pin_sscb_scl = SIOC_GPIO_NUM;
	  __config.pin_pwdn = PWDN_GPIO_NUM;
	  __config.pin_reset = RESET_GPIO_NUM;
	  __config.xclk_freq_hz = 10000000;
	  __config.pixel_format = PIXFORMAT_JPEG;
	  __config.frame_size = FRAMESIZE_SVGA;
	  __config.jpeg_quality = 40;
	  __config.fb_count = 2;

	  // camera init
	  ESP_ERROR_CHECK(esp_camera_init(&__config));
	  sensor_t * s = esp_camera_sensor_get();
	  s->set_framesize(s, FRAMESIZE_SVGA);
	  s->set_vflip(s, 0);
	  s->set_hmirror(s, 1);
```

I also have enabled PSRAM in menuconfig.

Are there any other dependencies, which I have to consider?
For testing purpose, I do capture the image only. I have no routine sending the image.

```
void TestingGrapCamFrames(void *parameter){
  tools_print_debug(DEBUG_LEVEL_0, "Running Task %s on Core: %d.\n", pcTaskGetTaskName(NULL), xPortGetCoreID());

  (void) parameter;
  int64_t t_start = 0;
  int64_t t_idle = 0;
  int64_t t_capture = 0;
  int64_t t_last_frame = 0;
	uint32_t ulNotifiedValue;
	BaseType_t  xResult;
  char s_frame[3] = "f1";

  //This is to check if there are tweo frame buffers used.
  fb_second = (camera_fb_t *) camera_get_camera_fb_int_t();     
  fb_first = (camera_fb_t * ) camera_get_camera_fb_int_t()->next;

  t_last_frame = esp_timer_get_time();
  if((fb_second != fb_first) && (fb_first != NULL)) {
	esp_camera_fb_return(fb_first);
	esp_camera_fb_return(fb_second);
    tools_print_debug(DEBUG_LEVEL_0, "FB1 is on: %p\n", fb_first);
    tools_print_debug(DEBUG_LEVEL_0, "FB2 is on: %p\n", fb_second);
  }
  else{
    tools_print_debug(DEBUG_LEVEL_0, "FB1 is on: %p and FB2 is on: %p\n", fb_first, fb_second);
    esp_camera_fb_return(fb_first);
    esp_camera_fb_return(fb_second);
  }

  while(true){
    t_start = esp_timer_get_time();
    
    //this is intend for future use. The sending task should notify the frame grabber task, to grab a new frame.
//    xResult = xTaskNotifyWait( pdFALSE,    /* Don't clear bits on entry. */
//  							ULONG_MAX,        /* Clear all bits on exit. */
//  							&ulNotifiedValue, /* Stores the notified value. */
//  							xMaxBlockTime );
    
    //For testing purpose, do not wait for notification of sending task
    xResult = pdPASS;
    ulNotifiedValue = CAPTURE_NEXT_FRAME;

    if( xResult == pdPASS ){
  	  //Notified to get next frame
  	  if((ulNotifiedValue & CAPTURE_NEXT_FRAME) != 0){
  		  t_idle = esp_timer_get_time();
    		fb = esp_camera_fb_get();
    		t_capture = esp_timer_get_time();
    		t_capture = (t_capture - t_idle) / 1000;
		    t_idle = (t_idle - t_start) / 1000;
	
    		if (!fb) {
    		  tools_print_debug(DEBUG_LEVEL_0, "Camera capture failed\n");
    		  my_res = ESP_FAIL;
    		}
    		else{
    		  if(fb->format != PIXFORMAT_JPEG){
    			   bool jpeg_converted = frame2jpg(fb, 80, &_jpg_buf, &_jpg_buf_len);
    			   esp_camera_fb_return(fb);
			       fb = NULL;
			       if(!jpeg_converted){
        				tools_print_debug(DEBUG_LEVEL_0, "JPEG compression failed\n");
        				my_res = ESP_FAIL;
			       }
		      }
    		  else{
    			  if(fb == fb_first){
      				fb_1 = fb;
      				tools_print_debug(DEBUG_LEVEL_2, "New frame 2 notify sender Task\n");
              //notify sending Task, to send new frame
      //				xTaskNotify(xHandleSender, NEW_FRAME_1, eSetBits);
      				s_frame[1] = '1';
    			  }
  			    else if(fb == fb_second){
  				    fb_2 = fb;
  				    tools_print_debug(DEBUG_LEVEL_2, "New frame 2 notify sender Task\n");
              //notify sending Task, to send new frame
    //				xTaskNotify(xHandleSender, NEW_FRAME_2, eSetBits);
  	     			s_frame[1] = '2';
  			    }
  			     else{
    				    tools_print_debug(DEBUG_LEVEL_0, "Unknown Frame Buffer %p\n", fb);
  			     }
		      }
		    }
    		int64_t fr_end = esp_timer_get_time();
    		int64_t frame_time = fr_end - t_last_frame;
    		frame_time /= 1000;
    		float fps_captured = 1000.0f / (float) frame_time;
    		uint32_t avg_frame_time = ra_filter_run(&ra_filter, frame_time);
    		tools_print_debug(DEBUG_LEVEL_1, "Capture %s t_idle %lld t_cap %lld fps %.0f avg %u avg_fps %.1f t_fr %lld t_last %lld t_this %lld\n", s_frame, t_idle, t_capture, fps_captured, avg_frame_time, (float) 1000/avg_frame_time, frame_time, t_last_frame, fr_end);
    		t_last_frame = fr_end;
  	  }//End if((ulNotifiedValue & CAPTURE_NEXT_FRAME) != 0){
  	  else{
  		  tools_print_debug(DEBUG_LEVEL_0, "Unknown Bit set in Frame Grabber\n");
  	  }
  	}//end   if( xResult == pdPASS ){
    else{
    	tools_print_debug(DEBUG_LEVEL_0, "Timeout while waiting for CAPTURE_NEXT_FRAME Frame Grabber\n");
    }
	  esp_camera_fb_return(fb);
  }//while true
}
```

Does anybody have an idea why I get 13fps when I use esp-idf and 26 when I use arduino?
Due to the fact, that I have half the number of frames, I guess the configuration of the 'experimental double' fps is not configured properly.

me-no-dev, 2020-08-04:
set XCLK to 20MHz

closed 2020-08-04 by me-no-dev

NikolasPascher, 2020-08-04:
> [EXTERNAL EMAIL] DO NOT open attachments or click on links from unknown senders or unexpected emails.

It is running.

Thank you so much

This message is for the designated recipient(s) only and may contain privileged or confidential information. If you are not the intended recipient, please advise the sender immediately by reply email and delete this message and any attachments without retaining a copy. No representation is made that this email or any attachments are free of viruses.