espressif/esp32-camera #174: 0x20004 Esp32 cam error

NikunjChiluvuri, 2020-09-08:
Hello, I'm using a esp32 cam. it suddenly threw this error at me, 
[image]

can someone help me??

NikunjChiluvuri, 2020-09-11:
- [ ] fhdhdjjdjd

NikunjChiluvuri, 2020-09-11:
closed because no one responded

NikunjChiluvuri, 2020-09-11:
ok reopened

bingulo, 2020-11-27:
I'm with same issue... I've a esp32cam (I think that it's a AI TINKER clone) and got the same error code. I noticed too that the camera is very hot when esp is on :(

Did you solved this problem?

NikunjChiluvuri, 2020-11-28:
intresting

FrancescoBaroneItaly, 2020-12-04:
I'm trying to re-initialize the OV cam of AI-Thinker board after deep sleep wakeup, but I'm not able to initialize with camera probe error. I also tried to invoke an ESP.restart after wake up, but the only way to initialize the camera is an hardware reset ( and I didn't understand, because hardware reset did not reset camera because camera reset pin is always pulled up ).
Any suggestion? thks!

gin66, 2020-12-08:
check this out: https://github.com/gin66/ESP32_projects/blob/master/prj_05_heizung/src/main.cpp

1., do not let the camera driver power down the module.
2. power on the module before camera i it
3. retry several times with power down/on, if i it fails
4. entering deep sleep: ensure camera/board/camera module power line pull downs/ups set appropriately

gin66, 2020-12-08:
search for init_camera() and case DEEPSLEEP

closed 2021-01-15 by ElectronicsBoy