espressif/esp32-camera #189: Unable to compile for the AI-thinker board

jonathan84clark, 2020-10-22:
Hello,

I was wondering if I could get some help with an issue I am running into. I have installed the libraries for the ESP-32 AI thinker board (with the camera). The pre-compiled libraries work just fine with the board, however I need to be able to modify the library. I was able to compile the library (.a file) with the idf tools, however when I replaced my pre-compiled libraries with the one I compiled I got errors when I attempted to build my Arduino project.

I was wondering if there is a fork or branch of this library that is specifically for the AI-thinker board. I noticed that the library I compiled was larger than the one that was originally in my Arduino libraries folder. I also noticed that the library in my Arduino folder had a data stamp of sometime in 2019, so it is older. I am guessing I just need to find the fork and branch used for the AI-thinker. 

Any help would be great!

Jonathan L Clark

me-no-dev, 2020-10-23:
Which IDF version were you using? Headers have also changed since the version in your Arduino install.

jonathan84clark, 2020-10-23:
I just installed the IDF tools for windows, version 2.3. Yes when I built the camera library I copied the following files.
libesp32-camera.a into Arduino15\packages\esp32\hardware\esp32\1.0.4\tools\sdk\include\esp32-camera\lib
then I copied the following header files
esp_camera.h
esp_jpg_decode.h
img_converters.h
sensor.h
into Arduino15\packages\esp32\hardware\esp32\1.0.4\tools\sdk\include\esp32-camera

I got the following error:
`undefined reference to `rtc_io_num_map'
	
	collect2.exe: error: ld returned 1 exit status
	
	exit status 1
	
Error compiling for board AI Thinker ESP32-CAM.`

What I am hoping is that I'm not getting into dependency soup. It could be that esp_camera.h depends on a specific version of sensor.h which depends on the correct versions of files x, y, z...n which depend on the correct versions of files x1, y1, z1...n and so forth until I have replaced the entire esp32 driver.

me-no-dev, 2020-10-26:
you have definitely compiled the camera driver with a different version of ESP-IDF than 3.3
The error you are getting is caused by the fact that there was a change in the naming of the io map in later esp-idf versions.

ntasher, 2020-10-28:
I also want to use the latest camera drivers - especially to support windowing (zooming). I tried to install ESP-IDF but it fails installation. Is there a way to get all the relevant files (.h, .a) in some ZIP to replace the files in my current 1.0.4 ESP32 package?

closed 2021-09-03 by jonathan84clark