espressif/esp32-camera #192: Precise Frame Sync with 2 ESP32 Cameras

ibprojects, 2020-10-29:
Hello,

I am exploring how to do precise frame sync between two or more cameras and I appreciate any help you can offer.

Is there a way to externally trigger a photo capture on more than one device at the exact same moment using GPIO pins or software?  

Thanks again,
IB

Schaggo, 2020-10-29:
I second this.
I was able to achieve a delay <3 ms (I cannot measure more precisely)  via NTP. 
If anyone knows a way how to achieve complete synch, I would love to know!

me-no-dev, 2020-10-29:
I do not think that his is possible, even if you run the cameras from the same clock, you can not write to them at the same time (setup over I2C). There will always be a difference, unless some default settings of the cameras work for you and you power them up at the same time. I see no other way really.

ibprojects, 2020-12-07:
@me-no-dev would it be possible with two camera connected to two different ESP32s but have a shared clock?

skinkie, 2020-12-18:
What you would actually want here is not frame sync, but even row synchronisation. Typically that requires an FPGA interface to the sensor itself. The datasheet https://datasheetspdf.com/datasheet/OV2640.html suggest the option for external frame synchronisation.

skinkie, 2021-09-03:
Think this should stay open, to have an answer.