espressif/esp32-camera #193: ov5640 camera using the arduino esp32 library 1.04

jameszah, 2020-10-29:
Am I correct that I cannot use the ov5640 camera with the Arduino system and the 1.04 library from Oct 2019?

But the PlatformIO system and libraries will let me use the 5640 code I see here in the github??

me-no-dev, 2020-10-29:
pio is probably using the esp32s2 branch of Arduino which has updated everything. You can try it too :) just install from github using that branch (make sure you uninstall from the board manager first)

closed 2020-10-29 by me-no-dev

jameszah, 2020-11-08:
Ok, I updated my esp32 system to the following ...

`https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json`

... and I tried the ov5640 camera and I got the following (chart below) frame-rates on a test program that just takes pictures and discards them - no processing.  I'm doing a video camera application, so I'm interested in the frame-rates.

The first observation is that the UXGA framerate doubled from 6 to 12 on the old ov2640 camera using the 1.05rc1 software over the 1.04 version - which is quite close the the camera max of 15 fps. 😄 

But the ov5640 camera UXGA dropped down to 4 fps out of a camera max of 30.  And the higher resolutions where all in the same ballpark of 5-6 fps out of a possible 15 fps camera max.

I know the esp32 cannot necessarily keep up with the speed of the ov5640, but do you think these are the maximum rates?  or are there opportunities to dig into the ov5640 code and speed it up.  If the ov2640 can send 12 fps UXGA, you would think the ov5640 could too, and if the esp32 can accept 12 fps from the ov2640, then it should be able to swallow the same from the ov5640.  

I suppose the ov5640 might transmit in a inefficient format designed for the much higher resolutions??

And the lower resolutions which world run at 25 fps or 50 fps on the ov2640, seem to be stuck at 22.7 fps, out of a possible 60, 90, or 120 fps.  The exception is the 1280x720 (720p) which ran at the 25 fps on the test program ... although it seems to produce a video of 1280x1440 on my avi recording program. 😕 

So the question is, Is this the limit of the esp32 abilities, or could a volunteer speed some time to improve it???

- James

[image]

jameszah, 2020-11-08:
One more thing ... I used the regular Arduino compiler here ... not the PlatformIO.

Not sure if anyone will see this inside a closed issue. 😕