espressif/esp32-camera #196: No noise reduction and sharpness with ov2640

raduprv, 2020-11-01:
Is there a way for you guys to add the noise reduction and sharpness control for ov2640 for Arduino?
I found that ov3660 supports those modes, and there is info on the web for the ov2640 sensor as well, for example here: http://www.pudn.com/Download/item/id/573503.html
I tried to do it manually, but unfortunately set_reg is also not supported for this sensor.

Is there any reason why this sensor doesn't have those 2 modes?

me-no-dev, 2020-11-02:
use the esp32s2 branch of Arduino till 2.0.0 is released. It has the updated driver

raduprv, 2020-11-03:
Should I uninstall the current board and install the one from here? https://github.com/espressif/arduino-esp32/tree/esp32s2

me-no-dev, 2020-11-03:
as of now you can also use the package manager and install 1.0.5-rc1 from the development link (replace the one you have): https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json

raduprv, 2020-11-03:
Thanks. I tried that, but I don't see any noticeable difference in NR or sharpening. 

Btw, my pictures come way to dark. This is my code:

sensor_t * s = esp_camera_sensor_get();  
  
  s->set_wpc(s, 1);
  s->set_denoise(s,0);
  s->set_sharpness(s,0);
  s->set_quality(s,5);
  s->set_exposure_ctrl(s,1);
  //s->set_gain_ctrl(s,1);
  //s->set_gainceiling(s,GAINCEILING_2X);
  //s->set_agc_gain(s,10);
  //s->set_aec_value(s,400);
  s->set_dcw(s,1);
  //s->set_dcw(s,1);
  s->set_aec2(s,1);
  s->set_whitebal(s,1);
  s->set_awb_gain(s,1);
  s->set_raw_gma(s,1);
  s->set_lenc(s,1);  

Any idea what I am doing wrong? I tried everything, and I can't make my pics come as they do in the webserver example.

raduprv, 2020-11-04:
Ok, I got it to work!
First, I had to add a delay(500) (might work with a smaller delay) after setting the nee things.
set_sharpness() is NOT implemented in the newer version either, however set_reg() is.
I modified the code I found here: http://read.pudn.com/downloads135/sourcecode/embed/573503/ov2640/ov2640.c__.htm and now changing sharpness works, and it is MUCH better without it (smaller pictures allowing for higher quality settings, and also no more noise sharpening) :)

Let me know if you are interested in the code.

dungdao191299, 2024-06-25:
> Ok, I got it to work! First, I had to add a delay(500) (might work with a smaller delay) after setting the nee things. set_sharpness() is NOT implemented in the newer version either, however [...]

can you share your method? that link is died and i still have serious problem with sharpness with esp32 CAM

raduprv, 2024-06-25:
> can you share your method? that link is died and i still have serious problem with sharpness with esp32 CAM

Yes, take a look at my code here: https://github.com/raduprv/esp32-cam_ov2640-timelapse

dungdao191299, 2024-06-25:
> > can you share your method? that link is died and i still have serious problem with sharpness with esp32 CAM
> 
> Yes, take a look at my code here: https://github.com/raduprv/esp32-cam_ov2640-timelapse

thanh you, i can see you adjust 0x92 and 0x93 to change sharpness value, but how can i change to maximum sharpness with there register? sorry for my bad knowledge

raduprv, 2024-06-25:
You don't want to add sharpness. You want to remove it. If you add sharpness, it will sharpen the noise and cause larger image files that will sometimes cause problems. If you want sharper images, you should process them on a computer (and maybe run them trough a denoising step first).