espressif/esp32-camera #198: Esp32 Camera library does not release I2C driver after initialization

FourFUN, 2020-11-05:
Hello everyone,

I am in the process of building a board that has the I2C lines used by the camera shared with other devices. While writing the code to talk to the other devices, I noticed that after the esp camera initialization, `esp_camera_init() -> camera_probe() -> SCCB_Init() -> i2c_driver_install() `the code does not free the I2C driver by calling `i2c_driver_delete()`.

I understand why this is done by the way, since if the user decides they want to change some camera setting after initialization, the code needs to talk over the I2C lines. If all you are running on the I2C lines is the camera this works great. Unfortunately in the case of my custom, crowded board that is a problem.

I am not here to advocate that a whole project accommodate my needs (I can monkey patch a driver uninstall for my particular needs) but a suggestion might be to allow the user to pass in a `SemaphoreHandle_t` (created as a FreeRTOS mutex) during init that could be used to negotiate access to the I2C driver. If the user passes in `NULL` then operate as before, otherwise make sure to acquire the mutex before using the I2C driver, and uninstall it after using it. Since this is specifically for accommodating multiple use on the same wires, the call to `i2c_param_config()` is unnecessary (unless you want to change the clock speed for each device but that could be left up to user code to address).

This does change the `esp_camera_init()` function parameters but maybe a dedicated `esp_camera_init_with_mutex()' function can be defined instead that is called by `esp_camera_init()` with `NULL` as the mutex value.

In any case I thought I would bring this up as a warning for those trying to do what I do plus perhaps a feature enhancement.

apiesse, 2020-11-13:
Another thing I've noticed (which I might be able to work on someday) is that the `camera_probe()`/`SCCB_Probe()` functions just use the first device found on the I2C bus.
If using esp32-camera lib with other I2c peripherals on a shared bus, I think the `camera_probe()` function should loop trying all devices found until it reaches address 0x7F.

The real problema with shared bus access, to be honest, is the whole Espressif implementation which assumes you either have a single device per bus, or you write you own "bus handler" because no semaphores/mutextes are present.
If you're interested integrating this functionality, I've got an unterminated mockup which has been partially tested. just let me know and I can share while I have more time to work on it.

FrancescoBaroneItaly, 2020-12-04:
#apiesse , I'm trying to re-initialize the OV cam of AI-Thinker board after deep sleep wakeup, but I'm not able to initialize with camera probe error. I also tried to invoke an ESP.restart after wake up, but the only way to initialize the camera is an hardware reset ( and I didn't understand, because hardware reset did not reset camera because camera reset pin is always pulled up ).
Any suggestion? thks!

FrancescoBaroneItaly, 2020-12-05:
Resolved, after resume from deep sleep I have to specify rtc_gpio_hold_dis for POWER CAMERA PIN and FLASH LED. Without rtc_gpio_hold_dis, the camera was not powered on failing the probe initialization. Hope will help others... bye