espressif/esp32-camera #201: Framerates on ov5640 camera using 1.05rc1

jameszah, 2020-11-08:
Using a ESP32-CAM module with a ov5640 camera:

Ok, I updated my esp32 system to the following ...

`https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json`

... and I tried the ov5640 camera and I got the following (chart below) frame-rates on a test program that just takes pictures and discards them - no processing.  I'm doing a video camera application, so I'm interested in the frame-rates.

The first observation is that the UXGA framerate doubled from 6 to 12 on the old ov2640 camera using the 1.05rc1 software over the 1.04 version - which is quite close the the camera max of 15 fps. 😄 

But the ov5640 camera UXGA dropped down to 4 fps out of a camera max of 30.  And the higher resolutions were all in the same ballpark of 5-6 fps out of a possible 15 fps camera max.

I know the esp32 cannot necessarily keep up with the speed of the ov5640, but do you think these are the maximum rates?  or are there opportunities to dig into the ov5640 code and speed it up.  If the ov2640 can send 12 fps UXGA, you would think the ov5640 could too, and if the esp32 can accept 12 fps from the ov2640, then it should be able to swallow the same from the ov5640.  

I suppose the ov5640 might transmit in a inefficient format designed for the much higher resolutions??

And the lower resolutions which world run at 25 fps or 50 fps on the ov2640, seem to be stuck at 22.7 fps, out of a possible 60, 90, or 120 fps.  The exception is the 1280x720 (720p) which ran at the 25 fps on the test program ... although it seems to produce a video of 1280x1440 on my avi recording program. 😕 

So the question is, Is this the limit of the esp32 abilities, or could a volunteer spend some time to improve it???

- James

[image]

Followup to this post in the "close" department. https://github.com/espressif/esp32-camera/issues/193#issue-732023104

raduprv, 2020-11-12:
Hi
Did you modify the board in any way? What sensor are you using? Can you post a link please?

jameszah, 2020-11-13:
Hi -- I got a ov5640 camera to see if I could get it working on my https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior at Full-HD 30fps like a real movie camera.

But then I had to find the 5640 code, which I discovered here https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json

.. and then tried the original ov2640 camera with the new drivers ... and found the much-improved UXGA speed using the original hardware.  But had less success with the Full-HD 30fps.

I'm just changing the video-recorder code to accommodate the 1.05 drivers.   It seems to use much more sram, and has different constants for framesize where I had just used the numbers.

I've been scanning through the register setups, etc. in the original 1.04 ov2640 code, and the new 1.05 ov2640 code to see how they improved the ov2640 performance for uxga and sxga.  

And then next step, to try to understand the registers and code for the 5640, which I think can be improved as well -- within the limits of the esp32 capabilities. 

I'll update https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior in a day or two for the ov2640 stuff.

raduprv, 2020-11-13:
Can you post a link to the camera module also? I'd like to get one as well. I have a few ov5640  around from other 'toys', but the connectors are not compatible with the ESP32 CAM.

jameszah, 2020-11-13:
[image]

https://usa.banggood.com/Mini-OV7670-or-OV2640-or-OV5640-AF-Camera-Module-CMOS-Image-Sensor-Module-p-1444381.html

raduprv, 2020-11-13:
Thanks, I'll get a few of these and then work on them.
Were you able to capture static full res frames with it? If so, what's the size?

jameszah, 2020-11-20:
I see 1.0.5-rc2 has been released, and I redid a few tests with it.

HD 1280x720 ~25 fps and still 1280x1440 
I think this is a "binning" problem where 2x2 bits get switched to 1 bit, but it must not be binning is the 720 direction.  
I have not figured out all the registers yet.

FHD 1920x1080 ~ 6 fps with correct dimensions

So, no change 😄

jameszah, 2021-01-23:
Doing some tests with 1.05rc6
It seems setting the fb_count = 2 will no longer turn the camera into continuous mode, but rather it must be fb_count = 3.
Also, when fb_count = 1, some of the smaller frame sizes are wrong in the height dimension being twice specified.
If fb_count = 2, then the dimensions are correct, but the frame rates is the same as fb_count = 1.

This was all done with xclk = 200... 

These are all done with the camera init'ed with the largest framesize and quality 6, and then the framesize changed on the fly without re-init.  The new version seems to let you init the camera with a small framesize and then change the framesize and it re-allocates memory, until you get to the largest framesize and bump into the 384000 (?) frame limit.

The super-fast 1280x720 from the rc1 seems to have disappeared.

These are written as a mjpeg movie onto a v30 sd card, so the frame-rate may be slower than the test above which was just running the camera, and not the SD card.
[image]

jameszah, 2021-01-23:
The fb_count = 3 seems to apply to the ov2640 camera as well in 1.05rc6.
xclk = 200... seems to be the faster setting.  Tried 300... didn't work 😄 

[image]

jameszah, 2021-01-23:
https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior

Look in the "test" folder for the program to run through all frame-sizes of the camera, and recording a 15 second mjpeg avi video for each framesize.  The serial monitor will show framesize number, plus the height, width, etc, and speed of picture and sd writer every 100 frames, and again at the end of the 15 seconds.

The framesize is set at the maximun with the camera_init, and then modified for each framesize without redo-ing the init.

closed 2021-01-27 by jameszah

Brainnyblinxy, 2022-09-27:
Hello, I am using ESP32 but with an ov5640 cam. streaming imaging orientation works fine using the setting, however, orientation capture and saves upside down 
[image]

kindly assist

me-no-dev, 2022-09-28:
There is a "H-Mirror" setting that controls that :)

jameszah, 2022-09-28:
What software are you using?  You can flip it or un-flip it with a line of code.

Brainnyblinxy, 2022-10-20:
Thank you, I am using Arduino kindly assist

Brainnyblinxy, 2022-10-24:
> What software are you using? You can flip it or un-flip it with a line of code.

Thank you, I am using Arduino kindly assist

jameszah, 2022-10-24:
See here for adjusting the camera settings

https://dronebotworkshop.com/esp32-cam-microsd/

```
  // Camera quality adjustments
  sensor_t * s = esp_camera_sensor_get();
 
  // HORIZ MIRROR (0 = Disable , 1 = Enable)
  s->set_hmirror(s, 0);
  // VERT FLIP (0 = Disable , 1 = Enable)
  s->set_vflip(s, 0);
```

Brainnyblinxy, 2022-10-26:
> See here for adjusting the camera settings
>
> https://dronebotworkshop.com/esp32-cam-microsd/
>
> ```
>   // Camera quality adjustments
>   sensor_t * s = esp_camera_sensor_get();
>
>   // HORIZ MIRROR (0 = [...]

Thank you is it possible to connect through skype or email?

Brainnyblinxy, 2022-11-15:
Hello @jameszah is it possible to have a Skype exchange. I need assistance with esp32 cam