espressif/esp32-camera #202: Support for custom frame size

vikramdattu, 2020-11-09:
It is very much useful to have a custom framesize support.
This will complement the standard ones defined in sensor.h.

I think this can be done easily by just having another separate variable for custom_resolution alongside resolution table.
And let the user modify it using a custom API.

In `sensor.c`:

```
resolution_info_t custom_resolution = {
    // Custom resolution
    640,  480, ASPECT_RATIO_4X3 /* Default to VGA   */
};

void sensor_set_custom_res(resolution_info_t *resolution_info)
{
    custom_resolution.width = resolution_info->width;
    custom_resolution.height = resolution_info->height;
    custom_resolution.aspect_ratio = resolution_info->aspect_ratio;
}
```

User:
```
  resolution_info_t custom_res = {
    160, 160, ASPECT_RATIO_1X1
  };

  sensor_set_custom_res(resolution_info_t *custom_res);
```

tomte76, 2021-12-18:
I also think it would be nice to set custom resolutions. I would need something like 768x432. I there meanwhile any way to set a custom resolution?

me-no-dev, 2021-12-20:
This is very hard to implement for the different sensors. They all set the resolution by more or less different rules. There is an attempt to expose an API that people can use to set custom resolution for particular sensor: https://github.com/espressif/esp32-camera/blob/master/driver/include/sensor.h#L234

Each parameter might mean a different thing for each sensor, so you would need to look at the sensor source and figure out how each resolution is defined and calculated.

tomte76, 2022-01-07:
Thank you @me-no-dev I tried some configurations on my OV2640 but I did not succeed by now. One question I could not answer by now is whether it will work in JPEG mode at all? I tried to set e.g. 

`s->set_res_raw(s, 0, 0, 1600, 1200, 0, 0, 1600, 1200, 768, 432, true, true);`

This will result in a feed with a 768x432 Part with "blocked" color image at the top and a green part below. The allover size of the image is not matching  768x432. Is there any documentation I can read? My goal is to achieve a resolution of 768x432 in JPEG on an OV2649.

tomte76, 2022-01-14:
I figured it out. The best example I found for Arduino is the esp-camera example. There is a switch in the UI to set the window and this is easy to reverse engineer. My solution on OV2640 is

s->set_res_raw(s, 0, 0, 0, 0, 32, 168, 1536, 864, 768, 432, true, true);

while having config.frame_size set to  FRAMESIZE_UXGA. This results in 768x432 as image information in fb which I can read and stream with MicroRTSP. Having totalX and totalY set to the double of outputX and outputY and enable binning and scaling seems to improve image quality. But maybe I'm just imagining it. Also it is a funny thing to set total and output to a small resolution and then pan around the window by using offsetX/Y to the region of interest. I attached the values of the offset onto a mqtt topic so you can pan around interactively. Very nice.

RicardoBianchi, 2023-04-21:
@tomte76 , where did you insert that line? straight into the .ino file or some library? thank you very much in advance

shapingstuff, 2023-11-15:
> I figured it out. The best example I found for Arduino is the esp-camera example. There is a switch in the UI to set the window and this is easy to reverse engineer. My solution on OV2640 is [...]

Thanks for this. Can you advise how you got rid of the green part below. I'm using this:

s->set_res_raw(s, 0, 0, 0, 0, 0, 80, 1600, 1000, 1600, 960, true, true);

The dimensions are correct however I have a blocked green area on the left-hand side as part of the total_x/width. It appears to be at 1280 so I'm guessing the resolution is not UXGA despite this having been set in the code.

Any solutions?