espressif/esp32-camera #208: abort in SCCB_Init()

DmytroZakharov, 2020-11-23:
SCCB_Init() sometimes causing abort when initializing camera,  that can be fixed by adding   `conf.clk_flags = 0;` before i2c_param_config().

Jermolinski, 2020-12-13:
Hi Dmitry,

the camera initialization always fails in my case (Eclipse + ESP-IDF) due to the same reason. 
Unfortunately I've found your comment only after I've found the error reason by myself.
It's clear for me, that the c compiler does not initialize the i2c_config_t structure in the function SCCB_Init at all because of the wrong declaration:

`i2c_config_t conf;
 conf.mode = I2C_MODE_MASTER;`

The right way to initialize the i2c_config_t  structure in the SCCB_Init on my opinion would be:

`i2c_config_t conf= {
        .mode = I2C_MODE_MASTER
       ,.sda_io_num = pin_sda
       ,.sda_pullup_en = GPIO_PULLUP_ENABLE
       ,.scl_io_num = pin_scl
       ,.scl_pullup_en = GPIO_PULLUP_ENABLE
       ,.master.clk_speed = SCCB_FREQ
    };`

or 

`i2c_config_t conf={};
 conf.mode = I2C_MODE_MASTER;
...`

Best regards,
Igor

yorickvP, 2021-03-02:
Fixed in #219?

closed 2021-09-03 by DmytroZakharov