espressif/esp32-camera #212: Somestimes too short sensor resets during camera init

adeuring, 2020-12-10:
This project has several bug reports mentioning issues with calls of esp_camera_init(). Here is another one, hopefully somewhat useful. 

 I discovered the problem with these cheap boards called ESP32_CAM. The schematics are here: https://raw.githubusercontent.com/SeeedDocument/forum_doc/master/reg/ESP32_CAM_V1.6.pdf 

 I am using the camera library with IDF version 4.2-rc; most config settings are the defaults. 

 Like other people, I had the problem that calls of `esp_camera_init()` occasionally fail with ESP_ERR_CAMERA_NOT_SUPPORTED. After some debugging I decided to look with a DSO what happens during the sensor reset in camera_probe(). The board does not connect any GPIO pin of the ESP32 module with the reset line of the sensor, so the "reset" is done by power-cycling the camera: GPIO32 is driven high, thus switching off a MOSFET (Q2 in the schematics) which connects 2.8V and 1.2V voltage regulators with the board's "main power line" of 3.3V. 

 To see what happens during esp_camera_init(), I connected channel 1 of the DSO to the drain of Q2 and channel 2 to the 3.3V line. (The latter just to ensure that any glitches that might be seen in channel 1 are not caused by some kind of general power supply problems. Unlikely because I used a lab power supply that can deliver more than 10 Ampere but you never know...) 

 The DSO was in "single shot" mode, triggered by a negative slope on channel 1, trigger threshold was 3.2V, i.e., slightly less than the "main power line". The ESP32 was remote controlled via some sort of restful web API from a laptop which also had a USB connection to the DSO. A call of esp_camera_init() can be directly issued via an HTTP connection; a test script on the laptop put the DSO into acquire mode before letting the ESP32 invoke esp_camera_init(); after the call the script checked if the DSO noticed a trigger event. If yes, a screenshot was taken on the DSO. 

 Some results:

 [image]
The above image shows that the camera power was switched off for about 8 ms; 

[image]
Here we have a "switched off" time of 3 ms, and

[image]
[image]
here we have very short pulses, more like something caused by a bad power supply. (Keep in mind that on the output side of the 2.8V and .2V regulators, U1 and U3 respectivly, are 10 µF capacitors, C2, C8, which may be able to just "filter away" this short drop. Might be worth another test run.) 

 The question is: Why are the "power off" times so different? 

 These are the code lines in camera.c that do the reset; 

 ```
    if(config->pin_pwdn >= 0) {
        ESP_LOGD(TAG, "Resetting camera by power down line");
        gpio_config_t conf = { 0 };
        conf.pin_bit_mask = 1LL << config->pin_pwdn;
        conf.mode = GPIO_MODE_OUTPUT;
        gpio_config(&conf);

        // carefull, logic is inverted compared to reset pin
        gpio_set_level(config->pin_pwdn, 1);
        vTaskDelay(10 / portTICK_PERIOD_MS);
        gpio_set_level(config->pin_pwdn, 0);
        vTaskDelay(10 / portTICK_PERIOD_MS);
    }
``` 

 So the only "thing" between th two gpio_set_level() calls is a vTaskDelay(). 

 A caveat, or disclainer or whatever: I am not familiar with the internals of FreeRTOS, so the following is a bit of guessing combined with a real test. I may very well be wrong, but I think I have at least a point that is worth to discuss. 

 Firstly, I checked what the value of portTICK_PERIOD_MS is in my setup: 10. As I understand it, this means that the "clock" of the task has a period of 10 ms. This value is derived from the config parameter `CONFIG_FREERTOS_HZ=100`. This seems to be the default value for IDF. (I am pretty sure that I did not change it but did not double-check in a fresh setup...) 

 This means that vTaskDelay(1) is called. Now my guess how FreeRTOS handles such a call: The affected task is suspended for some time. Where "some time" does not mean "for roughly 1 clock cycle" but "until the next clock tick occurs", or "until the next point in time when the scheduler will check again which tasks should continue/resume". Sine the call of vTaskDelay() is not synchronized with the scheduler's clock, the net "clock tick" can occur a few microseconds later or relatively late, after close to 10 msec. The documentation (https://www.freertos.org/a00127.html) is unfortunately not very enlightening regarding such fine details. 

 But I tried two more tests: The first was to double the value passed to vTaskDelay(), the second was to set `CONFIG_FREERTOS_HZ` to 1000. 

 The results of both tests support my theory about the "timing mechanism" of vTaskDelay(): With `vTaskDelay(20 / portTICK_PERIOD_MS);` and the usual `CONFIG_FREERTOS_HZ=100` I got reset times between 10ms and 20ms, while they were between 9ms and 10ms for `vTaskDelay(10 / portTICK_PERIOD_MS);` and `CONFIG_FREERTOS_HZ=100`. Though the latter tests failed overall utterly with the error ESP_ERR_CAMERA_NOT_SUPPORTED. 

Anyway, I'd like to suggest to replace calls of `vTaskDelay(10 / portTICK_PERIOD_MS)` with something like 

```
#define DELAY_10_MS ((10 / portTICK_PERIOD_MS) < 2? 2:(10 / portTICK_PERIOD_MS))
vTaskDelay(DELAY_10_MS);
```

(Note, BTW, that an integer calculation `10 / portTICK_PERIOD_MS` yields 0 if portTICK_PERIOD_MS is, for example, 20.)

me-no-dev, 2020-12-11:
You might be onto something here... :)
`vTaskDelay(10 / portTICK_PERIOD_MS);` is equal to 1 tick when `CONFIG_FREERTOS_HZ=100` which could lead to some weird delays, depending on when the next RTOS tick is. Maybe we should force `CONFIG_FREERTOS_HZ=1000` in sdkconfig.defaults to ensure that more proper timing is given.

adeuring, 2020-12-11:
> Maybe we should force `CONFIG_FREERTOS_HZ=1000` in sdkconfig.defaults to ensure that more proper timing is given.

I am not sure:

1. people can always change CONFIG_FREERTOS_HZ to any value they like.
2. As I wrote, there seems to lurk another problem in the driver when CONFIG_FREERTOS_HZ is 1000: While the reset behaviour looked great on the DSO, camera_probe() failed consistently. I have absolutely no clue why…

I would prefer a fix that ensures that `vTaskDelay()` is called with a value greater or equal to 2 when there is a hard lower bound for the required delay. Be it an ugly looking macro for the parameter as I suggested or a helper function like this:

```
void vTaskMinDelay(int delay) {
  vTaskDelay(delay > 1? delay : 2);
}
```
(I am not sure about the name. Perhaps better vTaskGuaranteedDelay() or whatever else.)

But since you are a member of the Espressif team: Do you know somebody who could confirm or reject my suspicion about the behaviour of `vTaskDelay()`? Otherwise we are risking to look at the wrong place…

raduprv, 2020-12-30:
So the following lines in Arduino should turn off the camera, right?

  pinMode(32 , OUTPUT);              //GPIO for turning off camera
  digitalWrite(32 , HIGH);

I am having some issues with an ov3660 sensor where it behaves differently after the first power up (I am doing soft resets of the ESP32). However, I tried this method to restart the sensor (in addition to sending it a software reset and powerdown command) and it still does this :/