espressif/esp32-camera #214: ESP32-CAM with Ngrok can capture pics but no streaming

MerlynMetal, 2020-12-13:
Hi,

ESP32-CAM is streaming in browser without problems. However when I try streaming using Ngrok the settings server loads and I can capture stills (where do they go by the way if at all or just use Save picture as?) but when I hit Start Stream it hangs for a while and then I get a broken picture icon (regardless whether I open the link in my browser or on a remote PC). Turning antivirus/firewall off didn't help.
Any help would be appreciated.

[image]

[image]

raduprv, 2020-12-17:
I think it's a software or hardware problem with the board. Mine have very bad wifi connections. They connect to the wifi, but when trying to upload files via ftp to my PC sometimes it works but most of the times it doesn't. The same code worked fine with an ESP8266

MerlynMetal, 2020-12-18:
The board is 50cm from the WiFi router. As I said it works well in the browser but when trying to connect with Ngrok it can only capture stills. So the camera is working but in the same time not working? I don't quite get that.

me-no-dev, 2020-12-29:
`/capture` is for stills `/stream` is for video

MerlynMetal, 2020-12-29:
OK, let me put it this way: WHY the cam can connect through Ngrok and can capture stills BUT can't stream??
How come it works locally in browser but doesn't when accessed through Ngrok tunnel? 
Is this a hardware problem (bad piece) or else?

me-no-dev, 2020-12-31:
the browser calls the `/stream` URL for video. up there you seem to be calling the `/capture` URL. Probably that is the reason. Also, I have no idea what Ngrok is and how it implements MJPEG :)

MerlynMetal, 2020-12-31:
As I already mentioned the browser will stream the video LOCALLY without problems no matter if I use only the cam IP or the IP:81/stream format.
Ngrok is a web service which creates a secure tunnel to access the video stream from anywhere in the world. This is when the stream is not working, only the stills capture.

sachinthaD, 2021-01-12:
> As I already mentioned the browser will stream the video LOCALLY without problems no matter if I use only the cam IP or the IP:81/stream format.
> Ngrok is a web service which creates a secure tunnel [...]

I'm having the same problem. Did you resolve it?

MerlynMetal, 2021-01-12:
Not yet. It looks like the problem is not well known which means adequate help should not be expected. 
I was able to upload pics to a website and take pics in Blynk but live stream is a no go yet.
I recently acquired a 3G mobile WiFi router so further experiments are ahead.

Paulson-G-Jose, 2021-03-12:
Note the port mentioned for stream on your arduino serial monitor, 
Ex: 14:54:02.019 -> Starting web server on port: '80'
     14:54:02.066 -> Starting stream server on port: '81'
     14:54:02.066 -> Camera Ready! Use 'http://192.168.147.41' to connect
now, here the port for stream is 81.
next, on ngrok command line, ngrok.exe http http://192.168.147.41:81
(after your ip, ':' then the port.)
now, once you get the url from ngrok, add '/stream' at the end.
Ex: http://d320bc12ccb4.ngrok.io/stream
That's all.

MerlynMetal, 2021-03-13:
That worked. Thank you very much! 
It looks like most tutorials didn't explain this detail correctly.

Paulson-G-Jose, 2021-03-14:
Welcome.

chupocro, 2021-06-05:
What I did was I moved the stream server to port 80 (same as for the still images) by registering stream URL among other URLs with:

`httpd_register_uri_handler(stream_httpd, &stream_uri);`

Then I commented out the code for starting the stream server on port 81:

```
/*
if (httpd_start(&stream_httpd, &config) == ESP_OK) {
    httpd_register_uri_handler(stream_httpd, &stream_uri);
 }
*/
```

If you want to after moving the stream server to port 80 still use the control panel you have to edit the HTML served by the ESP32 which is in:

`const uint8_t index_ov2640_html_gz[] = { ... }`

inside file **camera_index.h**

Use [Cyberchef](https://gchq.github.io/CyberChef/) to decode the byte array into HTML, recipe =

```
From_Hex('Auto')
Gunzip()
```

change the line:

`var streamUrl = baseHost + ':81'`

to

`var streamUrl = baseHost`

and convert the HTML back into the byte array, recipe =

```
Gzip('Dynamic Huffman Coding','','',false)
To_Hex('0x with comma',16)
```

The last thing is to count the bytes end to edit:

`#define index_ov2640_html_gz_len`

and now everything (control panel + still image + stream) works on port 80.

In the same way you can use any other port too - just change the default port, register stream URL among other URLs, remove the code for starting the stream server on another port and remove concatenation of the string `':81'` from HTML code served by ESP32.

05PRUTHVI11, 2021-06-07:
@Paulson-G-Jose It is working fine in PC but with mobile it is showing "431 Request Header Fields Too Large " what might be the issue can you please help me out

Paulson-G-Jose, 2021-06-15:
@05PRUTHVI11 Go to your android app settings and clear cache for the browser, restart the app and try again. Or try clearing 'cookies and cache' from browser settings as well. This is all about 431, I haven't tried tunneling from a mobile device, assume you are using termux on android??

05PRUTHVI11, 2021-06-30:
yes thanks,  it worked for me by clearing the cache in the browser

05PRUTHVI11, 2021-06-30:
@Paulson-G-Jose  Thanks,  it worked for me by clearing the cache in the browser

closed 2021-09-03 by MerlynMetal

LuizMosciaro, 2022-03-27:
@JosePaulson i hope you get a blessed life for as long as you live, thank you very much. Been researching for it about a month. Might make a video about this tiny and important detail you gave us, and post it on youtube.

samirakhrif1982, 2022-06-06:
> What I did was I moved the stream server to port 80 (same as for the still images) by registering stream URL among other URLs with:
>
> `httpd_register_uri_handler(stream_httpd, &stream_uri);`
>
> Then I [...]

Please post the full code

ousaghro, 2023-02-22:
please i use ESP 32 and model YOLOV3 i test it in pc work fine with video but when i execute it in esp32 not work give me this error :
**AssertionError: No images or videos found in http:/192.168.43.210:81/stream**
ihelp me please this is the command in python : 
python3 detect.py --source http://192.168.1.74:81/stream  --weights converted_ouhda.weights --cfg training/yolov3_ouhda.cfg --names training/object.names