espressif/esp32-camera #217: Cannot Connect ESP-32 Cam to MotionEyeOS

chmod92, 2020-12-16:
Hello,

I've configured 2 ESP-32 Cams and connected them to the network. They are working and I can see the live stream.

Guide I followed:
https://www.instructables.com/A-Super-Easy-Security-Camera-With-the-ESP32-CAM/

I connect to my Raspberrypi3 running MotioneyeOS (it has a USB webcam added already) and I keep getting "Empty reply from server" or "Camera not supported" when trying to connect to the ESP32 cams.

I'm not sure what I'm entirely doing wrong. When trying to connect the cams I've tried the following URLs in MotioneyeOS:

http://192.168.4.33/
http://192.168.4.33:80/
http://192.168.4.33/cam
http://192.168.4.33/stream/1
http://192.168.4.33/stream
http://192.168.4.33:80/cam
http://192.168.4.33:80/stream/1
http://192.168.4.33:80/stream

and same thing for the second camera

I've also tried connecting through Network camera and Simple MJPEG Camera and that didn't work either.

I've seen a few guides that say to set up home assistant, do i need to do that? can it connect without it?

Any help is appreciated.

me-no-dev, 2020-12-16:
Can you open the stream using VLC?

chmod92, 2020-12-16:
> Can you open the stream using VLC?

I will test tomorrow once I'm able to access the cameras. They are at a relatives house.

chmod92, 2020-12-19:
> Can you open the stream using VLC?

it was unable to open the stream. what should i do?

me-no-dev, 2020-12-29:
I would suggest you have a look at https://github.com/espressif/esp-who/tree/master/examples/single_chip/camera_web_server and try that

mariocondesantos, 2021-03-09:
I am having the same problem, in ESP32-CAM I used the following:

```
# Author by airijia.com
# ESP32-CAM
# Lovelace camera card https://github.com/custom-cards/camera-card
# Video tutorial https://youtu.be/-towjof_ixU
substitutions:
  # Modify variables based on your settings
  hostname: 'esp32cam_02'
  ssid: 'm_yssid'
  password: 'my_password'
esphome:
  name: $hostname
  platform: ESP32
  board: esp32dev
wifi:
  ssid: $ssid
  password: $password
  fast_connect: True
api:
  reboot_timeout: 0s
ota:
logger:

# ESP32-CAM
esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32

  name: bedroom
  # ... Frame Settings

# Flashlight
output:
  - platform: gpio
    pin: GPIO4
    id: gpio_4
light:
  - platform: binary
    output: gpio_4
    name: $hostname light
```

edchilvers, 2021-09-16:
> > Can you open the stream using VLC?
> 
> it was unable to open the stream. what should i do?

I had success using port 81. For example:
http://192.168.4.33:81/stream