espressif/esp32-camera #221: How can ı reach 25fps with ov3660

metehanemlik, 2020-12-29:
I made custom esp cam board with 4 mb psram. Currently, max fps is around 15fps. I have tried whole sizes. Is it possible to get higher fps rate. Thanks

me-no-dev, 2020-12-29:
FPS depends on XCLK and what the ESP hardware can take as clocks. I've done my best to get the highest possible stable FPS on any resolution.

closed 2020-12-29 by me-no-dev

alorbach, 2021-01-06:
I have got my hands on a m5stack timer x module with OV3660 and are playing with the sensor. 
I have many OV2640 modules which reach far better performance. However even now I can tell that the OV3660 gives far better image quality results. For

I am using very low resolution (160x120) and direct mjpeg streaming to a http client. For me, performance is most important, I need to process as most images as possible per second even with a low resolution. 

With the OV2640 modules, the FPS was always stable at 34 FPS per second, sometimes even 45-50 FPS. Most modules used xclk at 20MHZ. 

With the new OV3660 sensor , no matter what I set the xclk to, the  best I can achieve is about 28 FPS with 20 MHZ xclk and tripple buffering (fb_count = 3).

To my understanding, with my settings I end up with "50MHz SYSCLK and 10MHz PCLK" here: 
https://github.com/espressif/esp32-camera/blob/0347377b67c2d16996ba6532202f22943018b962/sensors/ov3660.c#L363

Is there any possibility to reach the performance level of the ov2640 sensor? The image quality of the ov3660 is really great, also the Whitebalance modes work far better.