espressif/esp32-camera #222: How to convert an image buffer into Base64  (IDFGH-4506)

NJenny, 2020-12-29:
I am making a project in which i am using OV2640 Ai thinking camera to take images and then convert this image into base64 and send it to the cloud. I was able to take pictures i can see the buffer size around 1.5kb but i am unable to access the raw data and then can someone help me convert this image into Base64 format on esp-idf and **not on arduino.** 
I used this link for [https://github.com/espressif/esp32-camera](url) for taking images .

me-no-dev, 2020-12-31:
You could extract the base64 files from Arduino and use them or use another base64 lib?

NJenny, 2021-01-03:
can you just recommend some library for base64 that i can use in esp-idf and not on arduino

me-no-dev, 2021-01-04:
The one I have tested and can recommend is in Arduino. It's just two files, you can grab just those and include them in your project without having the full Arduino lib there.

niemeyer, 2021-01-14:
@NJenny Have a look at "mbedtls/base64.h", stock in esp-idf.