espressif/esp32-camera #223: Is psram size important?

metehanemlik, 2020-12-29:
I saw that some board has 4mb psram and some board 8mb psram. Which one is better? Can ı increase fps if i use big psram

raduprv, 2020-12-29:
No, FPS has nothing to do with the RAM (so long as you have enough for 2 frames).
My understanding is that the MCU can only access 4mb of RAM at a time, and special tricks are needed to use all of it.

metehanemlik, 2020-12-30:
According to M5stack PSRAM Timer Camera features, it is using 8MB Psram, OV3660 and it has the ability to provide SVGA: 30fps. 

I got only SVGA 15fps with ESP-CAM (psram 4MB).  I think the reason might be psram size. What do you think? 

PSRAM | 8MB
Flash | 4M
Image Sensor | OV3660
Maximum resolution | 3M pixels
Output format | YUV(422/420)/YCbCr422,RGB565/555,8-bit compressed data,8-/10-bit Raw RGB data
Maximum image transmission rate | UXGA/SXGA: 15fps, SVGA: 30fps, CIF: 60fps

me-no-dev, 2020-12-30:
as @raduprv said. PSRAM has nothing to do with it. M5Stack uses this same driver, so FPS is the same too :)

raduprv, 2020-12-30:
OV3550 is a better sensor than the OV2640, that's why it gets more fps.