espressif/esp32-camera #224: Sensor ov2640: set_window does not work for offset_y > 256, possible fix included

fsck-block, 2020-12-30:
Fiddling around with the function set_window(), it shows up that using values for offset_y > 255  will move the window also in x-direction.

This seems to be caused by an incorrect setting of the VHYX register. According to the datasheet VHYX is used like:
```
VHYX[7:0]
  Bit[7]:   V_SIZE[8]
  Bit[6:4]: OFFSET_Y[10:8]
  Bit[3]:   H_SIZE[8]
  Bit[2:0]: OFFSET_X[10:8]

```
This is in contrast to sensors/ov2640.c [function set_window, line 152]
`    {VHYX, ((max_y >> 1) & 0X80) | ((offset_y >> 4) & 0X70) | ((max_x >> 5) & 0X08) | ((offset_y >> 8) & 0X07)},
`
where the variable offset_y is also transferred  into Bit[2:0]

Possible fix: sensors/ov2640.c [function set_window, line 152]
```
152c152
<         {VHYX, ((max_y >> 1) & 0X80) | ((offset_y >> 4) & 0X70) | ((max_x >> 5) & 0X08) | ((offset_y >> 8) & 0X07)},
---
>         {VHYX, ((max_y >> 1) & 0X80) | ((offset_y >> 4) & 0X70) | ((max_x >> 5) & 0X08) | ((offset_x >> 8) & 0X07)},
```

SrijanChaurasiya, 2021-01-11:
It's the right answer for me to day hauling problem, though the image shifts to blue but panning all the way.
Thanks!!

gcalhoun1949, 2021-02-28:
Can anyone provide an example using the set_window function? I have been experimenting with the esp32cam and would like to be able to Pan so I can zoom in and examine a ROI. I am working with Arduino IDE and cannot figure out how to access the function. Newbie so please be kind.
Thanks in advance.

fsck-block, 2021-03-01:
> Can anyone provide an example using the set_window function? 
It's a bit tricky because the function set_window is not directly available as sensor function pointer (see typedef t_sensor).
You have to use the sensor function set_res_raw which calls the hidden function set_window.
Maybe you want to try something like this (not tested by me):
`sensor_t * s = esp_camera_sensor_get();`
`int unused   = 0;`
`s->set_res_raw(s, OV2640_MODE_SVGA, unused, unused, unused, offset_x, offset_y, total_x, total_y, width, height, unused, unused);`

gcalhoun1949, 2021-03-03:
fsck-block wrote: It's a bit tricky because the function set_window is not directly available as sensor function pointer (see typedef t_sensor).
First let me say Thank You for responding. I tried as you suggested and ran into a few problems.  "set_res_raw" is not included in the sensor_t structure in Arduino ESP32 1.0.4 library. I found that it is in the 1.0.5 development release so I installed that. I also got an error trying to use OV2640_MODE_SVGA.  It is enumerated in a private include. I just set it to "0" (OV2640_MODE_UXGA) in my call. The sketch compiled and ran without error. I then tried to change offsets to other than "0". Again it compiled and ran, but instead of moving the region of interest the video was broken into multi-colored streaks. I will keep experimenting and give you an update if I make any progress. Thanks again for the help.

fsck-block, 2021-03-04:
It took a while for me to understand the function. 
My two cents:
- total_x and total_y defines the size on the sensor
- width and height defines the resulting image (may be smaller than the size on the sensor)
- keep the aspect total_x:total_y == width:height
- use values divisible by 8 without remainder
- start with total_x = width and total_y = height, reduce both values by eg. 32 pixels
- next try moving with offset_x or offset_y by 8 pixels

gcalhoun1949, 2021-03-04:
I have been playing with set_window all day. Keeping aspect ratio the same is giving me trouble since both offset_x and offset_y need to vary at the same time. I am beginning to think this is not really the best way to pan. It seems to be more for setting window size and aspect ratio rather than panning. I wish there was a way to access the registers (sensor side) described for windowing in the datasheet.
Thanks again for all you help.

gcalhoun1949, 2021-03-06:
fsck-block wrote: It took a while for me to understand the function.
I got it to work! After reading you explanation I started to understand how set_window works. I was trying to look at the set_framesize function to determine how to use the variables. In that that function the offset variables are used to adjust total_x and total_y to get the correct aspect ratio as well as center the chosen 'mode' frame size. Your explanation made me realize that the total values are independent of the offset values. 
I set total x & y to 1200 to get aspect ratio of 1:1, mode to SVGA (800x600) and w & h to 240 so my frame would match my display. I can now vary the offsets from 0 - 252 and move the image around. I cannot figure the 252 limit, but I can accomplish my goal.
Thanks again for you help. I really appreciate it. I have been playing with Arduino and the ESP for 3-4  years and refused to post anything because of all the negative responses. I am a firm  believer in "see one, do one, teach one" not I learned the hard way so you do the same. It appears you believe the same thing.
Regards!

nqh2412, 2021-07-20:
> ```
> 152c152
> <         {VHYX, ((max_y >> 1) & 0X80) | ((offset_y >> 4) & 0X70) | ((max_x >> 5) & 0X08) | ((offset_y >> 8) & 0X07)},
> ---
> >         {VHYX, ((max_y >> 1) & 0X80) | ((offset_y >> 4) & 0X70) | ((max_x >> 5) & 0X08) | ((offset_x >> 8) & 0X07)},
> ```

Hi. community, 
I am so new to this, I'm using the Arduino IDE and Arduino-esp32 (https://github.com/espressif/arduino-esp32) boards, Although I cannot find the ov2640.c driver in Arduino-esp32 folder, I think it linked to this espressif/esp32-camera somehow. I have pull a request to fix the offset_y issue in the master branch. But Im not sure that it will be effected to the code of Arduino-esp32 (https://github.com/espressif/arduino-esp32)

Is there any way I can implement a custom version of espressif/esp32-camera to Arduino IDE?

Thanks a lot.

me-no-dev, 2021-07-20:
The camera driver in Arduino is precompiled from this repository. Which version of the ESP32 Arduino are you using?

nqh2412, 2021-07-21:
@me-no-dev thank you for your reply,
I used the v2.0 of Arduino-esp32, I found the compile file (arduino-esp32/tools/sdk/esp32/lib/libesp32-camera.a) but it was created 4 days ago. So I guest it will be fixed in the next version.

In the meantime, I was able to get over it by compile the esp32-camera using esp-idf, and use the created libesp32-camera.a to replace the one in arduino-esp32 directory of my computer and it works.

Sincerely,

me-no-dev, 2021-07-21:
You can make a pull request to Arduino with the modified libesp32-camera.a, or it will come next time that libs are updated (which is not guaranteed to be before release)

closed 2021-07-21 by me-no-dev

gcalhoun1949, 2024-04-06:
Garvit,
It has been a while since I worked on this, but I did find my notes on how
to get Pan and Zoom to work. I have attached my notes. I hope it helps.
Regards,
Glen

> Hey @gcalhoun1949 <https://github.com/gcalhoun1949>
>
> I know it's probably been a long time since you worked on this, but can
> you explain a bit more about how you made the set_res_raw function work
> exactly?
>
> int (*set_res_raw) (sensor_t *sensor, int startX, int startY, int endX,
> int endY, int offsetX, int offsetY, int totalX, int totalY, int outputX,
> int outputY, bool scale, bool binning);
>
> I can try according to your workaround,
>
> mode: SVGA
> offset x & y: 0-252
> output X & Y = 240
>
> s->set_res_raw(s, startX, startY, endX, endY, xOffset, yOffset, total_x,
> total_y, outputX, outputY, true, true);
>
> What about the startX, startY, endX, endY?
> What is the resulting image size according to your function call?
>
> I couldn't get it to work, and this is important for me for a project.
>
> Thank you. Best,
> Garvit

OV2640 Specs
Array Size:	1600 x 1200 (UXGA)
Image Sensor Format	Type: 1/4″
Maximum Image Transfer Rate: ***@***.***
                             ***@***.***
                             ***@***.***
Resolution:	UXGA SVGA and below
Output Format(8 bit): YUV(422/420)/YCbCr422
                      RGB565/555
                      8-bit compressed data
                      8/10 bit Raw RGB Data
                      
Camera has Windowing, Panning and Zoom features

Arduino IDE does not have access to the 'set_window' function, but 'set_res_raw' is available. You can control windowing 
by using 'set_res_raw' which calls 'set_window'.

set_res_raw function:

static int set_res_raw(sensor_t *sensor, int startX, int startY, int endX, int endY, int offsetX,
 int offsetY, int totalX, int totalY, int outputX, int outputY, bool scale, bool binning)
 {
   return set_window(sensor, (ov2640_sensor_mode_t)startX, offsetX, offsetY, totalX, totalY, outputX, outputY);
 }

Accessing 'set_window' using 'set_res_raw' requires remapping variables:

       set_window         RAW     

        sensor          sensor
        mode            startX
        unused          startY
        unused          endX
        unused          endY
        offX            offsetX
        offY            offsetY
        totalX          totalX
[... 365 lines cut ...]
    HSTART                  HREFST[7:0]     (0x17)      LSBs in REG32
    HSTOP                   HREFEND[7:0]    (0x18)      LSBs in REG32
    REG32                   REG32[2:0]      (0x32)      Horz. start LSBs [2:0]; end LSBs [5:3]; Pixel Clock {7:6}
    VSTART                  VSTRT[7:0]      (0x19)      LSBs in REG32
    VSTOP                   VEND[7:0]       (0x1A)      LSBs in REG32
    COM1                    COM1            (0x03)      Vert. start LSBs [1:0]; end LSBs [3:2]

/* Sensor register bank FF=0x01*/
//Initialized values
const DRAM_ATTR uint8_t ov2640_settings_cif[][2] = {
    {HSTART, 0x11},  
    {HSTOP, 0x43},
    {REG32, 0x89},
    {VSTART, 0x00},
    {VSTOP, 0x25},
    {COM1, 0x0A},
    
    
Zooming and Panning Registers from description
    ZOOMS[7:0]  (0x49)
    COM19[1:0]  (0x48)
    ARCOM2[2]   (0x34)
 
   ov2640 register addresses listed in:
 https://github.com/espressif/esp32-camera/blob/master/sensors/private_include/ov2640_settings.h

garvitkhera, 2024-04-07:
Thanks for sharing your notes Glen. Really appreciate it.