espressif/esp32-camera #227: ESp32 cam and OV3660 lightmeter register problem

raduprv, 2020-12-30:
So I got a M5stak timer camera with an ov3660 sensor, and after some frustrations with that board (mainly that the only drivers I found crash my Windows 10 sometimes, and the lack of an SD card slot) I moved the sensor on an ESP32 cam.

I am trying to do a timelapse program where I do the exposure manually (the auto exposure is not good enough for all light situations). The sensor has a few registries that will give you the total amount of light it senses (much like the ov2680 but 20 bits instead of 8). The registers are 0x3C1B - 0x3C1D, and I think, based on the documentation, that this is used for light band detection.

This works nicely after I burn the program then do a hard reset. However, after soft resets the value of that register is 0. It does change if I shine a bright light in it, but if I take the light away it goes back to 0, instead of ~20 or so.

My program (in Arduino) looks like so:

Init the camera
Set the registers
Get the image
Reset the sensor and put it in standby
Save the image
Put the camera to sleep and tell it to wake up in 10 seconds

So the first time I start the program, the light value is ok, after that it's not ok anymore.
However, I noticed something strange. If I leave the wire from GPIO 0 connected to ground while the program runs, the sensor gives me the right light value after each soft reset...

Any ideas why this happens, or what else can I do?

chalabit, 2021-06-18:
I'm struggling to turn off the M5 Timer Camera.  While on USB power, the camera module stays on and gets HOT.

? How did you reset the camera sensor and put it in standby?
? How did you put the camera to sleep

raduprv, 2021-06-18:
Hi
I don't remember exactly, but take a look here: https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/datasheet/unit/OV3660_CSP3_DS_1.3_sida.pdf
Basically, you have to write a register to the sensor with the proper value.

chalabit, 2021-06-19:
hmm... I haven't done registers / hex / binary since college (I won't tell ya how long ago that was!!).

Here's what I came up with... would you mind looking at it to see if it's correct... I believe there is only 1 bank on the OV3360., so the first line may not be needed.

Also, how do you determine the mask?

Here's the code and OV3360 specification

      if (s->id.PID == OV3660_PID) {
          s->set_reg(s,0x3008,0xff,0x01);//banksel --- ? Is this needed for ov3660? Is there only 1 bank
          delay(100);
          s->set_reg(s,0x3008,0xff,0x80);//reset 
          delay(100);
          s->set_reg(s,0x3008,0x10,0x40); //stand by   ?? Is the mask correct? 

          delay(100);        
      }
  
For OV3660 -- See Table Page 36 2.6 in the documentation for control registers
>>> Register 0x3008 --- Bit 7 -- Software Reset         10000000  0x80
			Bit 6 -- Software Power Down	01000000  0x40

[OV3660_CSP3_DS_1.3_sida.pdf](https://github.com/espressif/esp32-camera/files/6679875/OV3660_CSP3_DS_1.3_sida.pdf)

raduprv, 2021-06-19:
There is no need to change the bank, that's for OV2660.
Also, the mask is not good. You want to change bit 6, so that's 0x40h. Your mask should also be 0x40, because you only want to change that bit.