espressif/esp32-camera #230: ESP32-CAM over 2G/GPRS - is it possible?

MerlynMetal, 2021-01-03:
Hi,

Can I use the camera at a remote location without WiFi with and access to the GSM network only using a SIM900 module for example?
I don't need it to stream at all times only couple of times per day for 5-10 secs. If not possible taking stills and sending them to a server/cloud is also an option.

raduprv, 2021-01-03:
What you can do is get a 3g/4g modem with a hotspot (you can get them for like 20 Euro from banggood). Then using a mosfet or something turn on the modem and send the data through wifi, then turn the modem off.

MerlynMetal, 2021-01-04:
Thanks for your suggestion. I'm aware of that option although I don't completely understand how the cam will get an IP from the hotspot. The thing is I have couple of SIM900 modules sitting around doing nothing so my idea was to use them with the cam if possible.

raduprv, 2021-01-04:
If you want to stream video, you can't do it via gprs, it's way too slow.

MerlynMetal, 2021-01-04:
Even for 5-10 secs at a time just to take a quick peak what's going on? Even if so is it possible at least to take stills and send them to ftp or to cloud? 
If too much trouble can you recommend a solution on a budget?

raduprv, 2021-01-04:
It doesn't matter how long it is streaming, you can't stream on a 56Kbps or so connection.
The solution is to save the image on the SD, and once or twice a day upload them via your GSM module on a ftp server.
A better solution is what I suggested, using a 3g/4g hotspot instead.

Paulson-G-Jose, 2021-03-14:
If you are looking for a live feed, then No. But if you wanna record a footage with detected motion or such:
Choose the Esp32CAM to capture at 10 frames per second(you can implement a delay of 100ms). If you want a footage of 60s, that's 600 captures, 600 * ~8kbytes @ QVGA Resolution (I chose 100ms sec delay, because ESP needs 70-78ms to process the light at QVGA resolution). Now, store the JPG's to the SD and send it over the 2G connection, higher latency to be expected. Remember, with longer feeds, it'll be difficult. Also, compress it so you may not lose any data, you can inflate once received and convert it to motion JPG, it won't be a good video, but still you'll have something.

MerlynMetal, 2021-03-14:
Thanks again for your input. I got an Alcatel Onetouch 3G modem and managed to upload pics 30 sec on my web page (and I can make a time lapse video from the sequence). Also got the cam working with Telegram so I can take picture whenever I want.

closed 2021-09-03 by MerlynMetal