espressif/esp32-camera #231: ov5640 AF module

raduprv, 2021-01-03:
I have an ov5640 with an AF module, and the focus part just doesn't work. It is permanently set at macro, so it is useless for what I need.
I got the AF firmware from here: https://github.com/xupsh/Vision-Zynq/blob/master/ov5640_init/OV5640.h
I send the commands, and the return code is valid. For example, it says the firmware is running, or that it is in the process of focusing, or the focus is released, etc. But there is no actual movement of the focus, it is always set at macro. Has anyone played with that module?

jameszah, 2021-01-23:
I assume you can query a register to find the state of the focusing.  Can you give us the code for that?  
I have restarted some work on my movie camera.  I am wondering when / how often it focuses while running in continuous picture mode.  Or if that tiny camera that I have could have a motor in it to move the lens.

raduprv, 2021-01-24:
My code looks like so:
  firmware_regs = ov5640_setting_init_embedded_firmware;
  for (i = 0; i < sizeof(ov5640_setting_init_embedded_firmware)/sizeof(ov5640_setting_init_embedded_firmware[0]);++i, ++firmware_regs) 
  {
    //ov5642_write_reg(firmware_regs->u16RegAddr,firmware_regs->u8Val);
    s->set_reg(s,firmware_regs->reg,0xff,firmware_regs->val);
  }

  //set the regs
  s->set_reg(s,0x3000,0x60,0x0);
  s->set_reg(s,0x3004,0x60,0x60);
  s->set_reg(s,0x3001,0x40,0x0);
  s->set_reg(s,0x3005,0x40,0x40);
  
  af_set_step(200);
  single_af();
  delay(200);
  Serial.print("Step response: ");
  Serial.println(s->get_reg(s,0x3029,0xff));

  Serial.print("Focus response 3027: ");
  Serial.println(s->get_reg(s,0x3027,0xff));

You can find more code and how to use it in the link I posted above.
The problem is, it doesn't do anything for me. I get the response that the firmware is loaded and working, and after issuing focusing commands I get the "now the lens is idle" thing, and nothing happens.

jameszah, 2021-01-24:
Interesting ... I was puzzled about the locations 3027 and 3029 as I couldn't find them in ov5640 manual, but now I see they communicate with the focusing code in the MCU, and 3022 is the command, which you must be using in your single_af ().

Have you found a manual (or the source code) for the autofocus microcode?

I see it here at ArduCam https://github.com/ArduCAM/Arduino/blob/6efd4eac93342d5c530e09b06ea26d4636f0d55b/ArduCAM/examples/mini/ArduCAM_Mini_5MP_OV5640_AutoFocus/ArduCAM_Mini_5MP_OV5640_AutoFocus.ino#L208

and the microcode code is there too https://github.com/ArduCAM/Arduino/blob/6efd4eac93342d5c530e09b06ea26d4636f0d55b/ArduCAM/ov5640_regs.h#L9

but I do not see any explanation of how to interact with it other than the example in that ... AutoFocus.ino above.

There is a reasonably active Issues section over there for questions.
https://github.com/ArduCAM/Arduino/issues

If you wanted to shut off the autofocus, you might have to change the microcode?  or fiddle with the registers at 3600 ... I assume your af_set_step () is setting one of the 3600 registers.

Is it your assumption that the autofocus only operates when you make that 3022 call?

It would be interesting to see that autofocus microcode source code to see what it is looking at to decide on focus.

raduprv, 2021-01-24:
There are multiple AF modules (or variations of the sensors). I got the info from here: https://www.arducam.com/downloads/modules/OV5640/OV5640_Software_app_note_parallel.pdf
However, the registers are not for this module, the real registers are at a slightly different location. The link I posted for the firmware has the right registers and also some sample code. It shows how to do AF and manual focus, just look at that code.

hrshovon, 2021-05-04:
Any update on this issue? 
Has anyone got it working?
I am also trying to do the same.

raduprv, 2021-05-04:
I haven't worked at it for a while, but last time I didn't get it to work, so I abandoned the idea.

Schaggo, 2021-05-04:
I would also like to experiment with af...
Don’t have the time or even the hardware yet, but looked a little bit into it. I don’t have the Link at Hand, but somebody seems to have got it to work by calling continuous af for a couple of seconds. If I remember correctly, continuous af is only available for certain models. 

Sorry I can’t be of more help, right now.

hrshovon, 2021-05-06:
Found another code while digging
could it be that the firmware is this?
https://android.googlesource.com/kernel/msm.git/+/511b239792d76efc7456ef92735a1257c95eac44/drivers/media/video/msm/ov5640.h

lancefi, 2021-09-24:
Anyone else tried this? I tried it and I get back the status that the AF is running, however it will not change the focus, it says at macro

beniroquai, 2021-09-24:
+1 I would also be very interested to see if someone got the OV650-AF working with autofocus turned on!

nilspupils, 2021-09-28:
These drivers from the OpenMV project implement autofocus with the ov5640 camera. Perhaps their code can be used?  

https://github.com/openmv/openmv/tree/f0903dd67f01cae68bbe50f090aba0a90064b52f/src/omv/sensors

nilspupils, 2021-12-15:
This is now availeable. See here: 
https://github.com/0015/ESP32-OV5640-AF

tebb, 2023-12-27:
> This is now availeable. See here: https://github.com/0015/ESP32-OV5640-AF

Just in case this helps someone ...

It's possible that the software was 'working' @raduprv, but the camera focus motor wasn't powered up.  Presumably the camera never finished 'focusing'?

The Voice Coil Motor (often refered to as VM on boards and in docs) that physically implements the camera focus, needs power.  Other cameras, for example the ov2640, don't need this, so it's not easily available on most camera modules.

The required mod to supply the power is shown last in the README.md for 0015's library you mention @nilspupils.

The mod is quite difficult on eg esp32-cam board I have, because the motor power pin is not broken out and connector dimensions small for soldering to.

A solution I'm trying:  

I have ordered some flat cable extender boards and 30 pin flat cables see if that's a practical solution that *does not* require hacking the original board.

This will also test whether the camera works with longer cables to allow more flexible camera positioning.

jameszah, 2023-12-30:
Adafruit claims they have the focus-motor wired up in this s3 ov5640 module ... but only 2mb psram?

https://www.cnx-software.com/2023/12/27/memento-esp32-s3-circuitpython-arduino-programmable-diy-camera-module/

jasaw, 2025-02-12:
I accidentally bumped into this thread. For what it's worth, I have ov5640 autofocus and manual focus working as expected.

Regular ESP32cam boards require 2.8V LDO output to be soldered to AF-VDD and AF-GND obviously to GND.

https://github.com/jomjol/AI-on-the-edge-device/pull/3504