espressif/esp32-camera #232: ov3660 FPS Performance Questions

alorbach, 2021-01-06:
I am not sure if this repo is the right one to ask these questions, but it would be great to get some answers or even solutions. 
So I recently got this module from m5stack with an OV3660 sensor: 
https://docs.m5stack.com/#/en/unit/timercam_x

According to the document here https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/datasheet/unit/OV3660_CSP3_DS_1.3_sida.pdf
, the sensor can handle up to 120 FPS in 320x240 (FRAMESIZE_QVGA):
[image]

However in reality I barely reach 30 FPS in 160x120 (FRAMESIZE_QQVGA) and tripple buffering(fb_count = 3) - direct mjpeg streaming to a http client. With ov2640 sensor modules, I easily reach 35-50 FPS with the same code (And only double buffering fb_count = 2).

I have xclk set to 20 MHZ for both modules, and to my understanding I should be at 50MHz SYSCLK and 10MHz PCLK:
https://github.com/espressif/esp32-camera/blob/0347377b67c2d16996ba6532202f22943018b962/sensors/ov3660.c#L362

Anything we can do to improve processing speed?

me-no-dev, 2021-01-06:
We are more or less limited to what the ESP can output as XCLK and what it can read as PCLK from the camera. While the sensor itself can reach such frame rates, we can not handle them on the ESP :)

alorbach, 2021-01-06:
@me-no-dev I understand, but why do we get almost double framerate with ov2640 sensor?

me-no-dev, 2021-01-06:
ov2640 subsamples from CIF to get 320x240 and ov3660 subsamples from 1280x960. set_pll is sensor specific. You can look at the source and the sensor documentation to understand what it does :) not all parameters exist for every sensor.

alorbach, 2021-01-06:
Yes I saw that in the code ;). 

I took a look to the ov3660 sensor doc , but not fully understand it yet. 

I played with the values (set_pll), and for some reason this gives me about ~ 50% more FPS which is more than I get with my ov2640 modules, so it is pretty fine for me:  
set_pll(sensor, false, 16, 1, 3, false, 0, true, 30);

I hope I am overclocking the something to much ;)?

alorbach, 2021-01-07:
Some more results: 

With xclk_freq_hz = 20000000U, these values work best: 
`set_pll(sensor, false, 15, 1, 3, false, 0, true, 30);`

With xclk_freq_hz = 10000000U, these values work best: 
`set_pll(sensor, false, 30, 1, 3, false, 0, true, 30);`

liamplm, 2023-11-08:
> Yes I saw that in the code ;).
>
> I took a look to the ov3660 sensor doc , but not fully understand it yet.
>
> I played with the values (set_pll), and for some reason this gives me about ~ 50% more FPS [...]

I know it's been quite a long time but what was the max fps you were able to achieve with these setting and in what resolution ?

Hfffliao, 2025-10-23:
我遇到了相同的问题，我想用esp32-cam模块传输视频到手机上，使用ov3660摄像头，以此来控制我的小车，我使用QVGA大小，并希望有尽可能低的延迟，我使用单缓冲区，看cam库的源码和ov3660文档之后也认为要调这个set_pll函数。另外我发现每次设置config.frame_size 函数都会自动调用set_pll，我目前毫无进展，我要试试你的参数，看看是否能加快获取图像的速度