espressif/esp32-camera #233: How to enable Antiflicker Support / 50&60Hz Auto

alorbach, 2021-01-07:
ov2640 and ov3660, both Sensors have  50/60 Hz detection and antiflicker controls. 
However I do not find any methods to control those settings. 

Could somebody tell me how to make sure those options are enabled, or how I can enable them?

raduprv, 2021-01-07:
I've been investigating those things for other purposes for ov3660 and ov5660. Do you need it for video or for stills?
The thing is, if the screen is just part of the picture, and there is natural or artificial light in the room, the sensor gets confused because it doesn't know which light is the correct one, so you have to set it manually. Basically, disable the band detection and set the band to whatever band you want.

alorbach, 2021-01-07:
Well basically I am capturing TV Screen, analyzing the image and send it to a LED Stripe behind the TV. Which works basically very well, however on one TV I have, I sometimes get this flickering - I am not using the internal auto exposure control. Only the auto white balance. 

Do you have any hint how to set the correct registers to set band detection manual, and switch between 50 / 60 hz ?

raduprv, 2021-01-07:
Well, first, what is the refresh rate of the tv? If it is not 50 or 60 hz, then I am not sure if this can help.
Read the ov3660 documentation if you haven't done so already: https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/datasheet/unit/OV3660_CSP3_DS_1.3_sida.pdf
Search in it for "band", there are quite a few registers. Try to modify the obvious bits, like 0x3C01 for example.

alorbach, 2021-01-21:
FYI: I tried setting these registers but I proberly am not setting all necessary ones. 
However it appears that if the aec value is to low, the flickering gets bad - meaning the light sensor does not appear to auto detect the frequency if aec value goes below a specific value. 

I am setting the AEC value manually using "set_aec_value" based on the light value of the TV content ignoring everything else. 
So I highered the minimum aec value and it got a lot better.