espressif/esp32-camera #234: Im getting endless dots on my serial monitor

WildCD, 2021-01-09:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8

..............................................................................

What could be the problem. My Password and ssid are correct and everything else is working.

Naheel-Azawy, 2021-01-23:
Same here. I have no idea why, but for some reason [THIS](https://github.com/easytarget/esp32-cam-webserver) worked in the access point mode. Connecting to an existing AP still doesn't work there though

Naheel-Azawy, 2021-09-03:
I'm replying to keep this issue active. I believe it is a significant issue as it's making my ESPs unusable for what I need them for.

me-no-dev, 2021-09-08:
neither of you have posted an actual issue report :)
I have no idea what code you are running, if you are using Arduino or ESP-IDF, which version, which board, which sensor, what means "it's making my ESPs unusable"?

Please file a new proper report, so that we can help.

closed 2021-09-08 by me-no-dev