espressif/esp32-camera #238: Accessing raw pixel values with frame2bmp: Guru Meditation Error

kmhaeren, 2021-01-21:
Hi

I'm a bit of an esp32 noob, but I did my fair share (2days) of googling before asking this question

board: AI-thinker esp32 cam
Camera: label says "AF2569"

I'm trying to take a picture with size XGA, I read that this is best done with pixel format set to JPEG and then convert with 'frame2bmp'. I also read that the buffer past to the latter function should be freed again. 
The code below is an example where the esp32 prints out the pixel values to serial (this is NOT what I want to do in my final application, but it shows the error I get).

Strangely enough, the error is only thrown after some values from fb->buf are printed already. 

The error I get is:

Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d153e  PS      : 0x00060630  A0      : 0x800d29b6  A1      : 0x3ffb1f00  
A2      : 0x3ffc07a4  A3      : 0x00000001  A4      : 0x00000400  A5      : 0x3ffc3b4c  
A6      : 0x00000300  A7      : 0x4000034d  A8      : 0x000000f2  A9      : 0x3ffb1ee0  
A10     : 0x00000001  A11     : 0x3f40b11b  A12     : 0x0000000a  A13     : 0x00000003  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000003  
EXCVADDR: 0x4000034d  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d153e:0x3ffb1f00 0x400d29b3:0x3ffb1fb0 0x4008cead:0x3ffb1fd0

Could anyone give me some pointers on how to resolve this error? 

minimal working sketch: 

'
#include "esp_camera.h"
#include "Arduino.h"
#include "FS.h"                // SD Card ESP32
#include "SD_MMC.h"            // SD Card ESP32
#include "soc/soc.h"           // Disable brownour problems
#include "soc/rtc_cntl_reg.h"  // Disable brownour problems
#include <EEPROM.h>            // read anpyd write from flash memory

// define the number of bytes you want to access
#define EEPROM_SIZE 1

// Pin definition for CAMERA_MODEL_AI_THINKER
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

void setup() {
  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector 
 
  Serial.begin(115200);

  camera_config_t config;
  
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;
  config.frame_size =  FRAMESIZE_XGA; 
  config.jpeg_quality = 12;
  config.fb_count = 1;
  // Init Camera
  Serial.println((bool)psramFound());

  esp_err_t err = esp_camera_init(&config);
  
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    return;
  }
    
  camera_fb_t * fb = NULL;
  
  fb = esp_camera_fb_get();  
  if(!fb) {
    Serial.println("Camera capture failed");
    return;
  }

  Serial.println("Camera capture success");
    
  uint8_t * buf = NULL;
  size_t buf_len = 0;

  bool converted = frame2bmp(fb, &buf, &buf_len);
  Serial.println(converted);
  free(buf);
  
  uint8_t *img = (uint8_t *)fb->buf;  

  size_t width = fb->width;
  size_t height = fb->height;

  Serial.println("START");

  for(int c = 1; c < width; c+=1){
   for(int r = 1; r < height; r+=1){
      Serial.print(img[c+r*width]);
      Serial.print(" ");
    }
    Serial.println();
  }
  Serial.println("STOP");
}
void loop() {  
}
`

fsck-block, 2021-01-23:
Hi
I think there are multiple issues here (as far as i can see without compiling and flashing your code)
`bool converted = frame2bmp(fb, &buf, &buf_len);
Serial.println(converted);
free(buf);`
buf and buf_len are holdig the bmp data converted from the framebuffer fb.

`uint8_t *img = (uint8_t *)fb->buf;`
fb->buf is the pointer to the camera buffer. So this buffer hold's jpeg data not bmp data

`for(int c = 1; c < width; c+=1){
for(int r = 1; r < height; r+=1){`
I think the for-loop should start with c=0 and r=0.

The crash results from access to fb->buf instead of buf.
The jpeg framebuffer fb->buf does only hold a small amount of width*height bytes based on the jpeg_quality. 
So you would reach the fb->buf end with low values of r, due to the index calculation of [c+r*width]

Best regards

kmhaeren, 2021-01-23:
Thanks so much for the reply, reading the `fp->buf` instead of `buf` did cause the crash!

I also found the structure of the bitmap format in the source code [here](https://github.com/espressif/esp32-camera/blob/a5ccbecf08f98eb84c7443d8aebe4a529ba737a7/conversions/to_bmp.c#L236).

it is still unclear to me though how the pixels are stored. I suppose they are stored as RGB values, where each colour component takes up 1 byte. 
the size of the bmp corresponds to this assumption, but the data I get out of an image is complete garbage.

fsck-block, 2021-01-24:
Yes, if not specified other, it's RGB data. But be sure to skip data of the bitmap header (bmp_header_t).
I'm using the frame2bmp function too. The following code fragment may help

    int indx;
    unsigned char red_val, green_val, blue_val;
    
    for (int row_no=0; row_no<height; row_no++) {
        for (int col_no=0; col_no<width; col_no++) {
            indx = sizeof(bmp_header_t) + (row_no * height + col_no) * 3;
            red_val   = buf[indx];
            green_val = buf[indx+1];
            blue_val  = buf[indx+2];        
            
            // output data to console
        }
    }