espressif/esp32-camera #247: OV5640 does not power off/sleep

Saur0o0n, 2021-02-19:
I have ov5640 AF 72deg camera, and recently used it in place of some OV2640 cams. It battery photo trap project - so I try to keep power consumption on low. But I've noticed that device with OV2640 (esp32 in deep sleep) consumes 0,032W, and with OV5640 somewhere close to 0,5W.
I've tried to use camera_disable_out_clock(); - but without any change in power consumption.

Is this "by design" for OV5640, or there can be something done on software (or hardware) level? Camera is all the time a bit hot - so It's definitely not desire situation (both lost of energy and hot sensor).

Saur0o0n, 2021-02-23:
I've checked and PWRDOWN pin (32 in case of my ESPCam) on camera is properly set to HIGH when ESP32 goes to sleep - but the camera is still drawing energy.

jameszah, 2021-02-28:
I think you would have to shut it off before going to sleep - maybe just shutting off the PWDN gpio - since it would need to be re-init-ed after sleep.

[image]
[image]

jameszah, 2021-02-28:
This seems to work.  My esp32-cam board goes from about 110 mA with the ov5640 camera on, to about 20 mA with camera off, and back to 110 mA with camera on, but otherwise idling.  That business about hardware/sccb standby would be more appropriate to a cpu that can wakeup where is was, rather than the esp32 arduino that has to re-run the setup.

```
  Serial.printf("Camera power is %d\n",digitalRead(PWDN_GPIO_NUM));
  delay(1000);
  
  Serial.printf("Shutting off camera - look at power consumption\n");
  digitalWrite(PWDN_GPIO_NUM, 1);
  
  delay(15000);

  Serial.printf("Camera power is %d\n",digitalRead(PWDN_GPIO_NUM));
  delay(1000);
  
  Serial.printf("Turning on camera - look at power consumption\n");
  digitalWrite(PWDN_GPIO_NUM, 0);
  delay(1000);

  Serial.printf("Camera power is %d\n",digitalRead(PWDN_GPIO_NUM));
  delay(15000);

```

Saur0o0n, 2021-02-28:
I'll retest it, but I've checked previously (on oscilloscope) and the PWRDOWN (#define PWDN_GPIO_NUM     32) pin of the camera WAS high during deep sleep of ESP32 - and still device draw 5W.
I was a bit unsure whether PWRDOWN pin is even connected on AI_THINKER type of board, because I've seen two different schematic for that board and on one it is NC on another is GPIO32. But I've checked, and verified that this GPIO32 is connected to camera ribbon.

Saur0o0n, 2021-03-02:
If I'm reading this right, my ESP32cam board (Chinese AIthinker clone) does not have actually camera PWDOWN control, it's permanently LOW - connected to GND (green box on the diagram).
And instead ESP32 controls power (2,8V and 1,2V) with GPIO32 and transistor - red boxes. But (and this I  guess is the problem here) the 3,3V voltage is connected to camera permanently
Tomorrow I should have new boards (I've fried on my test specimen RX/TX pins - so it's hard to debug it ;) ) and will continue to test.
[image]

Saur0o0n, 2021-03-02:
So I did retrace all the paths and above diagram is consist with my board. It looks like my p-mosfet is faulty (on the picture) - it does not close completely and leaks around 1,2V all the time. This allows 1,2V camera rail to be around 1,2V and 2,8V rail to be 1,2V.
I would swear this was the way since I've started to use this board - so it's seems like it was faulty from factory.
I've checked on other board (the same type - but installed in device) and this mosfet is working properly.
Interesting thing, LDO regulator for 1,2V rail (when powered 3,3V - so normally) produces 1,5V - and this one is different on my two boards.
[image]

Saur0o0n, 2021-03-17:
So it seems I was wrong... again :)
With new boards and some tests on it, it looks like P-MOS was ok all the time. I got wrong conclusions because I've measured one board with 5640 assuming transistor is faulty on this one, and another board board with 2640 - where everything was fine.
So the conclusion is that camera 5640 is somehow backfeeding low voltage rails from always connected 3,3V, when they are disconnected. That's why I was able to measure 1,2xV on mosfet drain.
I'm not sure if it's just mine camera - or every 5640 (I have only one).
Also this happens only after camera initialization - before that, it looks like there is no power issue.
Quick test was to just set HIGH/LOW voltage levels of GPIO32, to enable P-MOS. This allowed to power 2,8 and 1,2V lines, and disable them properly without strange voltages on P-MOS drain (and general power usage of the board was 0,1W).
When doing the same with initialized camera - even when GPIO32 was HIGH, thus P-MOS blocked. There was still voltage on the drain (and board uses 0,57W - esp32 in deep sleep).
So it's probably something that could be solved on software level.
Easier (for me) solution will be to power also 3,3V of the camera with the same line from P-MOS - and I don't know why it hasn't been done by default here.

Saur0o0n, 2021-03-17:
So the "physical" solution was to cut the 3,3V supply line to camera, and power it with output of the P-MOS. Lucky we have quite well exposed 3,3V path and we can solder wire to capacitor (it's bigger then camera socket pins).
[image]
This way during deep sleep of ESP32, whole boards draws 0,045W (8,5mA @ 5,2V) and during taking pictures - 1,15-1,35W.

eried, 2022-05-31:
> So the "physical" solution was to cut the 3,3V supply line to camera, and power it with output of the P-MOS. Lucky we have quite well exposed 3,3V path and we can solder wire to capacitor (it's [...]

But why is the 2.5µA hibernation mode never possible with this boards :/ ? did you ever achieved some long running battery life?

NRollo, 2023-07-05:
> So the "physical" solution was to cut the 3,3V supply line to camera, and power it with output of the P-MOS. Lucky we have quite well exposed 3,3V path and we can solder wire to capacitor (it's [...]

@Saur0o0n
Maybe everyone is gone on this old issue, but I will try anyway.
Did you consider/try placing a wire from the CAM_PWR on the ESP32 to the PWDN on the Cam1 (e.g. from R15 to R3), instead of cutting and wiring? I am "joggling" the same issue but I don't have a component placement for the board in order to try it out.

Saur0o0n, 2023-07-05:
Hi, sorry but it was too long ago and I don't remember much details on it. I finally gave up (not because of this power issue) - but all the tested motion sensors where unreliable (outside environment, snow, wind etc), combine with poor picture quality (low lights) and slow startup... there was no point

NRollo, 2023-07-05:
> Hi, sorry but it was too long ago and I don't remember much details on it. I finally gave up (not because of this power issue) - but all the tested motion sensors where unreliable (outside [...]

@Saur0o0n Thanks for the reply, appreciate it. You don’t happen to have a component placement for this board that you can share?

FBMinis, 2024-01-26:
> @Saur0o0n Thanks for the reply, appreciate it. You don’t happen to have a component placement for this board that you can share?

https://easyeda.com/editor#id=5a6b4b9c6cde4d639ac2b8ab2624d384

aiquie, 2024-03-14:
> So the "physical" solution was to cut the 3,3V supply line to camera, and power it with output of the P-MOS. Lucky we have quite well exposed 3,3V path and we can solder wire to capacitor (it's [...]

This is not suitable for every board, the schematics may differ. I tried it and it de-energized half of the board, I had to restore contact :)

A more reliable option would be the one proposed by @NRollo, control the PWDN output: need to remove resistor R3 and connect PWDN with CAM_PWR through resistor R15.

[image]

NRollo, 2024-04-11:
@FBMinis Thanks for the reply! The link to the PCB design is not the board pictured in this thread, but thanks anyway.