espressif/esp32-camera #251: Capture image timeout when using 24M external crystal oscillator

Kevincoooool, 2021-03-02:
My hardware uses an external 24M active crystal oscillator as the XCLK input source of the OV2640. Both ADF and IDF use the latest commit. Although the OV2640 can be detected, the acquisition frame has been timed out. The log output is as follows:
`ions
/************app_main ESP32_V1.0.1***************/
I (203) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldow
n: 0| Intr:0
I (203) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldow
n: 0| Intr:0
I (210) gpio: GPIO[12]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo
wn: 0| Intr:0
I (219) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo
wn: 0| Intr:0
I (229) gpio: GPIO[16]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo
wn: 0| Intr:0
I (238) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo
wn: 0| Intr:0
I (247) gpio: GPIO[32]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo
wn: 0| Intr:0
I (257) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo
wn: 0| Intr:0
I (266) gpio: GPIO[37]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo
wn: 0| Intr:0
I (275) gpio: GPIO[38]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo
wn: 0| Intr:0
I (284) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo
wn: 0| Intr:0
E (294) camera: Initializing SSCB
I (298) sccb: pin_sda 22 pin_scl 21

E (302) camera: Searching for camera address
E (322) camera: Detected camera at address=0x30
E (322) camera: Resetting OV2640
E (348) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
E (348) camera: Doing SW reset of sensor
I (386) camera: Detected OV2640 camera
E (386) camera: in_bpp: 2, fb_bpp: 2, fb_size: 9600, mode: 3, width: 320 height:
 240
E (388) camera: Line width (for DMA): 2560 bytes
E (393) camera: DMA buffer size: 2560, DMA buffers per line: 1
E (399) camera: DMA buffer count: 4
E (403) camera: DMA buffer total: 10240 bytes
E (409) camera: Allocating DMA buffer #0, size=2560
E (414) camera: dma_buf[0]=0x3f805f7c
E (418) camera: Allocating DMA buffer #1, size=2560
E (424) camera: dma_buf[1]=0x3f806980
E (428) camera: Allocating DMA buffer #2, size=2560
E (434) camera: dma_buf[2]=0x3f807384
E (438) camera: Allocating DMA buffer #3, size=2560
E (444) camera: dma_buf[3]=0x3f807d88
I (448) camera: Allocating 1 frame buffers (9 KB total)
I (455) camera: Allocating 9 KB frame buffer in OnBoard RAM
E (461) camera: Setting frame size to 320x240
nvs_open [0x0]
nvs_get_blob [0x0]
SSID:du
PASSWORD:99999999
I (533) wifi: wifi driver task: 3ffd1ff4, prio:23, stack:3584, core=0
I (533) system_api: Base MAC address is not set, read default base MAC address f
rom BLK0 of EFUSE
I (541) system_api: Base MAC address is not set, read default base MAC address f
rom BLK0 of EFUSE
I (579) wifi: wifi firmware version: 0df1a2a
I (579) wifi: config NVS flash: enabled
I (579) wifi: config nano formating: disabled
I (579) wifi: Init dynamic tx buffer num: 32
I (583) wifi: Init data frame dynamic rx buffer num: 32
I (588) wifi: Init management frame dynamic rx buffer num: 32
I (594) wifi: Init management short buffer num: 32
I (598) wifi: Init static tx buffer num: 16
I (602) wifi: Init static rx buffer size: 1600
I (606) wifi: Init static rx buffer num: 10
I (610) wifi: Init dynamic rx buffer num: 32
I (615) wifi: init cache tx buffer number 32
wifi_scan
I (702) phy: phy_version: 4181, c9a03e2, Mar 10 2020, 12:45:06, 0, 0
I (702) wifi: mode : sta (24:6f:28:8f:b6:9c)
SYSTEM_EVENT_STA_START err
SYSTEM_EVENT_SCAN_DONE
sta_number = 30
ap_list_buffer addr_check: 0x3f80df60
ssid ChinaNet-xesv,bssid 18:52:07:EC:59:D9,level -23
wifi_arry.ssid addr_check: 0x3f80b6b0
wifi_array.bssid addr_check: 0x3f80cd04
ssid 66666666,bssid 98:00:6A:D9:BF:91,level -48
wifi_arry.ssid addr_check: 0x3f80cd1c
wifi_array.bssid addr_check: 0x3f80cd2c
ssid ,bssid 9A:00:6A:B9:BF:91,level -49
wifi_arry.ssid addr_check: 0x3f80cd44
wifi_array.bssid addr_check: 0x3f80cd4c
wifi_connect out
I (2834) BAIDUAI: baidu Start
nvs_open [0x0]
I (2838) BAIDUAI: Taking picture...
[data1]: 192.168.4.1 len:12
[data2]: ssid:KKKK passwd:99999999
app_set_ip:192.168.4.1
E (2844) camera: DMA desc  0: 2560 2560 0 1 1 1 0x3f805f7c 0x3ffb9fd0
W (2851) MYTCP-->: task tcp_conn...
E (2858) camera: DMA desc  1: 2560 2560 0 1 1 1 0x3f806980 0x3ffb9fdc
E (2869) camera: DMA desc  2: 2560 2560 0 1 1 1 0x3f807384 0x3ffb9fe8
E (2876) camera: DMA desc  3: 2560 2560 0 1 1 1 0x3f807d88 0x3ffb9fc4
E (2883) camera: Waiting for negative edge on VSYNC
E (2903) camera: Got VSYNC
E (4880) wifi: ----------------->wifi disconnect
E (6903) camera: Failed to get the frame on time!
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandle
d.
Core 0 register dump:
PC      : 0x40111dba  PS      : 0x00060330  A0      : 0x00000000  A1      : 0x3f
fdf3c0
0x40111dba: baiduai_test_task at E:/msys32/home/Administrator/New_baiduai/main/b
aidu.c:207

A2      : 0x3f40dcb4  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00
000000
A6      : 0x3f80cd7c  A7      : 0x00000000  A8      : 0x80111dba  A9      : 0x3f
fdf3a0
A10     : 0x00001af7  A11     : 0x3ffc521c  A12     : 0x3f805d34  A13     : 0x00
000000
A14     : 0x3f805d34  A15     : 0x3ffc4c20  SAR     : 0x00000004  EXCCAUSE: 0x00
00001c
EXCVADDR: 0x00000004  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xff
fffffe

ELF file SHA256:

Backtrace: 0x40111dba:0x3ffdf3c0 0x7ffffffd:0x3ffdf450
0x40111dba: baiduai_test_task at E:/msys32/home/Administrator/New_baiduai/main/b
aidu.c:207`

me-no-dev, 2021-03-02:
the problem is that the driver is geared towards using the internally generated XCLK. At max XCLK from the ESP (20MHz) OV's internal clocks are set to the max of what the ESP can handle, so having the clock at 24MHz would mean that the camera's output will go beyond the capabilities of the ESP. Will have to think of a way to allow such configuration :)

Kevincoooool, 2021-03-02:
> the problem is that the driver is geared towards using the internally generated XCLK. At max XCLK from the ESP (20MHz) OV's internal clocks are set to the max of what the ESP can handle, so having [...]

Thanks,i'll try to change 24 to 20

Kevincoooool, 2021-03-02:
I just changed 24M to 12M,there was the same error,  Faieled to get the frame on time!

Kevincoooool, 2021-03-02:
> the problem is that the driver is geared towards using the internally generated XCLK. At max XCLK from the ESP (20MHz) OV's internal clocks are set to the max of what the ESP can handle, so having [...]

Hey,I just connected the XCLK pin on the ESP32-CAM board that I purchased to my own hardware, but the problem still persists, and the frame is timed out.

Kevincoooool, 2021-03-03:
> the problem is that the driver is geared towards using the internally generated XCLK. At max XCLK from the ESP (20MHz) OV's internal clocks are set to the max of what the ESP can handle, so having [...]

hello，Is it possible that my IO port configuration is wrong?
`
#define CAM_PIN_PWDN -1
#define CAM_PIN_RESET -1
#define CAM_PIN_XCLK -1
#define CAM_PIN_SIOD 22
#define CAM_PIN_SIOC 21

#define CAM_PIN_D7 36
#define CAM_PIN_D6 37
#define CAM_PIN_D5 38
#define CAM_PIN_D4 39
#define CAM_PIN_D3 32
#define CAM_PIN_D2 12
#define CAM_PIN_D1 13
#define CAM_PIN_D0 2
#define CAM_PIN_VSYNC 19
#define CAM_PIN_HREF 5
#define CAM_PIN_PCLK 16
`

Kevincoooool, 2021-03-04:
ESP-CAM does not work normally under the branch stack_on_psram_v3.3, but it works normally in the release version V3.3

yorickvP, 2021-03-05:
@Kevincoooool Replying to your email here. We're using an external 20MHz oscillator. 10MHz did not work, but we're using an OV5640, which has a different frequency operating range.

Kevincoooool, 2021-03-18:
> @Kevincoooool Replying to your email here. We're using an external 20MHz oscillator. 10MHz did not work, but we're using an OV5640, which has a different frequency operating range.
Thanks!

closed 2021-03-18 by Kevincoooool