espressif/esp32-camera #255: Camera init failed with error 0x103

stodge, 2021-04-05:
I'm trying to use the ESP32-CAM module (AI Thinker) with the Arduino IDE. My loop function simply loops until the pin connected to a RCWL-0516 sensor shows HIGH, at which point it take a picture.

The code is:

```
#include "esp_camera.h"
#include "Arduino.h"
#include "soc/soc.h"           // Disable brownour problems
#include "soc/rtc_cntl_reg.h"  // Disable brownour problems
#include "driver/rtc_io.h"

#define CAMERA_MODEL_AI_THINKER

#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

// 1 = TXD
#define INPUT_PIN 13
#define LED_PIN 33  // RED built in LED

void loop_forever() {
  while (1) {
    delay(500);
    yield();
  }
}

void setup() {
  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector

  Serial.begin(115200);
  Serial.setDebugOutput(true);
  esp_log_level_set("*", ESP_LOG_VERBOSE);
  Serial.println();

  Serial.printf("Total heap: %d\n", ESP.getHeapSize());
  Serial.printf("Free heap: %d\n", ESP.getFreeHeap());
  Serial.printf("Total PSRAM: %d\n", ESP.getPsramSize());
  Serial.printf("Free PSRAM: %d\n", ESP.getFreePsram());

  // Setup motion sensor pin.
  pinMode(INPUT_PIN, INPUT);

  // Setup internal RED LED.
  pinMode(LED_PIN, INPUT);
  // Turn off internal LED.
  digitalWrite(LED_PIN, HIGH);
}

void take_picture() {

  camera_config_t config;

  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;

  if (psramFound()) {
    Serial.println("Using PSRAM");
    config.frame_size = FRAMESIZE_UXGA; // FRAMESIZE_ + QVGA|CIF|VGA|SVGA|XGA|SXGA|UXGA
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    Serial.println("NO PSRAM");
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }

  Serial.println("Initialising camera");

  // Init Camera
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    loop_forever();
  }

  Serial.println("Taking picture");

  camera_fb_t * fb = NULL;

  // Take Picture with Camera
  fb = esp_camera_fb_get();
  if (!fb) {
    loop_forever();
  }

  Serial.println("Freeing framebuffer");

  esp_camera_fb_return(fb);

  // Turns off the ESP32-CAM white on-board LED (flash) connected to GPIO 4
  pinMode(4, OUTPUT);
  digitalWrite(4, LOW);
}

void loop() {

  // Read motion sensor.
  if (digitalRead(INPUT_PIN) == HIGH) {
    // Turn on internal LED.
    digitalWrite(LED_PIN, LOW);
    take_picture();
    // Turn off internal LED.
    digitalWrite(LED_PIN, HIGH);
  } else {
    delay(250);
  }
}
```

The output in the serial monitor is:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

Total heap: 366640
Free heap: 340740
Total PSRAM: 4194252
Free PSRAM: 4194252
Using PSRAM
Initialising camera
Taking picture
Freeing framebuffer
Using PSRAM
Initialising camera
[E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x103
Camera init failed with error 0x103
```

What does the error 0x103 mean? Any help appreciated.
Thanks

me-no-dev, 2021-04-05:
why do you try to init for each frame? the error is `ESP_ERR_INVALID_STATE` because camera is already initialized.
The part of your code that you need to loop is:
```cpp
Serial.println("Taking picture");

  camera_fb_t * fb = NULL;

  // Take Picture with Camera
  fb = esp_camera_fb_get();
  if (!fb) {
    loop_forever();
  }

  Serial.println("Freeing framebuffer");

  esp_camera_fb_return(fb);
```

closed 2021-04-05 by me-no-dev

stodge, 2021-04-05:
Damn, of course; how stupid of me. Thanks.