espressif/esp32-camera #256: Exception while initialising camera when using interrupts

stodge, 2021-04-05:
I'm getting an exception in the esp_camera_init function. I'm using an interrupt from a pin connected to an RCWL-0516 to signal when motion is detected. However, this happens before I setup the interrupt. If I comment out the two marked lines that set the pin for input and install the interrupt handler, then the camera will initialise. I'm sure there's a gap in my knowledge here somewhere?

I'm using the Arduino IDE so who knows what additional code is included.

```
#include "esp_camera.h"
#include "Arduino.h"
#include "soc/soc.h"           // Disable brownour problems
#include "soc/rtc_cntl_reg.h"  // Disable brownour problems
#include "driver/rtc_io.h"

#define CAMERA_MODEL_AI_THINKER

#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

// 1 = TXD
#define INPUT_PIN 13
#define LED_PIN 33  // RED built in LED
camera_config_t config;
volatile bool is_triggered = false;

void IRAM_ATTR ISR_HANDLER() {
  is_triggered = true;
}

void loop_forever() {
  while (1) {
    delay(500);
    yield();
  }
}

void setup() {
  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector

  Serial.begin(115200);
  Serial.setDebugOutput(true);
  esp_log_level_set("*", ESP_LOG_VERBOSE);
  Serial.println();

  Serial.printf("Total heap: %d\n", ESP.getHeapSize());
  Serial.printf("Free heap: %d\n", ESP.getFreeHeap());
  Serial.printf("Total PSRAM: %d\n", ESP.getPsramSize());
  Serial.printf("Free PSRAM: %d\n", ESP.getFreePsram());

  // Setup internal RED LED.
  pinMode(LED_PIN, INPUT);
  // Turn off internal LED.
  digitalWrite(LED_PIN, HIGH);

  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = Y2_GPIO_NUM;
  config.pin_d1 = Y3_GPIO_NUM;
  config.pin_d2 = Y4_GPIO_NUM;
  config.pin_d3 = Y5_GPIO_NUM;
  config.pin_d4 = Y6_GPIO_NUM;
  config.pin_d5 = Y7_GPIO_NUM;
  config.pin_d6 = Y8_GPIO_NUM;
  config.pin_d7 = Y9_GPIO_NUM;
  config.pin_xclk = XCLK_GPIO_NUM;
  config.pin_pclk = PCLK_GPIO_NUM;
  config.pin_vsync = VSYNC_GPIO_NUM;
  config.pin_href = HREF_GPIO_NUM;
  config.pin_sscb_sda = SIOD_GPIO_NUM;
  config.pin_sscb_scl = SIOC_GPIO_NUM;
  config.pin_pwdn = PWDN_GPIO_NUM;
  config.pin_reset = RESET_GPIO_NUM;
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;

  if (psramFound()) {
    Serial.println("Using PSRAM");
    config.frame_size = FRAMESIZE_UXGA; // FRAMESIZE_ + QVGA|CIF|VGA|SVGA|XGA|SXGA|UXGA
    config.jpeg_quality = 10;
    config.fb_count = 2;
  } else {
    Serial.println("NO PSRAM");
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12;
    config.fb_count = 1;
  }

  Serial.println("Initialising camera");

  // Init Camera
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK) {
    Serial.printf("Camera init failed with error 0x%x", err);
    loop_forever();
  }

  // Setup motion sensor pin.
  // **** REMOVING THESE TWO LINES - THE CAMERA WILL INITIALISE
  // **** USING THESE TWO LINES - AN EXCEPTION OCCURS WHILE INITIALISING THE CAMERA
  pinMode(INPUT_PIN, INPUT);
  attachInterrupt(digitalPinToInterrupt(INPUT_PIN), ISR_HANDLER, RISING);
}

void take_picture() {
  Serial.println("Taking picture");

  camera_fb_t * fb = NULL;

  // Take Picture with Camera
  fb = esp_camera_fb_get();
  if (!fb) {
    loop_forever();
  }

  Serial.println("Freeing framebuffer");

  esp_camera_fb_return(fb);

  // Turns off the ESP32-CAM white on-board LED (flash) connected to GPIO 4
  pinMode(4, OUTPUT);
  digitalWrite(4, LOW);
}

void loop() {
  //if (is_triggered) {
    // Turn on internal LED.
    //digitalWrite(LED_PIN, LOW);
    //take_picture();
    // Turn off internal LED.
    //digitalWrite(LED_PIN, HIGH);
    //is_triggered = false;
  //}
}
```

The exception is probably a null pointer, but I don't know why:

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

Total heap: 366528
Free heap: 340664
Total PSRAM: 0
Free PSRAM: 0
NO PSRAM
Initialising camera
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400f2e7b  PS      : 0x00060830  A0      : 0x800d1594  A1      : 0x3ffb1f10  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x4008116c  A5      : 0x00000000  
A6      : 0x3ffc07d8  A7      : 0x00000000  A8      : 0x800d54a8  A9      : 0x3ffb1ec0  
A10     : 0x00000105  A11     : 0x0000040e  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x4008116c  A15     : 0x00000000  SAR     : 0x00000013  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400f2e7b:0x3ffb1f10 0x400d1591:0x3ffb1f30 0x400d15f9:0x3ffb1f60 0x400d150a:0x3ffb1f80 0x400d1fab:0x3ffb1fb0 0x4008ce1d:0x3ffb1fd0
```

Thanks

me-no-dev, 2021-04-05:
currently ESP-Camera and `attachInterrupt` do not work together. They will work in a near future

closed 2021-04-05 by me-no-dev

stodge, 2021-04-05:
Phew, I thought I was going crazy. Is there an alternative, or should I check the sensor in a loop instead of using interrupts? Thanks again.

me-no-dev, 2021-04-05:
you can use IDF's API to set pin interrupt
```cpp
void IRAM_ATTR ISR_HANDLER(void* arg) {
  is_triggered = true;
}
gpio_set_intr_type(INPUT_PIN, GPIO_INTR_POSEDGE);
gpio_isr_handler_add(INPUT_PIN, ISR_HANDLER, NULL);
```

stodge, 2021-04-05:
Of course, thanks so much!

ShubhamBhatt, 2022-02-20:
I think I am also having the same problem. I am trying to use mpu6050 with the camera. mpu6050 needs to interrupt. As soon as I am trying to open the camera 

`camera.begin(FRAMESIZE_QVGA, PIXFORMAT_JPEG);`

 I get this error.

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
192.168.1.128
Initializing I2C devices...
Testing device connections...
MPU6050 connection successful

Send any character to begin DMP programming and demo: 
Initializing DMP...
>**......>......

//           X Accel  Y Accel  Z Accel   X Gyro   Y Gyro   Z Gyro
//OFFSETS    -1762,   -1532,    3088,      52,      21,     -16
Enabling DMP...
Enabling interrupt detection (Arduino external interrupt 2)...
DMP ready! Waiting for first interrupt...
[E][camera.c:1327] camera_init(): gpio_install_isr_service failed (105)
[E][camera.c:1406] esp_camera_init(): Camera init failed with error 0x105
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d1318  PS      : 0x00060130  A0      : 0x800d5321  A1      : 0x3ffb1f20  
A2      : 0x00000005  A3      : 0x00000001  A4      : 0x3ffc2948  A5      : 0x4008132c  
A6      : 0x00000003  A7      : 0x00060a23  A8      : 0x800d1316  A9      : 0x3ffb1f00  
A10     : 0x00000000  A11     : 0x00000005  A12     : 0x00000001  A13     : 0x3f4002b7  
A14     : 0x3ffc78c0  A15     : 0x3f400fd0  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000040  LBEG    : 0x4008dd01  LEND    : 0x4008dd11  LCOUNT  : 0xfffffffb  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d1318:0x3ffb1f20 0x400d531e:0x3ffb1fb0 0x40090312:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
`

I am new to interrupts. I have tried many things no luck please help.
Here is my full code
`

// I2Cdev and MPU6050 must be installed as libraries, or else the .cpp/.h files
// for both classes must be in the include path of your project
#define CAMERA_MODEL_AI_THINKER
#include <WiFi.h>
[... 241 lines cut ...]
  WiFi.disconnect(true); // delete old config 
  WiFi.onEvent(WiFiEvent);//register event handler
  WiFi.begin(ssid, pwd);//Initiate connection
 // Serial.println("Waiting for WIFI connection...");
}
//wifi event handler
void WiFiEvent(WiFiEvent_t event)
{
    switch(event) 
    {
      case SYSTEM_EVENT_STA_GOT_IP:           
         // Serial.print("WiFi connected! IP address: ");//When connected set
          Serial.println(WiFi.localIP());      
          //initializes the UDP state
          //This initializes the transfer buffer
          udp.begin(WiFi.localIP(),udpPort);
          connected = true;
          break;         
      case SYSTEM_EVENT_STA_DISCONNECTED:
         // Serial.println("WiFi lost connection");
          connected = false;      
          break;
      default: break;
    }
}`

hugokernel, 2022-07-26:
> currently ESP-Camera and attachInterrupt do not work together. They will work in a near future

@me-no-dev Any news about that ?

Thx