espressif/esp32-camera #262: esp32s3 can't build and run camera

YanMinge, 2021-05-10:
[Development environment]
OS: ubuntu18.04
cmake: 3.16.9
esp-idf: release/v4.3   commit-id: e9cf9e2978d0187b0f83a9978391217e39946517
esp32-camera: commit-id: 579fe966aedd0de212ff2e0c8d47ceeea1a566d7

[test project]
[camera_test.zip](https://github.com/espressif/esp32-camera/files/6451481/camera_test.zip)

[issue Description]
1. Compile Error, since the esp-idf\components\soc\esp32s3\include\soc\gdma_struct.h is not same as esp32, The structure gdma_dev_t has changed,  so GDMA calls like the following will compile errors.
```
    GDMA.in[lcd_cam_obj->dma_num].conf0.val = 0;
    GDMA.in[lcd_cam_obj->dma_num].conf1.val = 0;
    GDMA.in[lcd_cam_obj->dma_num].int_clr.val = ~0;
    GDMA.in[lcd_cam_obj->dma_num].int_ena.val = 0;

    GDMA.in[lcd_cam_obj->dma_num].conf0.in_rst = 1;
    GDMA.in[lcd_cam_obj->dma_num].conf0.in_rst = 0;
    GDMA.in[lcd_cam_obj->dma_num].conf0.indscr_burst_en = 1;
    GDMA.in[lcd_cam_obj->dma_num].conf0.in_data_burst_en = 1;
    GDMA.in[lcd_cam_obj->dma_num].peri_sel.sel = 5;
    GDMA.in[lcd_cam_obj->dma_num].pri.rx_pri = 1;
```
2. I modified the structure to make the compilation pass, but when I flash the code to my esp32-S3-Addax-1_v1.0. it will crash the log is:
[image]

and from Backtrace, it crash here:
`xtensa-esp32-elf-addr2line -pfiaC -e build/camera_test.elf  Backtrace:0x420069db:0x3fcf3b10 0x42007298:0x3fcf3b50 0x4200617b:0x3fcf3b80 0x42005e4a:0x3fcf3bb0 0x42005e6f:0x3fcf3bd0 0x4201cea3:0x3fcf3bf0 0x4037cdc5:0x3fcf3c10
0x00000bac: ?? ??:0
0x42007298: cam_init at /home/myan/develop/camera_test/build/../components/esp32-camera/target/esp32s3/cam_hal.c:581
0x4200617b: esp_camera_init at /home/myan/develop/camera_test/build/../components/esp32-camera/driver/esp_camera.c:231
0x42005e4a: init_camera at /home/myan/develop/camera_test/build/../main/camera_test.c:155
0x42005e6f: app_main at /home/myan/develop/camera_test/build/../main/camera_test.c:167
0x4201cea3: main_task at /home/myan/develop/esp32_mpy/submodule/esp-idf/components/freertos/port/port_common.c:133 (discriminator 2)
0x4037cdc5: vPortTaskWrapper at /home/myan/develop/esp32_mpy/submodule/esp-idf/components/freertos/port/xtensa/port.c:168`

me-no-dev, 2021-05-10:
ok. after you have edited the code for GDMA, what is located in cam_init() at /home/myan/develop/camera_test/build/../components/esp32-camera/target/esp32s3/cam_hal.c:581

YanMinge, 2021-05-11:
I set one IO error, so you can close this issue