espressif/esp32-camera #271: Deinit the camera

maldivien, 2021-05-28:
Hi, is there any function deinit the camera ? I need to use pin 25 after I take picture which is used by camera. should write low to all pins?

me-no-dev, 2021-05-28:
something like this? https://github.com/espressif/esp32-camera/blob/master/driver/include/esp_camera.h#L163

maldivien, 2021-05-28:
yes, Thanks deinit works but reinit gives out the following error.

E (53863) i2c: i2c driver install error
[E][sccb.c:115] SCCB_Write(): SCCB_Write Failed addr:0x30, reg:0xff, data:0x01, ret:-1
[E][sccb.c:115] SCCB_Write(): SCCB_Write Failed addr:0x30, reg:0x12, data:0x80, ret:-1
[E][camera.c:1113] camera_probe(): Detected camera not supported.
[E][camera.c:1379] esp_camera_init(): Camera probe failed with error 0x20004

maldivien, 2021-05-28:
so I solved by writing LOW and HIGH on PWDN pin before init. it works but with the following msgs appear.

E (72532) i2c: i2c driver install error
E (73908) gpio: gpio_install_isr_service(412): GPIO isr service already installed

me-no-dev, 2021-05-28:
just to confirm, you are running the very latest code from this repo?

maldivien, 2021-05-28:
> just to confirm, you are running the very latest code from this repo?

Yes.

maldivien, 2021-05-28:
I'm using AI thinker Module with ov2640, regardless of these msgs camera initialize and works perfectly.

me-no-dev, 2021-05-28:
second message is expected. First one is not :)

maldivien, 2021-06-03:
> second message is expected. First one is not :)

Yes, found the issue its thrown by another library that im using. thanks.

closed 2021-06-03 by maldivien