espressif/esp32-camera #274: Streaming : image taken every 2000mx

loicguidoux, 2021-06-02:
With the "7a06a7e" release, with streaming application as described in your "readme", image are taken every 2 seconds:
I (104237) server: MJPG: 26KB 2157ms (0.5fps)
I (106377) server: MJPG: 26KB 2137ms (0.5fps)
I (108497) server: MJPG: 26KB 2124ms (0.5fps)
I (110537) server: MJPG: 26KB 2041ms (0.5fps)
I (112657) server: MJPG: 26KB 2118ms (0.5fps)

When I use capture the image is taken in 50ms.
I (170377) server: Picture taken! width: 800 height: 600 size: 27357 bytes
I (170537) server: image width:800 height:600
I (170537) server: JPG: 26KB 161ms

Hereafter my camera config ( I designed my own camera HW with ESP32-S2): 

#define CAM_PIN_PWDN -1  //power down is not used
#define CAM_PIN_RESET -1 //software reset will be performed
#define CAM_PIN_XCLK GPIO_NUM_12
#define CAM_PIN_PCLK GPIO_NUM_13
#define CAM_PIN_VSYNC GPIO_NUM_10
#define CAM_PIN_HREF GPIO_NUM_11

#define CAM_PIN_D0 GPIO_NUM_17
#define CAM_PIN_D1 GPIO_NUM_15
#define CAM_PIN_D2 GPIO_NUM_14
#define CAM_PIN_D3 GPIO_NUM_16
#define CAM_PIN_D4 GPIO_NUM_18
#define CAM_PIN_D5 GPIO_NUM_19
#define CAM_PIN_D6 GPIO_NUM_20
#define CAM_PIN_D7 GPIO_NUM_21

#define CAM_PIN_SIOC GPIO_NUM_9
#define CAM_PIN_SIOD GPIO_NUM_8

static camera_config_t camera_config = {
    .pin_pwdn = CAM_PIN_PWDN,
    .pin_reset = CAM_PIN_RESET,
    .pin_xclk = CAM_PIN_XCLK,
    .pin_sscb_sda = CAM_PIN_SIOD,
    .pin_sscb_scl = CAM_PIN_SIOC,

    .pin_d7 = CAM_PIN_D7,
    .pin_d6 = CAM_PIN_D6,
    .pin_d5 = CAM_PIN_D5,
    .pin_d4 = CAM_PIN_D4,
    .pin_d3 = CAM_PIN_D3,
    .pin_d2 = CAM_PIN_D2,
    .pin_d1 = CAM_PIN_D1,
    .pin_d0 = CAM_PIN_D0,
    .pin_vsync = CAM_PIN_VSYNC,
    .pin_href = CAM_PIN_HREF,
    .pin_pclk = CAM_PIN_PCLK,

    //XCLK 20MHz or 10MHz for OV2640 double FPS (Experimental)
    .xclk_freq_hz = 20000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,

    .pixel_format = PIXFORMAT_JPEG, //YUV422,GRAYSCALE,RGB565,JPEG
    .frame_size = FRAMESIZE_SVGA,   //QQVGA-UXGA Do not use sizes above QVGA when not JPEG

    .jpeg_quality = 10, //0-63 lower number means higher quality
    .fb_count = 2      //if more than one, i2s runs in continuous mode. Use only with JPEG
};
It seems there is a timer that allows picture taking (activation of "cam-task")
thanks for help.

me-no-dev, 2021-06-02:
your results do not look ok. which sensor is that?

loicguidoux, 2021-06-02:
the sensor is OV2640
When I write a program taking pictures in a loop, the problem disappears, successive pictures are taken in 5ms.
Il looks like there's a problem in the streaming program (http response??)

loicguidoux, 2021-06-02:
I continue to investigate

me-no-dev, 2021-06-02:
no issues on my end. I use the web server example from ESP-WHO

loicguidoux, 2021-06-02:
OK I'll try

loicguidoux, 2021-06-02:
I found the problem. I modified the streaming example you gave, by introducing a timer of 2 seconds !!!

Everything is working fine. The streaming rate is around 10frames/s with SVGA and JPEG.
Sorry for disturbing.

Thanks for your useful work to make camera working with different versions of ESP32, especially for me the ESP32-S2.

closed 2021-06-02 by loicguidoux