espressif/esp32-camera #282: Does esp32 support driving GC series chips

seannaaa, 2021-06-18:
Hello, I'd like to know if esp32 supports driving gc2145. We can see gc2145. C from the old version of ESP who project, but with the iteration of the version, GC series chips no longer appear in esp32 who project

TDA-2030, 2021-06-24:
I'm doing this work, but it's not finished yet. You can experience it on branch [feature/add_gcxxxx_sensors](https://github.com/espressif/esp32-camera/tree/feature/add_gcxxxx_sensors) first

seannaaa, 2021-06-26:
> I'm doing this work, but it's not finished yet. You can experience it on branch [feature/add_gcxxxx_sensors](https://github.com/espressif/esp32-camera/tree/feature/add_gcxxxx_sensors) first

I understand. Thank you.Because I have a project recently, customers want to use gc1054 / gc2053 cameras to evaluate the feasibility of direct driving them with esp32.I hope your work can be finished smoothly.

TDA-2030, 2021-06-26:
Oh, gc1054 / gc2053 is not in my plan. I'm afraid I can't help you much :anguished:

seannaaa, 2021-06-28:
> Oh, gc1054 / gc2053 is not in my plan. I'm afraid I can't help you much 😧

It doesn't matter. I'll evaluate the plan again.Thank you for your answer~