espressif/esp32-camera #286: I can't compile my project if I add the ESP32_camera library to the ESP-IDF components folder

AndrewLivesChaos, 2021-06-24:
I copied the _esp32_camera_ folder in my _C:\Users\Andrea\esp\esp-idf\components_ folder (esp-idf v4.2.1). I've created a new project coping the _hello_world_ template (I renamed it as _camera_) but now wen I compile I get the following error:

> D:\Documenti\VSCode\ESP32\camera>idf.py build
> Executing action: all (aliases: build)
> Running ninja in directory d:\documenti\vscode\esp32\camera\build
> Executing "ninja all"...
> [0/1] Re-running CMake...
> -- ccache will be used for faster recompilation
> -- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
> -- Building ESP-IDF components for target esp32
> -- Project sdkconfig file D:/Documenti/VSCode/ESP32/camera/sdkconfig
> -- Could NOT find Perl (missing: PERL_EXECUTABLE)
> -- App "camera" version: 1
> -- Adding linker script C:/Users/Andrea/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
> -- Adding linker script C:/Users/Andrea/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
> -- Adding linker script C:/Users/Andrea/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
> -- Adding linker script C:/Users/Andrea/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
> -- Adding linker script D:/Documenti/VSCode/ESP32/camera/build/esp-idf/esp32/esp32_out.ld
> -- Adding linker script C:/Users/Andrea/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
> -- Adding linker script C:/Users/Andrea/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
> -- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp32_camera esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_ipc esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
> -- Component paths: C:/Users/Andrea/esp/esp-idf/components/app_trace C:/Users/Andrea/esp/esp-idf/components/app_update C:/Users/Andrea/esp/esp-idf/components/asio C:/Users/Andrea/esp/esp-idf/components/bootloader C:/Users/Andrea/esp/esp-idf/components/bootloader_support C:/Users/Andrea/esp/esp-idf/components/bt C:/Users/Andrea/esp/esp-idf/components/cbor C:/Users/Andrea/esp/esp-idf/components/coap C:/Users/Andrea/esp/esp-idf/components/console C:/Users/Andrea/esp/esp-idf/components/cxx C:/Users/Andrea/esp/esp-idf/components/driver C:/Users/Andrea/esp/esp-idf/components/efuse C:/Users/Andrea/esp/esp-idf/components/esp-tls C:/Users/Andrea/esp/esp-idf/components/esp32 C:/Users/Andrea/esp/esp-idf/components/esp32_camera C:/Users/Andrea/esp/esp-idf/components/esp_adc_cal C:/Users/Andrea/esp/esp-idf/components/esp_common C:/Users/Andrea/esp/esp-idf/components/esp_eth C:/Users/Andrea/esp/esp-idf/components/esp_event C:/Users/Andrea/esp/esp-idf/components/esp_gdbstub C:/Users/Andrea/esp/esp-idf/components/esp_hid C:/Users/Andrea/esp/esp-idf/components/esp_http_client C:/Users/Andrea/esp/esp-idf/components/esp_http_server C:/Users/Andrea/esp/esp-idf/components/esp_https_ota C:/Users/Andrea/esp/esp-idf/components/esp_https_server C:/Users/Andrea/esp/esp-idf/components/esp_ipc C:/Users/Andrea/esp/esp-idf/components/esp_local_ctrl C:/Users/Andrea/esp/esp-idf/components/esp_netif C:/Users/Andrea/esp/esp-idf/components/esp_ringbuf C:/Users/Andrea/esp/esp-idf/components/esp_rom C:/Users/Andrea/esp/esp-idf/components/esp_serial_slave_link C:/Users/Andrea/esp/esp-idf/components/esp_system C:/Users/Andrea/esp/esp-idf/components/esp_timer C:/Users/Andrea/esp/esp-idf/components/esp_websocket_client C:/Users/Andrea/esp/esp-idf/components/esp_wifi C:/Users/Andrea/esp/esp-idf/components/espcoredump C:/Users/Andrea/esp/esp-idf/components/esptool_py C:/Users/Andrea/esp/esp-idf/components/expat C:/Users/Andrea/esp/esp-idf/components/fatfs C:/Users/Andrea/esp/esp-idf/components/freemodbus C:/Users/Andrea/esp/esp-idf/components/freertos C:/Users/Andrea/esp/esp-idf/components/heap C:/Users/Andrea/esp/esp-idf/components/idf_test C:/Users/Andrea/esp/esp-idf/components/jsmn C:/Users/Andrea/esp/esp-idf/components/json C:/Users/Andrea/esp/esp-idf/components/libsodium C:/Users/Andrea/esp/esp-idf/components/log C:/Users/Andrea/esp/esp-idf/components/lwip D:/Documenti/VSCode/ESP32/camera/main C:/Users/Andrea/esp/esp-idf/components/mbedtls C:/Users/Andrea/esp/esp-idf/components/mdns C:/Users/Andrea/esp/esp-idf/components/mqtt C:/Users/Andrea/esp/esp-idf/components/newlib C:/Users/Andrea/esp/esp-idf/components/nghttp C:/Users/Andrea/esp/esp-idf/components/nvs_flash C:/Users/Andrea/esp/esp-idf/components/openssl C:/Users/Andrea/esp/esp-idf/components/partition_table C:/Users/Andrea/esp/esp-idf/components/perfmon C:/Users/Andrea/esp/esp-idf/components/protobuf-c C:/Users/Andrea/esp/esp-idf/components/protocomm C:/Users/Andrea/esp/esp-idf/components/pthread C:/Users/Andrea/esp/esp-idf/components/sdmmc C:/Users/Andrea/esp/esp-idf/components/soc C:/Users/Andrea/esp/esp-idf/components/spi_flash C:/Users/Andrea/esp/esp-idf/components/spiffs C:/Users/Andrea/esp/esp-idf/components/tcp_transport C:/Users/Andrea/esp/esp-idf/components/tcpip_adapter C:/Users/Andrea/esp/esp-idf/components/tinyusb C:/Users/Andrea/esp/esp-idf/components/ulp C:/Users/Andrea/esp/esp-idf/components/unity C:/Users/Andrea/esp/esp-idf/components/vfs C:/Users/Andrea/esp/esp-idf/components/wear_levelling C:/Users/Andrea/esp/esp-idf/components/wifi_provisioning C:/Users/Andrea/esp/esp-idf/components/wpa_supplicant C:/Users/Andrea/esp/esp-idf/components/xtensa
> -- Configuring done
> -- Generating done
> -- Build files have been written to: D:/Documenti/VSCode/ESP32/camera/build
> [224/943] Performing build step for 'bootloader'
> [0/1] Re-running CMake...
> -- Building ESP-IDF components for target esp32
> -- Project sdkconfig file D:/Documenti/VSCode/ESP32/camera/sdkconfig
> -- Adding linker script C:/Users/Andrea/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
> -- Adding linker script C:/Users/Andrea/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
> -- Adding linker script C:/Users/Andrea/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
> -- Adding linker script C:/Users/Andrea/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
> -- Adding linker script C:/Users/Andrea/esp/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.ld
> -- Adding linker script C:/Users/Andrea/esp/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.rom.ld
> -- Components: bootloader bootloader_support efuse esp32 esp_common esp_rom esptool_py log main micro-ecc partition_table soc spi_flash xtensa
> -- Component paths: C:/Users/Andrea/esp/esp-idf/components/bootloader C:/Users/Andrea/esp/esp-idf/components/bootloader_support C:/Users/Andrea/esp/esp-idf/components/efuse C:/Users/Andrea/esp/esp-idf/components/esp32 C:/Users/Andrea/esp/esp-idf/components/esp_common C:/Users/Andrea/esp/esp-idf/components/esp_rom C:/Users/Andrea/esp/esp-idf/components/esptool_py C:/Users/Andrea/esp/esp-idf/components/log C:/Users/Andrea/esp/esp-idf/components/bootloader/subproject/main C:/Users/Andrea/esp/esp-idf/components/bootloader/subproject/components/micro-ecc C:/Users/Andrea/esp/esp-idf/components/partition_table C:/Users/Andrea/esp/esp-idf/components/soc C:/Users/Andrea/esp/esp-idf/components/spi_flash C:/Users/Andrea/esp/esp-idf/components/xtensa
> -- Configuring done
> -- Generating done
> -- Build files have been written to: D:/Documenti/VSCode/ESP32/camera/build/bootloader
> [1/100] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/src/memory_layout_utils.c.obj
> [2/100] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/src/cpu_util.c.obj
> [3/100] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/src/lldesc.c.obj
> [4/100] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/src/hal/rtc_io_hal.c.obj
> [5/100] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/src/hal/rmt_hal.c.obj
> [6/100] Building C object esp-idf/soc/CMakeFiles/__idf_soc.dir/src/hal/cpu_hal.c.obj
[... cut ...]
> C:/Users/Andrea/esp/esp-idf/components/esp32_camera/target/esp32/ll_cam.c:199:5: error: implicit declaration of function 'esp_rom_delay_us'; did you mean 'ets_delay_us'? [-Werror=implicit-function-declaration]
>      esp_rom_delay_us(1);
>      ^~~~~~~~~~~~~~~~
>      ets_delay_us
> cc1.exe: some warnings being treated as errors
> [686/941] Building C object esp-idf/expat/CMakeFiles/__idf_expat.dir/expat/expat/lib/xmlparse.c.obj
> ninja: build stopped: subcommand failed.
> ninja failed with exit code 1
> 
> D:\Documenti\VSCode\ESP32\camera>

TDA-2030, 2021-06-24:
This problem has been fixed. Maybe you won't encounter this problem with the latest code

Portstart, 2021-07-02:
I have the same problem. Have you solved it

rokahr, 2021-07-03:
I'm facing a similar issue. Any idea on how to solve it?

> ../main/main.c: In function 'camera_init':
> ../main/main.c:57:9: error: implicit declaration of function 'pinMode' [-Werror=implicit-function-declaration]
>          pinMode(CAM_PIN_PWDN, OUTPUT);
>          ^~~~~~~
> ../main/main.c:57:31: error: 'OUTPUT' undeclared (first use in this function)
>          pinMode(CAM_PIN_PWDN, OUTPUT);
>                                ^~~~~~
> ../main/main.c:57:31: note: each undeclared identifier is reported only once for each function it appears in
> ../main/main.c:58:9: error: implicit declaration of function 'digitalWrite' [-Werror=implicit-function-declaration]
>          digitalWrite(CAM_PIN_PWDN, LOW);
>          ^~~~~~~~~~~~
> ../main/main.c:58:36: error: 'LOW' undeclared (first use in this function)
>          digitalWrite(CAM_PIN_PWDN, LOW);
>                                     ^~~
> ../main/main.c:64:9: error: implicit declaration of function 'ESP_LOGE'; did you mean 'ESP_OK'? [-Werror=implicit-function-declaration]
>          ESP_LOGE(TAG, "Camera Init Failed");
>          ^~~~~~~~
>          ESP_OK
> ../main/main.c:64:18: error: 'TAG' undeclared (first use in this function)
>          ESP_LOGE(TAG, "Camera Init Failed");
>                   ^~~
> ../main/main.c: In function 'camera_capture':
> ../main/main.c:75:18: error: 'TAG' undeclared (first use in this function)
>          ESP_LOGE(TAG, "Camera Capture Failed");
>                   ^~~
> ../main/main.c:79:5: error: implicit declaration of function 'process_image' [-Werror=implicit-function-declaration]
>      process_image(fb->width, fb->height, fb->format, fb->buf, fb->len);

MarkusvonStaden, 2021-11-18:
> I'm facing a similar issue. Any idea on how to solve it?

I'm facing the same issue. Did you find a solution?

AndrewLivesChaos, 2021-12-03:
I solved the issue changing the library version, I don't remember which one.