espressif/esp32-camera #289: Getting a linker error when building the drivers

nalostta, 2021-06-28:
[esp_PIO_build_error.txt](https://github.com/espressif/esp32-camera/files/6728589/esp_PIO_build_error.txt)

I followed the steps to link the repository as a submodule and to test the code, copied and built the example that came with the repo. I am facing linking errors with undefined references to functions like cam_start, cam_task, ll_cam_stop, ll_cam_memcpy, ll_cam_set_pin etc...

I have attached the the complete log file with this issue. 
kindly help asap.

me-no-dev, 2021-06-28:
It looks like PIO did not compile all necessary files. Why that is I have no idea. I suggest you ask there :)

nalostta, 2021-06-29:
So I was able to build the project using the esp-idf framework in PIO before I added the esp32-cam in the lib section. If you look closely, the symbol errors come from HALs for the cam driver in the esp32cam module itself. I am not sure whether this is a config error or a bug. Could you share the method to build and the run the example code of the repository?

me-no-dev, 2021-06-29:
The problem is something connected to pio and cmake. On our side the configuration is correct to use with ESP-IDF (I build it quite often), but seems that PIO somehow did not pass CMakeLists.txt in the driver and did not build the relevant hal for the chip you are building for. If you do not use PIO to build your project, but instead go the direct ESP-IDF way, it should work. Preferably you should use the latest IDF master

nalostta, 2021-06-29:
Okay, so I installed the esp-idf extension for vs-code and tried out their blinky program. I am now facing extension errors when I tried to build the application *sigh* : 
[image]

closed 2021-09-03 by nalostta

wawan-ikhwan, 2022-07-27:
I have same issue

Bombkimo, 2022-09-24:
> I have same issue

you have found the solution?