espressif/esp32-camera #291: event_queue can't send event after calling esp_camera_fb_return()

Arsanios, 2021-07-05:
Hi,
I am using the master of esp32-camera at commit `7c5d8b2` , the latest one right now, with ESP-IDF v4.3 release version

I am having a trouble when I call ` esp_camera_fb_return()` 
After the initialization of esp-camera, i call the `esp_camera_fb_get()` and i take the photo correctly, then when i call ` esp_camera_fb_return()`  the system starts printing

E (50579) cam_hal: EV-OVF
E (51319) cam_hal: FB-SIZE: 1916800 != 1920000
E (51599) cam_hal: EV-OVF
E (52419) cam_hal: EV-OVF
E (52729) cam_hal: EV-OVF
E (53339) cam_hal: EV-OVF

and then crashes and so on.

From the line of the mismatched size of the frame `FB-SIZE: 1916800 != 1920000`, I can understand that the esp doesn't receive the whole frame.
From analyzing the source code I found that this happens when the event_queue is not able to send the data correctly, function `ll_cam_send_event()`  .

This problem occurs only when i run the camera alongside other tasks.

Thanks

me-no-dev, 2021-07-05:
Could you please also post your camera config?

Arsanios, 2021-07-05:
> 
> 
> Could you please also post your camera config?

static camera_config_t camera_config = {
    .pin_pwdn = PWDN_GPIO_NUM,
    .pin_reset = RESET_GPIO_NUM,
    .pin_xclk = XCLK_GPIO_NUM,
    .pin_sscb_sda = SIOD_GPIO_NUM,
    .pin_sscb_scl = SIOC_GPIO_NUM,

    .pin_d7 = Y9_GPIO_NUM,
    .pin_d6 = Y8_GPIO_NUM,
    .pin_d5 = Y7_GPIO_NUM,
    .pin_d4 = Y6_GPIO_NUM,
    .pin_d3 = Y5_GPIO_NUM,
    .pin_d2 = Y4_GPIO_NUM,
    .pin_d1 = Y3_GPIO_NUM,
    .pin_d0 = Y2_GPIO_NUM,
    .pin_vsync = VSYNC_GPIO_NUM,
    .pin_href = HREF_GPIO_NUM,
    .pin_pclk = PCLK_GPIO_NUM,

    .xclk_freq_hz = 20000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,

    .pixel_format = PIXFORMAT_RAW,
    .frame_size = FRAMESIZE_UXGA, 
    .fb_count = 1 
};

I am using ESP-EYE board with OV-2640 sensor camera

TDA-2030, 2021-07-05:
From your configuration, it may be that the output raw format in large resolution causes the PCLK to be too high.

Can you measure the frequency of PCLK? I found that ESP32 doesn't work well when the frequency exceeds 15MHz, and the phenomenon is very similar to what you described.

me-no-dev, 2021-07-05:
@Arsanios you are trying to acquire 2MB of data, which in reality means 4MB of DMA data on the ESP32. It is a lot of data/operations for the ESP32 to go through, even if you are having one core to do just that. One way would be to lower the XCLK to 16MHz or 10MHz, which will slow PCLK down and allow for more time to be available for copy and your other tasks.
Definitely the best would be to have the camera running on the second core on it's own and handle the resulting frames on the first core.

Arsanios, 2021-07-05:
Hi all,
I set the xclk to 10 MHZ and now it is going very well, thank you very much

@me-no-dev Later i will try the 2 cores solution.

Thanks

closed 2021-09-08 by me-no-dev