espressif/esp32-camera #295: OV2640 HREF signal timing for JPEG

LakyG, 2021-07-14:
Hi,

I am trying to look at the timing for the HREF, VSYNC and PCLK signals outputting from the OV2640 to identify the start and end of image frames on an oscilloscope. I am having some trouble doing so when I am outputting JPEG images.

-----------------------------------------------------------------------------------------------------------------------------------------

### **For uncompressed images,** HREF is asserted during the output of each line of the image.

This is one period of VSYNC (yellow) where you can see HREF (green) assertions for each image line. (Blue line is PCLK):
[image]

Here is a zoomed in image of the same:
[image]

-----------------------------------------------------------------------------------------------------------------------------------------

### **Now for JPEG (compressed) images,** HREF is not as frequent and is irregular:
[image]

Zoomed-in HREF:
[image]

More zoomed into one of those:
[image]

^ Here you can see that the HREF signal looks kind of like data. Why is it not a regular pulsing signal, but instead different period pulses?

-----------------------------------------------------------------------------------------------------------------------------------------

I also saw in the data sheet that in the IMAGE_MODE register (address 0xDA), for JPEG images, "HREF timing select in DVP JPEG output mode" is:

> HREF = VSYNC

Reference: https://www.uctronics.com/download/cam_module/OV2640DS.pdf

Does anyone have any suggestion as to why the HREF signal changes for JPEG vs Uncompressed?

Thank you! I appreciate any help!

me-no-dev, 2021-07-14:
What you see is normal. The MCU inside the image sensor grabs the frame and starts compressing into JPEG, then send the data out when it has some to send. For uncompressed, signals are standard and behave as they should.

LakyG, 2021-07-15:
> What you see is normal. The MCU inside the image sensor grabs the frame and starts compressing into JPEG, then send the data out when it has some to send. For uncompressed, signals are standard and behave as they should.

I see. That makes sense. So essentially the compression engine waits for some part of the image, compresses that and outputs. Then the next part of the same image, compress, transmit, etc. So does that mean that it doesn't wait for the entire frame before compressing the data?

Also, does that mean that the compression time is the time between the HREF for JPEG?

me-no-dev, 2021-07-15:
- The image is always captured during VSYNC (well starts within it and takes a smidge more).
- Compression happens when HREF is inactive
- When HREF is active, you get 8 bits of data each PCLK
- Think of HREF as Slave Select, PCLK is the clock and VSYNC is your IRQ that a new frame is coming (previous ended)

LakyG, 2021-07-15:
That is really helpful! For my research project, I was trying to find the amount of time it takes for the compression of each frame at different resolutions. So now I could just measure the amount of time the HREF signal is low, within the VSYNC signal, (for JPEG) in order to find the time it takes for compression. Does that sound like it would work?

Also, out of curiosity, do you know why the HREF signal is 'random' as shown in the last image?

TDA-2030, 2021-07-16:
I think the invalid level time of HREF signal is not equal to the compression time. It at least includes the exposure time and possible delay time.

me-no-dev, 2021-07-16:
@TDA-2030 exposure happens only around vsync, then image is compressed and sent out as it's being compressed. How much time is spent not compressing the image is a bit unknown, but it is known that capture happens at once around vsync, not line by line through the time.

closed 2021-07-20 by LakyG