espressif/esp32-camera #299: Using this Library with ArduinoIDE

LakyG, 2021-07-20:
Hi,

I'm quite new to using the ESP32 and this library for the cameras and so I had searched online for how to do so. I came across tutorials for using the ArduinoIDE to download/install this library in order to use the CameraWebServer example code. However, in the files for the project, none of this library's source code is there for me to edit, just the header files for the higher level code.

Is there any way for me to edit the source of the lower level library (e.g. OV2640_settings [register settings] ) on my local project files?

I have tried following some tutorials of "ESP-IDF" but I haven't had much luck with that.

Any suggestions for how I can do that?

I would appreciate any help I can get. Thanks!

me-no-dev, 2021-07-20:
You can not compile the source with Arduino, but you can use the web server example through esp-who: https://github.com/espressif/esp-who

closed 2021-07-20 by me-no-dev

LakyG, 2021-07-20:
Is there no way to just clone this repo and use it to download my code (along with this library) onto the ESP32?

viveky1794, 2021-09-17:
> Is there no way to just clone this repo and use it to download my code (along with this library) onto the ESP32?

Step-1
-------
clone Git-hub library in same folder where you create Arduino project.

git clone https://github.com/espressif/esp32-camera

Step-2
--------
copy and paste this Code.( camera OV7670 )
Remove SD card Code. if you don't need to store data into SD card

```c

#include "esp32_camera/driver/include/esp_camera.h"
#include "esp32_camera/conversions/include/img_converters.h"
#include "esp32_camera/driver/include/sensor.h"

#include "FS.h"
#include "SD.h"
#include "SPI.h"

//WROVER-KIT PIN Map
//#define CAM_PIN_PWDN    -1 //power down is not used
//#define CAM_PIN_RESET   -1 //software reset will be performed
//#define CAM_PIN_XCLK    21
//#define CAM_PIN_SIOD    26
//#define CAM_PIN_SIOC    27
//
//#define CAM_PIN_D7      35
//#define CAM_PIN_D6      34
//#define CAM_PIN_D5 		32 // 39
//#define CAM_PIN_D4 		33 // 36
//#define CAM_PIN_D3      19
//#define CAM_PIN_D2      18
//#define CAM_PIN_D1       5
//#define CAM_PIN_D0       4
//#define CAM_PIN_VSYNC   25
//#define CAM_PIN_HREF    23
//#define CAM_PIN_PCLK    22

#define CAM_PIN_PWDN -1  // Connected to GND
#define CAM_PIN_RESET -1 //3.3V
#define CAM_PIN_XCLK 2
#define CAM_PIN_SIOD 27
#define CAM_PIN_SIOC 13

#define CAM_PIN_D7 32
#define CAM_PIN_D6 4

#define CAM_PIN_D5 33 // 39
#define CAM_PIN_D4 16 // 36

#define CAM_PIN_D3 25
#define CAM_PIN_D2 17
#define CAM_PIN_D1 26
#define CAM_PIN_D0 22
#define CAM_PIN_VSYNC 34
#define CAM_PIN_HREF 15
#define CAM_PIN_PCLK 35

static camera_config_t camera_config = {
  .pin_pwdn  = CAM_PIN_PWDN,
  .pin_reset = CAM_PIN_RESET,
  .pin_xclk = CAM_PIN_XCLK,
  .pin_sscb_sda = CAM_PIN_SIOD,
  .pin_sscb_scl = CAM_PIN_SIOC,

  .pin_d7 = CAM_PIN_D7,
  .pin_d6 = CAM_PIN_D6,
  .pin_d5 = CAM_PIN_D5,
  .pin_d4 = CAM_PIN_D4,
  .pin_d3 = CAM_PIN_D3,
  .pin_d2 = CAM_PIN_D2,
  .pin_d1 = CAM_PIN_D1,
  .pin_d0 = CAM_PIN_D0,
  .pin_vsync = CAM_PIN_VSYNC,
  .pin_href = CAM_PIN_HREF,
  .pin_pclk = CAM_PIN_PCLK,

  .xclk_freq_hz = 16000000,//EXPERIMENTAL: Set to 16MHz on ESP32-S2 or ESP32-S3 to enable EDMA mode
  .ledc_timer = LEDC_TIMER_0,
  .ledc_channel = LEDC_CHANNEL_0,

  .pixel_format = PIXFORMAT_YUV422,//YUV422,GRAYSCALE,RGB565,JPEG
  .frame_size = FRAMESIZE_QQVGA,//QQVGA-QXGA Do not use sizes above QVGA when not JPEG

  .jpeg_quality = 10, //0-63 higher number means higher quality
  .fb_count = 1, //if more than one, i2s runs in continuous mode. Use only with JPEG
  .fb_location = CAMERA_FB_IN_DRAM,
  .grab_mode = CAMERA_GRAB_WHEN_EMPTY//CAMERA_GRAB_LATEST. Sets when buffers should be filled
};

esp_err_t camera_init() {
  //power up the camera if PWDN pin is defined
  if (CAM_PIN_PWDN != -1) {
    pinMode(CAM_PIN_PWDN, OUTPUT);
    digitalWrite(CAM_PIN_PWDN, LOW);
  }

  //initialize the camera
  esp_err_t err = esp_camera_init(&camera_config);
  if (err != ESP_OK) {
    ESP_LOGE(TAG, "Camera Init Failed");
    return err;
  }

  return ESP_OK;
}

/*<=============================================================>>*/

/*<! ========================================================== !>*/

void writeFile(const char *path, const uint8_t *userBuff, int len){
  File myFile;  
  myFile = SD.open(path, FILE_WRITE);
  
  if( !myFile ){
    Serial.println("File Writing Error...");
  }
  
  myFile.write(userBuff, len);
  myFile.close(); 
}

 
// our call back to dump whatever we got in binary format
size_t jpgCallBack(void * arg, size_t index, const void* data, size_t len)
{
  static uint8_t pictureNumber=0;
  //  uint8_t* basePtr = (uint8_t*) data;
  //  for (size_t i = 0; i < len; i++) {
  //    printf("%zu, ", basePtr[i]);
  //  }
  String imagePath = "/picture_" + String(pictureNumber) + ".jpg";
  Serial.print((const char *)data);
  writeFile(imagePath.c_str(), (const uint8_t *)data, len );
  pictureNumber++;
  Serial.printf("File has been written : %s\n", imagePath.c_str());
  return 0;
}

esp_err_t camera_capture() {
  static uint8_t pictureNumber=0;
  //acquire a frame
  camera_fb_t * fb = esp_camera_fb_get();
  if (!fb) {
    ESP_LOGE(TAG, "Camera Capture Failed");
    return ESP_FAIL;
  }
  //replace this with your own function
  //    process_image(fb->width, fb->height, fb->format, fb->buf, fb->len);
 //  frame2jpg_cb(fb,  camera_config.jpeg_quality, jpgCallBack, NULL );
  
  String imagePath = "/picture_" + String(pictureNumber) + ".jpg"; 
  writeFile(imagePath.c_str(), fb->buf, fb->len );
  pictureNumber++;
  Serial.printf("File has been written : %s\n", imagePath.c_str());
  
  //return the frame buffer back to the driver for reuse
  esp_camera_fb_return(fb);
  return ESP_OK;
}

void initSdCam() {
  if (!SD.begin()) {
    Serial.println("Card Mount Failed");
    return;
  }
  uint8_t cardType = SD.cardType();

  if (cardType == CARD_NONE) {
    Serial.println("No SD card attached");
    return;
  }

  Serial.print("SD Card Type: ");
  if (cardType == CARD_MMC) {
    Serial.println("MMC");
  } else if (cardType == CARD_SD) {
    Serial.println("SDSC");
  } else if (cardType == CARD_SDHC) {
    Serial.println("SDHC");
  } else {
    Serial.println("UNKNOWN");
  }

  uint64_t cardSize = SD.cardSize() / (1024 * 1024);
  Serial.printf("SD Card Size: %lluMB\n", cardSize);

}

void setup() {
  Serial.begin(115200);
  printf("Hello World\n\n");
  delay(2000);
  initSdCam();
  camera_init();

}

void loop() {
  camera_capture();
  printf("\n\n");
  delay(2000);
}

```