espressif/esp32-camera #301: esp 32 cam erreur :  [E][camera.c:1049] camera_probe(): Detected camera not supported. [E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004

KHALEDKHADRAOUI, 2021-07-22:
Hello
I have this problem I can not configure the module
**esp 32 cam error :  [E][camera.c:1049] camera_probe(): Detected camera not supported. [E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004**

esp 32-S cam :FCCID:2AHMR-ESP32S WIFI +BT SOC INSIDE
+ convertisseur CP2102
CAMERA ref:DCV-OC2  640-V2
[image]

>

me-no-dev, 2021-07-22:
what is the camera board? you probably have your pins defined incorrectly

KHALEDKHADRAOUI, 2021-07-22:
CONNECTION CONVERTER CP 2102 TO ESP32CAM
[image]

5v ==>5
GND==>GND
TXD==>UNR
RXD==>UOT
ESP32 CAM
GND+IO0
[image]
[image]

closed 2021-07-22 by KHALEDKHADRAOUI